project-chip/connectedhomeip #31191: TARGET_CPU=arm64 ./scripts/examples/android_app_ide.sh does  not produce jar files.

DoggyBadBoy, 2023-12-26:
### Build issue(s)

1. I've followed the steps outlined in the matter build documentation for android app build.
2. Opening android and making project results in undefined references to matter
3. TARGET_CPU=arm64 ./scripts/examples/android_app_ide.sh completes successfully.
4. Searching my system fails to show that any .jar files were built that would include the references for matter.
5. Am i missing something here? Any help would be appreciated.
6. 
[image]

...

### Platform

android

### Anything else?

[image]

DoggyBadBoy, 2023-12-29:
I have not tried the android studio build yet, but after pounding my head on the desk for several hours I managed to do a script build. The script build is what actually generates the JAR files. The documentation for the android build should be updated to reflect that if the CHIPController.jar and OnboardingPayload.jar is required to build the android studio project.

closed 2023-12-29 by DoggyBadBoy

VinhBH1, 2024-06-05:
i have the same issue .
must run 
./scripts/build/build_examples.py --target android-arm64-chip-tool build
for both option 
Building Android CHIPTool from scripts or 
Building Android CHIPTool from Android Studio

pls update the doc