project-chip/connectedhomeip #31205: [INFO] [Design] Matter end-device design using clusters and endpoints

gokul-bhalodiya-7, 2023-12-31:
### Reproduction steps

I want to design matter end-device with the 3 on/off light indications.

Which is the best/most efficient method to design this device type to meet my requirements?

1. I will create the 3 endpoints with OnOff cluster support and handle these 3 indications based on endpoints.

2. I will design three different color commands for these 3 on/off light indications and handle these 3 indications based on their color on the end-device side to on/off these 3 indications.

Developing multiple endpoints(3 endpoints) with the same clusters causes any high change in memory size?

Any help/support is truly appreciated!!

### Bug prevalence

No bug

### GitHub hash of the SDK that was being used

Any

### Platform

k32w

gokul-2000, 2024-01-04:
Thank you for being so helpful in the past @bzbarsky-apple 

I configured **lighting-app** zap file by cloning endpoint-1 of the lighting-app 3 times to create endpoint-2,3,4 and compiled binary. After commissioning it with the **chip-tool**, I read information about endpoints[endpoint-,0,1,2,3,4 is created].

I tried to send On/Off commands to endpoint-1 is successful. But On/Off commands to endpoint-2,3,4 failed.

When I sent chip-tool command to read the **accepted commands list** then I observed the below logs on chip-tool side.

```
[1704294568.504348][28037:28039] CHIP:TOO: Endpoint: 1 Cluster: 0x0000_0006 Attribute 0x0000_FFF9 DataVersion: 1706451401
[1704294568.505086][28037:28039] CHIP:TOO:   AcceptedCommandList: 6 entries
[1704294568.505185][28037:28039] CHIP:TOO:     [1]: 0
[1704294568.505253][28037:28039] CHIP:TOO:     [2]: 1
[1704294568.505315][28037:28039] CHIP:TOO:     [3]: 2
[1704294568.505371][28037:28039] CHIP:TOO:     [4]: 64
[1704294568.505426][28037:28039] CHIP:TOO:     [5]: 65
[1704294568.505479][28037:28039] CHIP:TOO:     [6]: 66
---------------------------------------------------------------------------------------
[1704294584.173556][28075:28089] CHIP:TOO: Endpoint: 2 Cluster: 0x0000_0006 Attribute 0x0000_FFF9 DataVersion: 3615275775
[1704294584.174087][28075:28089] CHIP:TOO:   AcceptedCommandList: 0 entries
---------------------------------------------------------------------------------------
[1704294584.173556][28075:28089] CHIP:TOO: Endpoint: 3 Cluster: 0x0000_0006 Attribute 0x0000_FFF9 DataVersion: 3615275775
[1704294584.174087][28075:28089] CHIP:TOO:   AcceptedCommandList: 0 entries
---------------------------------------------------------------------------------------
[1704294584.173556][28075:28089] CHIP:TOO: Endpoint: 4 Cluster: 0x0000_0006 Attribute 0x0000_FFF9 DataVersion: 3615275775
[1704294584.174087][28075:28089] CHIP:TOO:   AcceptedCommandList: 0 entries
```

Is there anything I need to take care of while cloning endpoint-2,3,4 from endpoint-1?

Any help/support would be much appreciated!!

bzbarsky-apple, 2024-01-08:
@gokul-2000 there's nothing special that should happen in the above scenario.  You can have the same cluster on multiple endpoints, and the in-tree lighting app example does that.

I'm not aware of any guides for custom clusters.  There are multiple different approaches you could take, depending on how you plan to implement it, etc....  The simplest one is probably to write XML for it and add it to the build system the same was as the "Unit Testing" cluster was added, which is basically an in-tree custom cluster.