project-chip/connectedhomeip #31206: [ICD] How can we get the old key when verification/old key has not yet been stored with plain text in controller side

yunhanw-google, 2024-01-01:
Currently we have not yet stored old/verification key via ICDClientInfo in  persistent storage, and we store this key in KeyStore. When we try to refresh key after validating check-in message, we need new key and old key, but it seems we don't have API to retrieve the key from key store. Clearly it is not recommended to store the old key with plain text in persistent storage. Any good idea on this? maybe i missed something?  @bzbarsky-apple  @mkardous-silabs  thanks

bzbarsky-apple, 2024-01-02:
> When we try to refresh key after validating check-in message, we need new key and old key

Why do you need the old key?

yunhanw-google, 2024-01-02:
@bzbarsky-apple  verification key is the old key, we use it when it is not admin.
 chip-tool is usually an admin and doesn't need to have the verificationKey.
https://github.com/project-chip/connectedhomeip/blob/6ea3c34c119ec56eb425d4486d0b6a3f62742ef6/src/app/clusters/icd-management-server/icd-management-server.cpp#L250

yunhanw-google, 2024-01-02:
@bzbarsky-apple this issue may relate to the design for the callback, OnRefreshKeyGenerate, https://github.com/project-chip/connectedhomeip/pull/30731/files#r1439672537, thanks

bzbarsky-apple, 2024-01-03:
Non-admin clients have to store the old key in plaintext somewhere, yes, given our spec setup.  So yes, we need some API to get the current key for cases when we are not an admin.

Clients that always expect to be admins can just return error from that API presumably.

mkardous-silabs, 2024-01-04:
@bzbarsky-apple @yunhanw-google
Since we refactored everything server side to enable PSA storage for the keys, i think we could modify the spec to avoid clients to have a plaintext version of the key.

yunhanw-google, 2024-01-05:
@mkardous-silabs does that mean we can remove verification key from RegisterClient command in ICD Management cluster? thanks

mkardous-silabs, 2024-01-05:
@yunhanw-google It would get transformed into something else but that change is not a high priority. In the mean time, boris is correct, non-admin clients would need to store it in plaintext some where

yunhanw-google, 2024-08-27:
thanks, close it now.

yunhanw-google, 2024-08-27:
> @bzbarsky-apple @yunhanw-google Since we refactored everything server side to enable PSA storage for the keys, i think we could modify the spec to avoid clients to have a plaintext version of the key.

@mkardous-silabs  have we resolved it in server side for old key storage?