project-chip/connectedhomeip #31223: [Test Issue] Tested for OTA in esp32, found it was with very a slow speed, how can I increase BDX transfer speed?

moculll, 2024-01-03:
### Test issue(s)

The ota-provider sample can work fine, but the speed of it is sadly slow, I have a 1.7mb *.ota file, and it made alomst 10 minutes to transfer into esp32, how can I resolve this situation, thank you guys very much!!

dhrishi, 2024-01-03:
@jadhavrohit924 PTAL

jadhavrohit924, 2024-01-04:
Hi @moculll, there is API to increase download block size, you have to set the `maxDownloadBlockSize` using `SetMaxDownloadBlockSize` API [here](https://github.com/project-chip/connectedhomeip/blob/master/examples/platform/esp32/ota/OTAHelper.cpp#L78)

moculll, 2024-01-08:
> Hi @moculll, there is API to increase download block size, you have to set the `maxDownloadBlockSize` using `SetMaxDownloadBlockSize` API [here](https://github.com/project-chip/connectedhomeip/blob/master/examples/platform/esp32/ota/OTAHelper.cpp#L76)

Thank you! So the solution means that if I want to increase OTA block size, the only way is to use this api in device side, right? And there is no way to increase this in ota provider side

jadhavrohit924, 2024-01-08:
There is, I want to confirm which provider are you using esp32 or linux?

moculll, 2024-01-08:
> There is, I want to confirm which provider are you using esp32 or linux?

thanks for your reply, I tried both linux and darwin ota-provider, their speed seems the same, and I also notice one thing in matter-1.2-core-specification in page795, there says "Maximum Block Size requested by OTA Requestors over non-TCP transports SHALL be no
larger than 1024 (2^10) bytes. OTA Providers SHALL support the Maximum Block Size of at
least 1024 bytes in those cases.", does that mean if I increase blocksize in thread device like esp32h2, its MaxBlockSize is still limited to 1024bytes?

moculll, 2024-01-09:
> There is, I want to confirm which provider are you using esp32 or linux?

I have another thought about ota retry times, I tried to ota several devices like : 8 ota-linux-providers on ubuntu22.04 server, 8 device waiting for ota command, and when I started the progress, I found that almost half of the ota-linux-providers meet "stop ota progress after 3 times retries", so how can I make sure a ota progress won't stop evenif there are a few retries?

GnarFang, 2024-04-17:
Hi all,
I add  SetMaxDownloadBlockSize 10240  in  /examples/ota-requestor-app/linux/main.cpp, but final transmit is 1024
How to increase transmit size

[1713358759.544753][1056:1056] CHIP:ATM: ReceiveInit
[1713358759.544769][1056:1056] CHIP:ATM:   Proposed Transfer Control: 0x20
[1713358759.544788][1056:1056] CHIP:ATM:   Range Control: 0x0
[1713358759.544806][1056:1056] CHIP:ATM:   Proposed Max Block Size: 10240
[1713358759.544823][1056:1056] CHIP:ATM:   Start Offset: 0x0000000000000000
[1713358759.544838][1056:1056] CHIP:ATM:   Proposed Max Length: 0x0000000000000000
[1713358759.544853][1056:1056] CHIP:ATM:   File Designator Length: 29

[1713358759.889212][1056:1056] CHIP:ATM: Handling received BDX Message
[1713358759.889231][1056:1056] CHIP:ATM: ReceiveAccept
[1713358759.889245][1056:1056] CHIP:ATM:   Transfer Control: 0x20
[1713358759.889260][1056:1056] CHIP:ATM:   Range Control: 0x0
[1713358759.889277][1056:1056] CHIP:ATM:   Max Block Size: 1024
[1713358759.889293][1056:1056] CHIP:ATM:   Length: 0x0000000000000000

closed 2024-06-06 by moculll

Trishia, 2024-10-11:
@GnarFang Did you 

> Hi all, I add SetMaxDownloadBlockSize 10240 in /examples/ota-requestor-app/linux/main.cpp, but final transmit is 1024 How to increase transmit size
>
> [1713358759.544753][1056:1056] CHIP:ATM: [...]

@GnarFang I also have same problem. Did you solve this problem? If so, advise me plz.