project-chip/connectedhomeip #31224: How does nordic52840-matter generate CD, PAI and DAC instead of default certificate generate firmware

chuckrui, 2024-01-03:
I generated CD, PAI, and DAC using the following command, but I don't know how to replace the default certificate configuration of the matter project in ncs, i.e., I want to use my own generated certificate.

1.Generate the PAA and private key
./out/host/chip-cert gen-att-cert --type a --subject-cn "Matter Development PAA 01" --lifetime 7305 --out-key ./Chip-PAA-Key.der --out ./Chip-PAA-Cert.der

2.Generate PAI and private key using PAA and private key
./out/host/chip-cert gen-att-cert --type i --subject-cn "Matter Development PAI 01" --subject-vid xxxx --lifetime 7305 --ca-key ./Chip-PAA-Key.der --ca-cert ./Chip-PAA-Cert.der --out-key ./Chip-PAI-Key.der --out ./Chip-PAI-Cert.der

3.Sign the DAC and private key using the generated PAI and private key
./out/host/chip-cert gen-att-cert --type d --subject-cn "Matter Development DAC 01" --subject-vid xxxx --subject-pid 0001 --lifetime 7305 --ca-key ./Chip-PAI-Key.der --ca-cert ./Chip-PAI-Cert.der--out-key ./Chip-DAC-Key.der --out ./Chip-DAC-Cert.der

4.Verify the generated DAC,PAI, and PAA certificates
./out/host/chip-cert validate-att-cert --dac ./Chip-DAC-Cert.der --pai ./Chip-PAI-Cert.der --paa ./Chip-PAA-Cert.der

bzbarsky-apple, 2024-01-03:
@Damian-Nordic

Damian-Nordic, 2024-01-04:
@chuckrui 

The answer depends on whether you use `FactoryDataProvider` (and separate factory data partition in flash) or not.

In the first case, take a look at: https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/matter/nrfconnect_factory_data_configuration.html#creating-the-factory-data-json-file-with-the-first-script. In particular, notice the factory data generation script parameters, such as `--dac_cert` that allow you to use custom certificates.

If you don't use a separate factory data partition to store the device information, please take a look at these Kconfig options: https://github.com/project-chip/connectedhomeip/blob/master/config/nrfconnect/chip-module/Kconfig#L209C4-L209C38. I haven't tested them myself but they were contributed by some Nordic user who had the same need as you :).