project-chip/connectedhomeip #31231: [BUG] I'm sorry to stir the pot for something that may not be a Documentation bug, but the VSCODE Readme doesn'

kencottrell, 2024-01-03:
### Reproduction steps

I followed instructions from the file "connectedhomeip\docs\VSCODE_DEVELOPMENT.md"

- Windows 11 Home, Visual Studio 1.85.1, Docker Desktop. 
- WSL Ubuntu installed

Visual Studio does open WSL Ubuntu bash as an integrated terminal from menu, however there doesn't appear to be a way to get the devcontainer.json file to open bash instead of cmd.exe when the initializeCommand is called as part of Docker startup.

### Bug prevalence

every time I reopen the VSCode in a devcontainer

### GitHub hash of the SDK that was being used

not sure where to find this?

### Platform

vscode

### Anything else?

I get this error, it seems that VSCode selection of bash as default shell is not being sent to Docker image. Numerous searches indicate this is a none issue but there aren't any answers that I can find to it. 

Running the initializeCommand from devcontainer.json...

[6581 ms] Start: Run: C:\windows\system32\cmd.exe /c  .devcontainer/build.sh --tag matter-dev-environment:local --version 22
'.devcontainer' is not recognized as an internal or external command,
operable program or batch file.

olavt, 2024-01-03:
Yes, I have noticed the same problem. The documentation for how to set this up seems a bit outdated / incorrect.

I asked about this on Stackoverflow:

https://stackoverflow.com/questions/77570688/how-to-get-vs-code-devcontainer-to-execute-bash-command

I think the documentation / and or the initializeCommand should be improved to avoid everyone getting into the same issue.

kencottrell, 2024-01-03:
@olavt can you think of any work arounds for this? I've tried all sorts of things to no avail. I guess it could be a problem with VSCode but I'd be ok with editing a config file and trying things out. Is there anyone on the CHIP team who would like to spend some time with me via zoom and work this out? I can document if needed

kencottrell, 2024-01-04:
Thx olavt.

hmmm. 

when I tried that I got a similar error message but now with "bash" passed as another argument to cmd.exe. Let me see if I can clear everything out and try again. I wonder if something is cached inside VS Code that needs to be cleared out.

kencottrell, 2024-01-04:
I'm redoing this on my linux box and it appears successful. Still not sure why everyone can reset VSCode DevContainer windows to use bash vs. cmd and not me :-)

closed 2025-03-12 by Alami-Amine