project-chip/connectedhomeip #31237: [Build]  Bootstrap.sh Failing

mttrRDRSDX, 2024-01-04:
### Build issue(s)

**Trying to build bridge app , previously it was working fine till last week of previous year 

but now am facing multiple issues while bootstrapping it self and unable to build :

with existing cloned repo i faced the following issue :** 

  WELCOME TO...
 
         █
         █
     ▄   █   ▄                                █     █
     ▀▀█████▀▀      ▄▀▀▀▄ ▄▀▀▀▄    ▄▀▀▀▀▄█  ▀▀█▀▀▀▀▀█▀▀   ▄▀▀▀▀▄    ▄▀▀
   ▀█▄       ▄█▀   █     █     █  █      █    █     █    █▄▄▄▄▄▄█  █   
     ▀█▄   ▄█▀     █     █     █  █      █    █     █    █         █   
  ▄██▀▀█   █▀▀██▄  █     █     █   ▀▄▄▄▄▀█    ▀▄▄   ▀▄▄   ▀▄▄▄▄▀   █   
▀▀    █   █    ▀▀
 
  BOOTSTRAP! Bootstrap may take a few minutes; please be patient.
 
Downloading and installing packages into local source directory:
 
  Setting up CIPD package manager...done (5.2s)
warning: pattern "/home/rf6000eng1/Desktop/new/connectedhomeip/examples/bridge-app/linux/third_party/connectedhomeip/.environment/pigweed.json" matched 0 files
  Setting up Project actions........skipped (0.1s)
  Setting up Python environment.....Traceback (most recent call last):
  File "/home/rf6000eng1/Desktop/new/connectedhomeip/examples/bridge-app/linux/third_party/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 982, in <module>
    sys.exit(main())
  File "/home/rf6000eng1/Desktop/new/connectedhomeip/examples/bridge-app/linux/third_party/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 974, in main
    return EnvSetup(**vars(parse())).setup()
  File "/home/rf6000eng1/Desktop/new/connectedhomeip/examples/bridge-app/linux/third_party/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 573, in setup
    result = step(spin)
  File "/home/rf6000eng1/Desktop/new/connectedhomeip/examples/bridge-app/linux/third_party/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 762, in virtualenv
    if not virtualenv_setup.install(
  File "/home/rf6000eng1/Desktop/new/connectedhomeip/examples/bridge-app/linux/third_party/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 443, in install
    install_packages(gn_target)
  File "/home/rf6000eng1/Desktop/new/connectedhomeip/examples/bridge-app/linux/third_party/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 402, in install_packages
    subprocess.check_call(
  File "/usr/lib/python3.8/subprocess.py", line 359, in check_call
    retcode = call(*popenargs, **kwargs)
  File "/usr/lib/python3.8/subprocess.py", line 340, in call
    with Popen(*popenargs, **kwargs) as p:
  File "/usr/lib/python3.8/subprocess.py", line 858, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.8/subprocess.py", line 1704, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
**FileNotFoundError: [Errno 2] No such file or directory: 'gn'**

**so i recloned the repo and tried again to bootstrap but now am facing the following issue :** 

source ../../../scripts/bootstrap.sh 

Synchronizing submodule url for '../../common/QRCode/repo'

Synchronizing submodule url for '../../common/m5stack-tft/repo'

Synchronizing submodule url for '../../../third_party/asr/asr550x/asr_sdk'

Synchronizing submodule url for '../../../third_party/asr/asr582x/asr_sdk'

Synchronizing submodule url for '../../../third_party/asr/asr595x/asr_sdk'

Synchronizing submodule url for '../../../third_party/asr/components'

Synchronizing submodule url for '../../../third_party/boringssl/repo/src'

Synchronizing submodule url for '../../../third_party/bouffalolab/repo'

Synchronizing submodule url for '../../../third_party/cirque/repo'
[... 140 lines cut ...]

     ▄   █   ▄                                █     █

     ▀▀█████▀▀      ▄▀▀▀▄ ▄▀▀▀▄    ▄▀▀▀▀▄█  ▀▀█▀▀▀▀▀█▀▀   ▄▀▀▀▀▄    ▄▀▀

   ▀█▄       ▄█▀   █     █     █  █      █    █     █    █▄▄▄▄▄▄█  █   

     ▀█▄   ▄█▀     █     █     █  █      █    █     █    █         █   

  ▄██▀▀█   █▀▀██▄  █     █     █   ▀▄▄▄▄▀█    ▀▄▄   ▀▄▄   ▀▄▄▄▄▀   █   

▀▀    █   █    ▀▀
 
  BOOTSTRAP! Bootstrap may take a few minutes; please be patient.
 
**bash: /home/rf6000eng1/Desktop/new/connectedhomeip/examples/bridge-app/linux/third_party/connectedhomeip/.environment/pigweed-venv/bin/python3: No such file or directory

Installing pip requirements for all...

bash: /home/rf6000eng1/Desktop/new/connectedhomeip/examples/bridge-app/linux/third_party/connectedhomeip/.environment/pigweed-venv/bin/pip: No such file or directory**

i even tried checking out to the previous commit which was working from me for last 6 Months : 
**commit-ID : 5313f741b3b8c0eaf2a0dc22157fc9016e4ec4bb**

above commit was working fine for me from last  couple of months but it is also troubling now with pip version even after upgrading pip and it is successful still it says its an older one

minq-park-lge, 2024-01-11:
maybe the same issue → https://github.com/project-chip/connectedhomeip/issues/31360

andy31415, 2024-01-11:
Duplicate of #30981 - please apply #30987

closed 2024-01-11 by andy31415