project-chip/connectedhomeip #31239: [1.3] The implementation of the Content App Observer Cluster is required

Rajashreekalmane, 2024-01-04:
**Reproduction steps**

-  Executed the below mentioned command to put DUT into a commissionable state

    ./chip-tv-app

-  Once DUT reach the commissionable state sent below mentioned command on TH.

  ./chip-tool pairing onnetwork 1 20202021

-  After the successful commissioning, I attempted to validate the Content App Observer Cluster, but it is not implemented in Reference app(tv-app)

./chip-tool contentappobserver read cluster-revision 1 1

> [1704368376.361503][4141:4143] CHIP:EM: Rxd Ack; Removing MessageCounter:178936953 from Retrans Table on exchange 55544i
> [1704368376.361570][4141:4143] CHIP:DMG: ReportDataMessage =
> [1704368376.361599][4141:4143] CHIP:DMG: {
> [1704368376.361622][4141:4143] CHIP:DMG: 	AttributeReportIBs =
> [1704368376.361656][4141:4143] CHIP:DMG: 	[
> [1704368376.361683][4141:4143] CHIP:DMG: 		AttributeReportIB =
> [1704368376.361722][4141:4143] CHIP:DMG: 		{
> [1704368376.361750][4141:4143] CHIP:DMG: 			AttributeStatusIB =
> [1704368376.361783][4141:4143] CHIP:DMG: 			{
> [1704368376.361815][4141:4143] CHIP:DMG: 				AttributePathIB =
> [1704368376.361850][4141:4143] CHIP:DMG: 				{
> [1704368376.361886][4141:4143] CHIP:DMG: 					Endpoint = 0x1,
> [1704368376.361923][4141:4143] CHIP:DMG: 					Cluster = 0x510,
> [1704368376.361961][4141:4143] CHIP:DMG: 					Attribute = 0x0000_FFFD,
> [1704368376.362018][4141:4143] CHIP:DMG: 				}
> [1704368376.362060][4141:4143] CHIP:DMG: 					
> [1704368376.362091][4141:4143] CHIP:DMG: 				StatusIB =
> [1704368376.362124][4141:4143] CHIP:DMG: 				{
> [1704368376.362157][4141:4143] CHIP:DMG: 					status = 0xc3 (UNSUPPORTED_CLUSTER),
> [1704368376.362192][4141:4143] CHIP:DMG: 				},
> [1704368376.362227][4141:4143] CHIP:DMG: 				
> [1704368376.362258][4141:4143] CHIP:DMG: 			},
> [1704368376.362294][4141:4143] CHIP:DMG: 			
> [1704368376.362321][4141:4143] CHIP:DMG: 		},
> [1704368376.362354][4141:4143] CHIP:DMG: 		
> [1704368376.362380][4141:4143] CHIP:DMG: 	],
> [1704368376.362413][4141:4143] CHIP:DMG: 	
> [1704368376.362439][4141:4143] CHIP:DMG: 	SuppressResponse = true, 
> [1704368376.362466][4141:4143] CHIP:DMG: 	InteractionModelRevision = 11
> [1704368376.362491][4141:4143] CHIP:DMG: }
> [1704368376.362642][4141:4143] CHIP:TOO: Response Failure: IM Error 0x000005C3: General error: 0xc3 (UNSUPPORTED_CLUSTER)
> 
**Bug prevalence**
Whenever I do

**Platform Version(s)**
1.3

**GitHub hash of the SDK that was being used**
8686f210408dc7f5d609537ef2a7501c4ad87993

bzbarsky-apple, 2024-01-04:
@decenzo @lazarkov

chrisdecenzo, 2024-01-04:
The tv-app implements content app observer cluster as a client. The tv-casting-app implements it as a server. 

@Rajashreekalmane - would you mind re-testing against the tv-casting-app please?

closed 2024-02-06 by Rajashreekalmane