project-chip/connectedhomeip #31248: [Feature] Support for sound level measurement

olavt, 2024-01-04:
### Feature description

As far as I could see, there is no support in Matter 1.2 for sound level measurement.

This should be added.

### Platform

all

bzbarsky-apple, 2024-01-04:
This doesn't seem like an SDK issue; there's no support for this in the spec, right?

closed 2024-01-04 by bzbarsky-apple

olavt, 2024-01-05:
Where do we submit feature requests for the spec?

bzbarsky-apple, 2024-01-08:
I think in general you need to either be a company that is a CSA member or get one of those companies to submit it....

(But also: in general the parts of the spec that get worked on are the one that someone steps up and works on.  So the most likely way to get this to happen is for someone to want to ship a device that does sound level measurement and step up to write the spec for it.)

olavt, 2024-02-18:
@bzbarsky-apple I have a hard time understanding how the Matter standard evolves. I would have thought that most manufacturers of equipment consuming electricity would be interesting in having Matter support for various forms of electricity measurement. Such support have been in Z-Wave and Zigbee for many many years. When you buy a smart plug with Matter support it's virtually impossible for a consumer to understand that no devices support electric measurement using Matter, simply because it doesn't exist in the standard. Even on the CSA product list of certified devices, you will find products descriptions like this: "With Eve Energy, control your lights and appliances using your voice or an app, see how much power they’re using,...". Very misleading.