project-chip/connectedhomeip #31252: [1.2] ubuntu VScode devcontainer creates docker artifacts but then fails upon reopen

kencottrell, 2024-01-04:
### Reproduction steps

VSCode devcontainer successfully created docker images (it appears) but I get an error when I try to reopen git  repo in container

### Bug prevalence

everytime 

### GitHub hash of the SDK that was being used

can you show me where this is isted?

### Platform

vscode, other

### Type

Platform Issue

### Anything else?

[error-msg.txt](https://github.com/project-chip/connectedhomeip/files/13836071/error-msg.txt)

kencottrell, 2024-01-04:
3312 ms] **Command failed: /usr/share/code/code --ms-enable-electron-run-as-node /home/ken/.vscode/extensions/ms-vscode-remote.remote-containers-0.327.0/dist/spec-node/devContainersSpecCLI.js up --user-data-folder /home/ken/.config/Code/User/globalStorage/m**s-vscode-remote.remote-containers/data --container-session-data-folder /tmp/devcontainers-d80dd24b-ef6c-4a22-8deb-c58ee8733df91704406398255 --workspace-folder /home/ken/GITREPOS/connectedhomeip --workspace-mount-consistency cached --id-label devcontainer.local_folder=/home/ken/GITREPOS/connectedhomeip --id-label devcontainer.config_file=/home/ken/GITREPOS/connectedhomeip/.devcontainer/devcontainer.json --log-level debug --log-format json --config /home/ken/GITREPOS/connectedhomeip/.devcontainer/devcontainer.json --default-user-env-probe loginInteractiveShell --mount type=volume,source=vscode,target=/vscode,external=true --mount type=bind,source=/run/user/1000/wayland-0,target=/tmp/vscode-wayland-a4599f97-f78c-4727-a286-71670d614f53.sock --skip-post-create --update-remote-user-uid-default on --mount-workspace-git-root
[3312 ms] Exit code 1

closed 2024-01-13 by kencottrell

haerrel, 2024-01-16:
how did you solve this issue?