project-chip/connectedhomeip #31254: [RPC] rpc_console is out of work

erwinpan1, 2024-01-05:
### Test issue(s)

1. Build an example application on Linux with RPC enabled. e.g. 
```
source scripts/bootstrap.sh
cd examples/chef/
./chef.py -czbr -d rootnode_onofflight_bbs1b7IaOV -t linux
rm -rf /tmp/chip*; ./linux/out/rootnode_onofflight_bbs1b7IaOV
```

2. Create another new terminal to build and run rpc_console
```
source scripts/activate.sh
cd examples/common/pigweed/rpc_console/
gn gen out/debug
ninja -C out/debug
cd out/debug
pip install chip_rpc_console_wheels/*.whl
chip-console -s 127.0.0.1:33000
```

3. Run the sample commands, it just showed `Running` without returning any message even error
```
>>> rpcs.chip.Device.GetDeviceInfo()
Running...
```

If I use another repository cloned during Nov, the rpc_console works well
```
>>> rpcs.chip.rpc.Device.GetDeviceInfo()
(Status.OK, chip.rpc.DeviceInfo(
    vendor_id=65521,
    product_id=32768,
    software_version=1,
    serial_number='TEST_SN',
    pairing_info=chip.rpc.PairingInfo(
        code=20202021,
        discriminator=3840,
        qr_code='MT:Y.K9042C00KA0648G00',
    ),
    software_version_string='1.0',
))
```

Note: Tested at commit: 
```
commit d38a6496c3abeb3daf6429b1f11172cfa0112a1a
Author: Boris Zbarsky <bzbarsky@apple.com>
Date:   Thu Jan 4 20:30:31 2024 -0500
```

### Platform

other

andy31415, 2024-01-05:
Tried to do a careful bisect (cleaning out output and fresh venv, and supposedly #30440 is at fault - which seems possible as it is a PW update).

andy31415, 2024-01-05:
bisect on pigweed says https://github.com/google/pigweed/commit/3d96f77e48a58d236cb5c891e72ccb7969a6a18a is what breaks things.

Seems to be hdlc-related, so seems to be a valid culprit.

andy31415, 2024-01-05:
I believe the problem is we were using the `Callable[[], bytes]]` method for the console and now we need to use a cancellable. Will ask around if there are any wrappers we can use to simplify our life.

PR #31267 (2024-01-08): Switch rpc console to use CancellableReader instead of a pure read callback
This fixes #31254

Pigweed updated in google/pigweed@3d96f77 to require usage of a cancellable reader.

Tested at least with sockets that this makes printing of Device information work.
closed 2024-01-08 by mergify[bot]