project-chip/connectedhomeip #31259: [Workflow] Submodule clone faild on ubuntu

solomondong, 2024-01-05:
### Workflow issues

on ubuntu 18.04 + python 11  + x86_64
on branch test_event_7.lts

--------------------------

git submodule update --init
Submodule 'qrcode' (https://github.com/nayuki/QR-Code-generator.git) registered for path 'examples/common/QRCode/repo'
Submodule 'm5stack-tft' (https://github.com/jeremyjh/ESP32_TFT_library.git) registered for path 'examples/common/m5stack-tft/repo'
Submodule 'bluez' (https://github.com/bluez/bluez.git) registered for path 'third_party/bluez/repo'
Submodule 'cirque' (https://github.com/openweave/cirque.git) registered for path 'third_party/cirque/repo'
Submodule 'efr32_sdk' (https://github.com/SiliconLabs/sdk_support.git) registered for path 'third_party/efr32_sdk/repo'
Submodule 'freertos' (https://github.com/FreeRTOS/FreeRTOS-Kernel.git) registered for path 'third_party/freertos/repo'
Submodule 'happy' (https://github.com/openweave/happy.git) registered for path 'third_party/happy/repo'
Submodule 'mbed-os-cypress-capsense-button' (https://github.com/ARMmbed/mbed-os-cypress-capsense-button.git) registered for path 'third_party/mbed-os-cypress-capsense-button/repo'
Submodule 'third_party/mbed-os-posix-socket/repo' (https://github.com/ARMmbed/mbed-os-posix-socket.git) registered for path 'third_party/mbed-os-posix-socket/repo'
Submodule 'third_party/mbed-os/repo' (https://github.com/ARMmbed/mbed-os.git) registered for path 'third_party/mbed-os/repo'
Submodule 'mbedtls' (https://github.com/ARMmbed/mbedtls.git) registered for path 'third_party/mbedtls/repo'
Submodule 'nanopb' (https://github.com/nanopb/nanopb.git) registered for path 'third_party/nanopb/repo'
Submodule 'nlassert' (https://github.com/nestlabs/nlassert.git) registered for path 'third_party/nlassert/repo'
Submodule 'nlfaultinjection' (https://github.com/nestlabs/nlfaultinjection.git) registered for path 'third_party/nlfaultinjection/repo'
Submodule 'nlio' (https://github.com/nestlabs/nlio.git) registered for path 'third_party/nlio/repo'
Submodule 'nlunit-test' (https://github.com/nestlabs/nlunit-test.git) registered for path 'third_party/nlunit-test/repo'
Submodule 'third_party/openthread/ot-efr32' (https://github.com/openthread/ot-efr32.git) registered for path 'third_party/openthread/ot-efr32'
Submodule 'third_party/openthread/ot-nxp' (https://github.com/openthread/ot-nxp.git) registered for path 'third_party/openthread/ot-nxp'
Submodule 'third_party/openthread/ot-qorvo' (https://github.com/openthread/ot-qorvo.git) registered for path 'third_party/openthread/ot-qorvo'
Submodule 'openthread' (https://github.com/openthread/openthread.git) registered for path 'third_party/openthread/repo'
Submodule 'ot-br-posix' (https://github.com/openthread/ot-br-posix.git) registered for path 'third_party/ot-br-posix/repo'
Submodule 'p6/TARGET_CY8CKIT-062S2-43012' (https://github.com/Infineon/TARGET_CY8CKIT-062S2-43012) registered for path 'third_party/p6/p6_sdk/libs/TARGET_CY8CKIT-062S2-43012'
Submodule 'p6/abstraction-rtos' (https://github.com/Infineon/abstraction-rtos) registered for path 'third_party/p6/p6_sdk/libs/abstraction-rtos'
Submodule 'p6/bluetooth-freertos' (https://github.com/Infineon/bluetooth-freertos) registered for path 'third_party/p6/p6_sdk/libs/bluetooth-freertos'
Submodule 'p6/btstack' (https://github.com/Infineon/btstack) registered for path 'third_party/p6/p6_sdk/libs/btstack'
Submodule 'p6/clib-support' (https://github.com/Infineon/clib-support) registered for path 'third_party/p6/p6_sdk/libs/clib-support'
Submodule 'p6/connectivity-utilities' (https://github.com/Infineon/connectivity-utilities) registered for path 'third_party/p6/p6_sdk/libs/connectivity-utilities'
Submodule 'p6/core-lib' (https://github.com/Infineon/core-lib) registered for path 'third_party/p6/p6_sdk/libs/core-lib'
Submodule 'p6/core-make' (https://github.com/Infineon/core-make) registered for path 'third_party/p6/p6_sdk/libs/core-make'
Submodule 'p6/freertos' (https://github.com/Infineon/freertos) registered for path 'third_party/p6/p6_sdk/libs/freertos'
Submodule 'p6/kv-store' (https://github.com/Infineon/kv-store) registered for path 'third_party/p6/p6_sdk/libs/kv-store'
Submodule 'p6/mbedtls' (https://github.com/ARMmbed/mbedtls) registered for path 'third_party/p6/p6_sdk/libs/mbedtls'
Submodule 'p6/mtb-hal-cat1' (https://github.com/Infineon/mtb-hal-cat1) registered for path 'third_party/p6/p6_sdk/libs/mtb-hal-cat1'
Submodule 'p6/mtb-pdl-cat1' (https://github.com/Infineon/mtb-pdl-cat1) registered for path 'third_party/p6/p6_sdk/libs/mtb-pdl-cat1'
Submodule 'p6/psoc6cm0p' (https://github.com/Infineon/psoc6cm0p) registered for path 'third_party/p6/p6_sdk/libs/psoc6cm0p'
Submodule 'p6/recipe-make-cat1a' (https://github.com/Infineon/recipe-make-cat1a) registered for path 'third_party/p6/p6_sdk/libs/recipe-make-cat1a'
Submodule 'p6/retarget-io' (https://github.com/Infineon/retarget-io) registered for path 'third_party/p6/p6_sdk/libs/retarget-io'
Submodule 'p6/secure-sockets' (https://github.com/Infineon/secure-sockets) registered for path 'third_party/p6/p6_sdk/libs/secure-sockets'
Submodule 'p6/whd-bsp-integration' (https://github.com/Infineon/whd-bsp-integration) registered for path 'third_party/p6/p6_sdk/libs/whd-bsp-integration'
Submodule 'p6/wifi-connection-manager' (https://github.com/Infineon/wifi-connection-manager) registered for path 'third_party/p6/p6_sdk/libs/wifi-connection-manager'
Submodule 'p6/wifi-host-driver' (https://github.com/Infineon/wifi-host-driver) registered for path 'third_party/p6/p6_sdk/libs/wifi-host-driver'
Submodule 'p6/wifi-mw-core' (https://github.com/Infineon/wifi-mw-core) registered for path 'third_party/p6/p6_sdk/libs/wifi-mw-core'
Submodule 'pigweed' (https://github.com/google/pigweed.git) registered for path 'third_party/pigweed/repo'
Submodule 'third_party/pybind11/repo' (https://github.com/pybind/pybind11) registered for path 'third_party/pybind11/repo'
Submodule 'qpg_sdk' (https://github.com/Qorvo/qpg-connectedhomeip) registered for path 'third_party/qpg_sdk/repo'
Submodule 'simw-top-mini' (https://github.com/NXP/plug-and-trust.git) registered for path 'third_party/simw-top-mini/repo'
Submodule 'zap' (https://github.com/project-chip/zap.git) registered for path 'third_party/zap/repo'
Cloning into '/root/workspace/github/matter/examples/common/QRCode/repo'...
Cloning into '/root/workspace/github/matter/examples/common/m5stack-tft/repo'...
Cloning into '/root/workspace/github/matter/third_party/bluez/repo'...
Cloning into '/root/workspace/github/matter/third_party/cirque/repo'...
Cloning into '/root/workspace/github/matter/third_party/efr32_sdk/repo'...

fatal: the remote end hung up unexpectedly
fatal: early EOF
fatal: index-pack failed
fatal: clone of 'https://github.com/SiliconLabs/sdk_support.git' into submodule path '/root/workspace/github/matter/third_party/efr32_sdk/repo' failed
Failed to clone 'third_party/efr32_sdk/repo'. Retry scheduled
Cloning into '/root/workspace/github/matter/third_party/freertos/repo'...
Cloning into '/root/workspace/github/matter/third_party/happy/repo'...
Cloning into '/root/workspace/github/matter/third_party/mbed-os-cypress-capsense-button/repo'...
Cloning into '/root/workspace/github/matter/third_party/mbed-os-posix-socket/repo'...
[... 78 lines cut ...]
Cloning into '/root/workspace/github/matter/third_party/p6/p6_sdk/libs/wifi-connection-manager'...
Cloning into '/root/workspace/github/matter/third_party/p6/p6_sdk/libs/wifi-host-driver'...
Cloning into '/root/workspace/github/matter/third_party/p6/p6_sdk/libs/wifi-mw-core'...
Cloning into '/root/workspace/github/matter/third_party/pigweed/repo'...
fatal: the remote end hung up unexpectedly
fatal: early EOF
fatal: index-pack failed
fatal: clone of 'https://github.com/google/pigweed.git' into submodule path '/root/workspace/github/matter/third_party/pigweed/repo' failed
Failed to clone 'third_party/pigweed/repo'. Retry scheduled
Cloning into '/root/workspace/github/matter/third_party/pybind11/repo'...
Cloning into '/root/workspace/github/matter/third_party/qpg_sdk/repo'...
Username for 'https://github.com': Password for 'https://github.com':
remote: Repository not found.
fatal: Authentication failed for 'https://github.com/Qorvo/qpg-connectedhomeip/'
fatal: clone of 'https://github.com/Qorvo/qpg-connectedhomeip' into submodule path '/root/workspace/github/matter/third_party/qpg_sdk/repo' failed
Failed to clone 'third_party/qpg_sdk/repo'. Retry scheduled
Cloning into '/root/workspace/github/matter/third_party/simw-top-mini/repo'...
[... cut ...] github/matter/third_party/zap/repo'...
Cloning into '/root/workspace/github/matter/third_party/efr32_sdk/repo'...
error: RPC failed; curl 92 HTTP/2 stream 0 was not closed cleanly: CANCEL (err 8)
fatal: the remote end hung up unexpectedly
fatal: early EOF
fatal: index-pack failed
fatal: clone of 'https://github.com/SiliconLabs/sdk_support.git' into submodule path '/root/workspace/github/matter/third_party/efr32_sdk/repo' failed
Failed to clone 'third_party/efr32_sdk/repo' a second time, aborting

rhymezxcode1, 2024-01-12:
I'm also facing the same issues with cloning the submodules
[image]

ryoya-1010, 2025-04-21:
Have those issues been resolved? I'm stuck at the same place...