project-chip/connectedhomeip #31269: [Build] failed to run all tests on darwin

imsadegh, 2024-01-06:
### Build issue(s)

1. On darwin with m1 CPU, When run all tests with `ninja -C out/host check` command it will gives me this error:

> ninja: Entering directory `out/host'
> [1/5] ACTION //src/platform/tests:TestDnssd.run(//build/toolchain/host:mac_arm64_gcc)
> FAILED: gen/src/platform/tests/TestDnssd.run.pw_pystamp
> python3 ../../third_party/pigweed/repo/pw_build/py/pw_build/python_runner.py --gn-root ../../ --current-path 
> ../../src/platform/tests --default-toolchain=//build/toolchain/host:mac_arm64_gcc --current-toolchain=//build/toolchain/host:mac_arm64_gcc --touch gen/src/platform/tests/TestDnssd.run.pw_pystamp --capture-output --module pw_unit_test.test_runner --python-virtualenv-config python/gen/matter_build_venv/venv_metadata.json --python-dep-list-files gen/src/platform/tests/TestDnssd.run_metadata_path_list.txt -- --runner ../../third_party/pigweed/repo/targets/host/run_test --test tests/TestDnssd
> ERR ../../third_party/pigweed/repo/targets/host/run_test exited with status 255
> OUT [Pid: 1192]
> '#0:','CHIP DeviceLayer mDNS tests'
> [1704505891544] [1193:14623] [DL] KVS will be written to: file:///tmp/chip_kvs
> [1704505891552] [1193:14623] [DL] _Init
> '#2:','Setup                       ','PASSED'
> [1704505891552] [1193:14623] [DIS] Responding with _mock._udp.local
> [1704505891552] [1193:14623] [DIS] Responding with INSTANCE._matterc._udp.local
> [1704505891552] [1193:14623] [DIS] Responding with INSTANCE._matterc._udp.local
> [1704505891552] [1193:14623] [DIS] Responding with resolve-tester._mock._udp.local
> [1704505891552] [1193:14623] [DIS] Responding with resolve-tester._mock._udp.local
> [1704505891567] [1193:14623] [DIS] Browsing for: _mock._udp
> [1704505891568] [1193:14623] [DL] Start EventLoop
> [1704505891568] [1193:14624] [DL] _OnPlatformEvent
> [1704505891671] [1193:14624] [DIS] Mdns: OnBrowseAdd  name: INSTANCE, type: _matterc._udp., domain: local., interface: 15
> [1704505891671] [1193:14624] [DIS] Mdns: OnBrowseAdd  name: INSTANCE, type: _matterc._udp., domain: local., interface: 14
> Browse mDNS service size 2
> Service[0] name INSTANCE
> Service[0] type _matterc
> [1704505891671] [1193:14624] [DIS] Resolve type=_matterc._udp name=INSTANCE interface=15
> Service[1] name INSTANCE
> Service[1] type _matterc
> [1704505891672] [1193:14624] [DIS] Resolve type=_matterc._udp name=INSTANCE interface=14
> [1704505891740] [1193:14624] [DIS] Mdns: OnResolve flags: 0, interface: 15, fullname: INSTANCE._matterc._udp.local., hostname: resolve-tester._mock._udp.local., port: 5540
> [1704505891740] [1193:14624] [DIS] Mdns : OnNewInterface hostname:resolve-tester._mock._udp.local. fullname:INSTANCE._matterc._udp.local. interface: 15 port: 5540 TXT:"key=val"
> [1704505891742] [1193:14624] [DIS] Mdns: OnResolve flags: 0, interface: 14, fullname: INSTANCE._matterc._udp.local., hostname: resolve-tester._mock._udp.local., port: 5540
> [1704505891742] [1193:14624] [DIS] Mdns : OnNewInterface hostname:resolve-tester._mock._udp.local. fullname:INSTANCE._matterc._udp.local. interface: 14 port: 5540 TXT:"key=val"
> [1704505891855] [1193:14624] [DIS] Mdns: OnGetAddrInfo flags: 3, interface: 15, hostname: resolve-tester._mock._udp.local.
> [1704505891855] [1193:14624] [DIS] Mdns: OnNewAddress interface: 15 ip:fe80::1010:6e8e:c605:0000
> [1704505891855] [1193:14624] [DIS] Mdns: OnGetAddrInfo flags: 3, interface: 15, hostname: resolve-tester._mock._udp.local.
> [1704505891855] [1193:14624] [DIS] Mdns: OnNewAddress interface: 15 ip:fd12:f420:8e7a:a44f:18de:b50f:a981:0000
> [1704505891856] [1193:14624] [DIS] Mdns: OnGetAddrInfo flags: 2, interface: 15, hostname: resolve-tester._mock._udp.local.
> [1704505891856] [1193:14624] [DIS] Mdns: OnNewAddress interface: 15 ip:192.168.0.201
> [1704505891856] [1193:14624] [DIS] Mdns: Resolve success on interface 15
> Service[0] at [fe80::1010:6e8e:c605:a983]:5540
> [1704505891856] [1193:14624] [DIS] Mdns: OnGetAddrInfo flags: 2, interface: 14, hostname: resolve-tester._mock._udp.local.
> [1704505891856] [1193:14624] [DIS] Mdns: OnNewAddress interface: 14 ip:fe80::f02f:4bff:fe02:0000
> [1704505891856] [1193:14624] [DIS] Mdns: Resolve success on interface 14
> Service[1] at [fe80::f02f:4bff:fe02:d523]:0000
> [1704505891856] [1193:14623] [DL] End EventLoop
> [1704505891856] [1193:14623] [DIS] Mdns: Browse failure (src/platform/Darwin/DnssdImpl.cpp:412: CHIP Error 0x00000074: The operation has been cancelled)
> '#3:','Test ChipDnssdBrowse        ','PASSED'
> [1704505891856] [1193:14623] [DIS] Registering service test on host MatterTest.local. with port 80 and type: _mock._tcp on interface id: 0
> [1704505891857] [1193:14623] [DIS] Browsing for: _mock._tcp
> [1704505891857] [1193:14623] [DL] Start EventLoop
> [1704505892546] [1193:14625] [DIS] Mdns: OnRegister name: test, type: _mock._tcp., domain: local., flags: 2
> [1704505892551] [1193:14625] [DIS] Status: Satisfied
> [1704505892551] [1193:14625] [DIS] 	lo0 (1)
> [1704505892551] [1193:14625] [DIS] 		* ipv4: 127.0.0.1
> [1704505892551] [1193:14625] [DIS] 		* ipv6: ::1
> [1704505892551] [1193:14625] [DIS] 		* ipv6: fe80::1
> [1704505892553] [1193:14625] [DIS] 	en0 (15 / WiFi)
> [1704505892553] [1193:14625] [DIS] 		* ipv4: 192.168.0.201
> [1704505892553] [1193:14625] [DIS] 		* ipv6: fe80::1010:6e8e:c605:0000
> [1704505892553] [1193:14625] [DIS] 		* ipv6: fd12:f420:8e7a:a44f:18de:b50f:a981:0000
> [1704505892554] [1193:14625] [DIS] 	utun8 (29 / Other)
> [1704505892554] [1193:14625] [DIS] 		* ipv4: 172.17.255.1
> [1704505892554] [1193:14625] [DIS] 		* ipv6: fe80::f22f:4bff:fe02:0000
> [1704505892554] [1193:14625] [DIS] 		* ipv6: fd12:3456:0000::1
> [1704505892556] [1193:14625] [DIS] Mdns: OnRegisterRecord flags: 1
> [1704505892556] [1193:14625] [DIS] Mdns: OnRegisterRecord flags: 1
[... 32 lines cut ...]
> [1704505892805] [1193:14625] [DIS] Mdns: Resolve success on interface 1
> Service[0] at [::1]:80
> [1704505892806] [1193:14625] [DIS] Mdns: OnGetAddrInfo flags: 1073741827, interface: 15, hostname: MatterTest.local.
> [1704505892806] [1193:14625] [DIS] Mdns: OnNewAddress interface: 15 ip:fe80::1010:6e8e:c605:0000
> [1704505892806] [1193:14625] [DIS] Mdns: OnGetAddrInfo flags: 1073741827, interface: 15, hostname: MatterTest.local.
> [1704505892806] [1193:14625] [DIS] Mdns: OnNewAddress interface: 15 ip:fd12:f420:8e7a:a44f:18de:b50f:a981:0000
> [1704505892806] [1193:14625] [DIS] Mdns: OnGetAddrInfo flags: 1073741826, interface: 15, hostname: MatterTest.local.
> [1704505892806] [1193:14625] [DIS] Mdns: OnNewAddress interface: 15 ip:192.168.0.201
> [1704505892806] [1193:14625] [DIS] Mdns: Resolve success on interface 15
> Service[1] at [fe80::1010:6e8e:c605:0000]:80
> [1704505896859] [1193:14625] [DL] mDNS test timeout, is avahi daemon running?
> [1704505896859] [1193:14623] [DL] End EventLoop
> src/platform/tests/TestDnssd.cpp:310: assertion failed: "!context.mTimeoutExpired"
> [1704505896859] [1193:14623] [DIS] Mdns: Resolve failure (src/platform/Darwin/DnssdImpl.cpp:409: CHIP Error 0x00000074: The operation has been cancelled)
> src/platform/tests/TestDnssd.cpp:105: assertion failed: "result != nullptr"
> [1704505896859] [1193:14623] [DIS] Mdns: Browse failure (src/platform/Darwin/DnssdImpl.cpp:412: CHIP Error 0x00000074: The operation has been cancelled)
> '#3:','Test ChipDnssdPublishService','FAILED'
> [1704505896859] [1193:14623] [DL] Inet Layer shutdown
> [1704505896859] [1193:14623] [DL] BLE shutdown
> [1704505896859] [1193:14623] [DL] System Layer shutdown
> '#4:','Teardown                    ','PASSED'
> '#6:','1','2'
> '#7:','2','49'
> INF Test 1/1: [FAIL] TestDnssd in 5.332 s
> ninja: build stopped: subcommand failed.

bzbarsky-apple, 2024-01-08:
The TestDnssd test is pretty broken.  It makes a lot of assumptions that happen to be false in various cases, and are definitely false on m1/2/3 laptops.

bzbarsky-apple, 2024-01-08:
@imsadegh In the meantime, you should be able to compile with `--args='chip_enable_dnssd_tests=false'` when doing your `gn gen` step to unblock you.

kiel-apple, 2024-05-06:
failing test introduced in #26143.  still reproducible at top of tree as of [e01eb4e](https://github.com/project-chip/connectedhomeip/commit/e01eb4ea5ce3009c1cc33e264a22210abd888653).

woody-apple, 2025-01-29:
@kiel-apple please see if we need to open an internal issue here

kiel-apple, 2025-04-16:
This is still happening - investigating.

kiel-apple, 2025-06-03:
In short, the mDNS layer on Darwin is "done" sending back browse results after fewer interfaces than we're publishing services to.  #39363 fixes this by only publishing services to interfaces that can be browsed.

closed 2025-06-05 by mergify[bot]