project-chip/connectedhomeip #31271: [Build] Failed to build Chip-tool for Android

adrien-tardy, 2024-01-06:
### Build issue(s)

1. Hi !! I'm on a VM Ubuntu 22.04.3 and I'm  trying to build Chip-tool for android by following the README file located in docs/guides/android_building.md. I'm at the step 4 of "Building Android CHIPTool from Android Studio" and when I click "Make project" the build fail. I'm new to this subject and I don't get what's happening. Thank you in advance. 
 
Here's the message from the error:

Build command failed.
Error while executing process /home/aaa/Android/Sdk/cmake/3.10.2.4988404/bin/ninja with arguments {-C /home/aaa/connectedhomeip/examples/android/CHIPTool/app/.cxx/cmake/debug/arm64-v8a default}
ninja: Entering directory `/home/aaa/connectedhomeip/examples/android/CHIPTool/app/.cxx/cmake/debug/arm64-v8a'
[1/892] Building CXX object CMakeFiles/src__lib__core_error.dir/home/aaa/connectedhomeip/src/lib/core/CHIPError.cpp.o
FAILED: CMakeFiles/src__lib__core_error.dir/home/aaa/connectedhomeip/src/lib/core/CHIPError.cpp.o 
/home/aaa/Android/Sdk/ndk/23.2.8568313/toolchains/llvm/prebuilt/linux-x86_64/bin/clang++ --target=aarch64-none-linux-android24 --gcc-toolchain=/home/aaa/Android/Sdk/ndk/23.2.8568313/toolchains/llvm/prebuilt/linux-x86_64 --sysroot=/home/aaa/Android/Sdk/ndk/23.2.8568313/toolchains/llvm/prebuilt/linux-x86_64/sysroot  -DCHIP_HAVE_CONFIG_H=1 -D_LIBCPP_ENABLE_THREAD_SAFETY_ANNOTATIONS=1 -I/home/aaa/connectedhomeip/src/include -I/home/aaa/connectedhomeip/src -I/home/aaa/connectedhomeip/out/android_arm64/gen/include -I/home/aaa/connectedhomeip/zzz_generated/app-common -I/home/aaa/connectedhomeip/config/standalone -g -DANDROID -fdata-sections -ffunction-sections -funwind-tables -fstack-protector-strong -no-canonical-prefixes -D_FORTIFY_SOURCE=2 -Wformat -Werror=format-security   -fno-limit-debug-info  -fPIC   -O0 -g2 -fno-common -ffunction-sections -fdata-sections -fvisibility=hidden -fno-exceptions -funwind-tables -fPIC -Wall -Werror -Wstack-usage=8192 -Wextra -Wshadow -Wunreachable-code -Wvla -Wformat -Wformat-nonliteral -Wformat-security -Wundef -Wimplicit-fallthrough -Wheader-hygiene -Wshorten-64-to-32 -Wformat-type-confusion -Wthread-safety -Wno-deprecated-declarations -Wno-missing-field-initializers -Wno-unknown-warning-option -Wno-unused-parameter -fdiagnostics-color -fno-strict-aliasing -fmacro-prefix-map=../../= -std=gnu++17 -fno-rtti -Wnon-virtual-dtor -MD -MT CMakeFiles/src__lib__core_error.dir/home/aaa/connectedhomeip/src/lib/core/CHIPError.cpp.o -MF CMakeFiles/src__lib__core_error.dir/home/aaa/connectedhomeip/src/lib/core/CHIPError.cpp.o.d -o CMakeFiles/src__lib__core_error.dir/home/aaa/connectedhomeip/src/lib/core/CHIPError.cpp.o -c /home/aaa/connectedhomeip/src/lib/core/CHIPError.cpp
In file included from /home/aaa/connectedhomeip/src/lib/core/CHIPError.cpp:26:
/home/aaa/connectedhomeip/src/lib/core/CHIPConfig.h:39:10: fatal error: 'core/CHIPBuildConfig.h' file not found
#include <core/CHIPBuildConfig.h>
         ^~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
[2/892] Building CXX object CMakeFiles/src__lib__support_memory.dir/home/aaa/connectedhomeip/src/lib/support/CHIPPlatformMemory.cpp.o
FAILED: CMakeFiles/src__lib__support_memory.dir/home/aaa/connectedhomeip/src/lib/support/CHIPPlatformMemory.cpp.o 
/home/aaa/Android/Sdk/ndk/23.2.8568313/toolchains/llvm/prebuilt/linux-x86_64/bin/clang++ --target=aarch64-none-linux-android24 --gcc-toolchain=/home/aaa/Android/Sdk/ndk/23.2.8568313/toolchains/llvm/prebuilt/linux-x86_64 --sysroot=/home/aaa/Android/Sdk/ndk/23.2.8568313/toolchains/llvm/prebuilt/linux-x86_64/sysroot  -DCHIP_HAVE_CONFIG_H=1 -D_LIBCPP_ENABLE_THREAD_SAFETY_ANNOTATIONS=1 -I/home/aaa/connectedhomeip/src/include -I/home/aaa/connectedhomeip/src -I/home/aaa/connectedhomeip/out/android_arm64/gen/include -I/home/aaa/connectedhomeip/zzz_generated/app-common -I/home/aaa/connectedhomeip/config/standalone -I/home/aaa/connectedhomeip/third_party/nlassert/repo/include -g -DANDROID -fdata-sections -ffunction-sections -funwind-tables -fstack-protector-strong -no-canonical-prefixes -D_FORTIFY_SOURCE=2 -Wformat -Werror=format-security   -fno-limit-debug-info  -fPIC   -O0 -g2 -fno-common -ffunction-sections -fdata-sections -fvisibility=hidden -fno-exceptions -funwind-tables -fPIC -Wall -Werror -Wstack-usage=8192 -Wextra -Wshadow -Wunreachable-code -Wvla -Wformat -Wformat-nonliteral -Wformat-security -Wundef -Wimplicit-fallthrough -Wheader-hygiene -Wshorten-64-to-32 -Wformat-type-confusion -Wthread-safety -Wno-deprecated-declarations -Wno-missing-field-initializers -Wno-unknown-warning-option -Wno-unused-parameter -fdiagnostics-color -fno-strict-aliasing -fmacro-prefix-map=../../= -std=gnu++17 -fno-rtti -Wnon-virtual-dtor -MD -MT CMakeFiles/src__lib__support_memory.dir/home/aaa/connectedhomeip/src/lib/support/CHIPPlatformMemory.cpp.o -MF CMakeFiles/src__lib__support_memory.dir/home/aaa/connectedhomeip/src/lib/support/CHIPPlatformMemory.cpp.o.d -o CMakeFiles/src__lib__support_memory.dir/home/aaa/connectedhomeip/src/lib/support/CHIPPlatformMemory.cpp.o -c /home/aaa/connectedhomeip/src/lib/support/CHIPPlatformMemory.cpp
In file included from /home/aaa/connectedhomeip/src/lib/support/CHIPPlatformMemory.cpp:22:
In file included from /home/aaa/connectedhomeip/src/lib/support/CHIPMem.h:27:
In file included from /home/aaa/connectedhomeip/src/lib/core/CHIPError.h:31:
/home/aaa/connectedhomeip/src/lib/core/CHIPConfig.h:39:10: fatal error: 'core/CHIPBuildConfig.h' file not found
#include <core/CHIPBuildConfig.h>
         ^~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
[3/892] Building CXX object CMakeFiles/src__lib__support_memory.dir/home/aaa/connectedhomeip/src/lib/support/CHIPMem.cpp.o
FAILED: CMakeFiles/src__lib__support_memory.dir/home/aaa/connectedhomeip/src/lib/support/CHIPMem.cpp.o 
/home/aaa/Android/Sdk/ndk/23.2.8568313/toolchains/llvm/prebuilt/linux-x86_64/bin/clang++ --target=aarch64-none-linux-android24 --gcc-toolchain=/home/aaa/Android/Sdk/ndk/23.2.8568313/toolchains/llvm/prebuilt/linux-x86_64 --sysroot=/home/aaa/Android/Sdk/ndk/23.2.8568313/toolchains/llvm/prebuilt/linux-x86_64/sysroot  -DCHIP_HAVE_CONFIG_H=1 -D_LIBCPP_ENABLE_THREAD_SAFETY_ANNOTATIONS=1 -I/home/aaa/connectedhomeip/src/include -I/home/aaa/connectedhomeip/src -I/home/aaa/connectedhomeip/out/android_arm64/gen/include -I/home/aaa/connectedhomeip/zzz_generated/app-common -I/home/aaa/connectedhomeip/config/standalone -I/home/aaa/connectedhomeip/third_party/nlassert/repo/include -g -DANDROID -fdata-sections -ffunction-sections -funwind-tables -fstack-protector-strong -no-canonical-prefixes -D_FORTIFY_SOURCE=2 -Wformat -Werror=format-security   -fno-limit-debug-info  -fPIC   -O0 -g2 -fno-common -ffunction-sections -fdata-sections -fvisibility=hidden -fno-exceptions -funwind-tables -fPIC -Wall -Werror -Wstack-usage=8192 -Wextra -Wshadow -Wunreachable-code -Wvla -Wformat -Wformat-nonliteral -Wformat-security -Wundef -Wimplicit-fallthrough -Wheader-hygiene -Wshorten-64-to-32 -Wformat-type-confusion -Wthread-safety -Wno-deprecated-declarations -Wno-missing-field-initializers -Wno-unknown-warning-option -Wno-unused-parameter -fdiagnostics-color -fno-strict-aliasing -fmacro-prefix-map=../../= -std=gnu++17 -fno-rtti -Wnon-virtual-dtor -MD -MT CMakeFiles/src__lib__support_memory.dir/home/aaa/connectedhomeip/src/lib/support/CHIPMem.cpp.o -MF CMakeFiles/src__lib__support_memory.dir/home/aaa/connectedhomeip/src/lib/support/CHIPMem.cpp.o.d -o CMakeFiles/src__lib__support_memory.dir/home/aaa/connectedhomeip/src/lib/support/CHIPMem.cpp.o -c /home/aaa/connectedhomeip/src/lib/support/CHIPMem.cpp
In file included from /home/aaa/connectedhomeip/src/lib/support/CHIPMem.cpp:26:
/home/aaa/connectedhomeip/src/lib/core/CHIPConfig.h:39:10: fatal error: 'core/CHIPBuildConfig.h' file not found
#include <core/CHIPBuildConfig.h>
         ^~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
[4/892] Building CXX object CMakeFiles/src__lib__support_text_only_logging.dir/home/aaa/connectedhomeip/src/lib/support/logging/TextOnlyLogging.cpp.o
FAILED: CMakeFiles/src__lib__support_text_only_logging.dir/home/aaa/connectedhomeip/src/lib/support/logging/TextOnlyLogging.cpp.o 
/home/aaa/Android/Sdk/ndk/23.2.8568313/toolchains/llvm/prebuilt/linux-x86_64/bin/clang++ --target=aarch64-none-linux-android24 --gcc-toolchain=/home/aaa/Android/Sdk/ndk/23.2.8568313/toolchains/llvm/prebuilt/linux-x86_64 --sysroot=/home/aaa/Android/Sdk/ndk/23.2.8568313/toolchains/llvm/prebuilt/linux-x86_64/sysroot  -DCHIP_HAVE_CONFIG_H=1 -D_LIBCPP_ENABLE_THREAD_SAFETY_ANNOTATIONS=1 -I/home/aaa/connectedhomeip/third_party/nlas [... cut ...] yptopal_mbedtls.dir/home/aaa/connectedhomeip/src/crypto/CHIPCryptoPALmbedTLSCert.cpp.o -c /home/aaa/connectedhomeip/src/crypto/CHIPCryptoPALmbedTLSCert.cpp
In file included from /home/aaa/connectedhomeip/src/crypto/CHIPCryptoPALmbedTLSCert.cpp:24:
/home/aaa/connectedhomeip/src/crypto/CHIPCryptoPAL.h:26:10: fatal error: 'crypto/CryptoBuildConfig.h' file not found
#include <crypto/CryptoBuildConfig.h>
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
ninja: build stopped: subcommand failed.

yunhanw-google, 2024-01-08:
Hi, @joonhaengHeo can you take a look? thanks

srdantas, 2024-01-16:
I have the same problem

srdantas, 2024-01-17:
@adrien-tardy have found solution?

CaoJieNan, 2024-02-27:
I have the same problem

soulzzz, 2024-03-12:
I have the same problem

q260751600, 2024-03-16:
I have the same problem

joonhaengHeo, 2024-04-02:
Among the causes of this build error, we found that the code below needs to be modified, but it has not been resolved. I think the timing of the Build.h file creation is the problem. However, I don't know much about that, so it's difficult to solve.

--- a/scripts/examples/android_app_ide.sh
+++ b/scripts/examples/android_app_ide.sh
@@ -40,6 +40,8 @@ source scripts/activate.sh
 # Set up JARs
 python3 third_party/android_deps/set_up_android_deps.py
 
+third_party/java_deps/set_up_java_deps.sh
+
 # Build CMake for Android Studio
 echo "build ide"
-gn gen --check --fail-on-unused-args out/"android_$TARGET_CPU" --args="target_os=\"android\" target_cpu=\"$TARGET_CPU\" android_ndk_root=\"$ANDROID_NDK_HOME\" android_sdk_root=\"$ANDROID_HOME\"" --ide=json --json-ide-script=//scripts/examples/gn_to_cmakelists.py
+gn gen --check --fail-on-unused-args out/"android_$TARGET_CPU" --args="target_os=\"android\" target_cpu=\"$TARGET_CPU\" android_ndk_root=\"$ANDROID_NDK_HOME\" android_sdk_root=\"$ANDROID_HOME\" chip_build_controller_dynamic_server=true" --ide=json --json-ide-script=//scripts/examples/gn_to_cmakelists.py

VinhBH1, 2024-06-05:
the same

vulpes2, 2024-07-21:
I was never able to get it to build in Android Studio either, but I was able to build it with the scripts after fixing a couple of issues. I've written a simple troubleshooting guide in a similar issue, please try it and see it works for you. https://github.com/project-chip/connectedhomeip/issues/33212#issuecomment-2241712830

Kartikgupta1502, 2024-08-16:
I am facing the same problem, anyone have to solution?