project-chip/connectedhomeip #31295: [Build] HomePod Mini cannot add the light example of ESP32H2

suktry, 2024-01-08:
### Build issue(s)

Hi,I'm facing an issue where HomePod Mini cannot add the light example of ESP32H2. I can connect and test with Android chip using esp32s3-br, but I cannot add accessories through HomePod Mini. The following prompt appears on HomePod Mini.When I was adding Matter device accessories through the home using HomePod Mini as the central hub, after completing the pairing, the Home app displayed the message 'No Home Hub Response' in the 'Home Hub and Bridging' option. Operations such as adding/updating and removing accessories cannot be performed. Please ensure that the device is powered on and connected to Wi-Fi.
[homepod mini 失败20240107.txt](https://github.com/project-chip/connectedhomeip/files/13862907/homepod.mini.20240107.txt)
[homepodmin失败20240108.txt](https://github.com/project-chip/connectedhomeip/files/13862908/homepodmin.20240108.txt)

aajain-com, 2024-01-08:
@suktry please file a feedback assistant ticket so we can track this request and correspond with you.

Feedback assistant ticket:
- File a feedback assistant ticket.
https://developer.apple.com/bug-reporting/#feedback-assistant
- Click on the link below using your account:
https://feedbackassistant.apple.com/new-form-response
- Choose iOS/iPadOS to file an issue
- Choose the area as Home App and fill in the details of the type of issue and the description
* Technical Information
Please provide the following
a) Sysdiagnose logs for iPhone / iPad with the timestamp / timeframe of the issue
b) Sysdiagnose logs for all the Home Hubs (HomePod mini / HomePod / Apple TV) with the timestamp / timeframe of the issue
c) Accessory logs with the timestamp / timeframe of the issue
d) mDNS network logs with timestamp / timeframe of the issue
e) Video or screenshot describing the issue (optional but beneficial)
Please make sure you download and install the profiles for iOS and tvOS as provided here. The instructions are also in these links.
* iOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=homekit
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=network
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=mdns
* tvOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=homekit
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=network
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=mdns
For installation instructions, refer to https://developer.apple.com/bug-reporting/profiles-and-logs.
Note: Once you have the profile on the iPhone, it will ask you if you want to install it on other devices. Follow the prompts to install on the HomePod, Apple TV, etc., where you have the described issue and reproduce it.
Apple Developer
Bug Reporting - Apple Developer
Now with Feedback Assistant available on iPhone, iPad, Mac, and the web, it's easier to submit effective bug reports and request enhancements to APIs and tools. (71 kB)
https://developer.apple.com/bug-reporting/#feedback-assistant
developer.apple.com
Profiles and Logs - Bug Reporting - Apple Developer
Get details on providing logs, reproducible test cases, and other information that will help us investigate and diagnose your reported issues.
developer.apple.com
Profiles and Logs - Bug Reporting - Apple Developer
Get details on providing logs, reproducible test cases, and other information that will help us investigate and diagnose your reported issues.
developer.apple.com
Profiles and Logs - Bug Reporting - Apple Developer
Get details on providing logs, reproducible test cases, and other information that will help us investigate and diagnose your reported issues.

woody-apple, 2025-04-02:
Please follow up via feedback assistant, Apple cannot provide support via this standard's owned respository

closed 2025-04-02 by woody-apple