project-chip/connectedhomeip #31308: [1.3] [chip-tool] How to get list of paired devices

nmtoan91, 2024-01-09:
### Reproduction steps

Is there any way to obtain the list of paired devices on chip-tool?
I tried to search in chip-tool commands but I believe there is no command for that.

### Bug prevalence

all

### GitHub hash of the SDK that was being used

latest

### Platform

darwin, other

### Type

Test Improvement

bzbarsky-apple, 2024-01-09:
There is no such concept (e.g. a different commissioner could commision a device into the same fabric), so this is impossible to provide in general.  chip-tool itself does not store a list of things it commissioned.

closed 2024-01-09 by bzbarsky-apple

nmtoan91, 2024-03-26:
@bzbarsky-apple 
Thank you,
However, is there any way to get all/any commissioned nodes in a specific fabric?

bzbarsky-apple, 2024-03-26:
@nmtoan91 You can get the nodes that are currently operational by looking for operational advertisements.  Past that, the only way to get this list is to have something keeping track of the list.

nmtoan91, 2024-03-27:
@bzbarsky-apple 
Can you show me exactly the chip-tool command for that?
I believe the chip-tool stores the node-id somewhere in the storage folder.

bzbarsky-apple, 2024-03-29:
There is no chip-tool command for locating operational advertisements.

>  I believe the chip-tool stores the node-id somewhere in the storage folder.

It stores its own node ID, not the node IDs of the things it commissions.