project-chip/connectedhomeip #31309: [1.2] [darwin] Error on commissioning step 'FindOperational'

robinmo, 2024-01-09:
### Reproduction steps

Xcode15.1
iPhone14Pro/iOS16.3
--
commission with MatterSupport
implementation 
```
@objc class func commission(with setupPayload: MTRSetupPayload?) throws
    {
        MTRLog("MatterSupport start request")
        let topology: MatterAddDeviceRequest.Topology = .init(
            ecosystemName: "SmartHome",
            homes: [
//                .init(displayName: "smart office"),
//                .init(displayName: "smart living"),
            ]
        )
        let request = MatterAddDeviceRequest(
            topology: topology,
            setupPayload: setupPayload
        )
        Task
        {
            do
            {
                try await request.perform()
            } catch {
               throw error
            }
        }
    }
```

Always failed with
```
Error on commissioning step 'FindOperational': '../../../../../../../../Sources/CHIPFramework/connectedhomeip/src/credentials/CHIPCert.cpp:488: CHIP Error 0x0000004A: CA certificate not found'
```

### Bug prevalence

Whenever I do this

### GitHub hash of the SDK that was being used

1.2

### Platform

darwin

### Platform Version(s)

iOS16.3

### Type

Platform Issue

robinmo, 2024-01-09:
detail log
```
(base) robin @ 10:45:46 24-01-09 ~/Project/myapp_iOS3.7.1/app/build_script [v3.6.3rd+F-Secure] $ cfgutil syslog  | grep commission
Jan  9 10:51:53 iPhone homed(Matter)[2952] <Notice>: \M-p\M^_\M^_\M-" [1704768713290] [2952:1193679] CHIP: [CTL] Shutting down the commissioner
Jan  9 10:51:53 iPhone homed(Matter)[2952] <Notice>: \M-p\M^_\M^_\M-" [1704768713290] [2952:1193679] CHIP: [CTL] Stopping commissioning discovery over DNS-SD
Jan  9 10:51:53 iPhone homed(Matter)[2952] <Notice>: \M-p\M^_\M^_\M-" [1704768713523] [2952:1193679] CHIP: [CTL] Stopping commissioning discovery over DNS-SD
Jan  9 10:51:53 iPhone homed(Matter)[2952] <Notice>: \M-p\M^_\M^_\M-" [1704768713532] [2952:1193856] CHIP: [CTL] Stopping commissioning discovery over DNS-SD
Jan  9 10:51:53 iPhone homed(Matter)[2952] <Notice>: \M-p\M^_\M^T\M-5 [1704768713532] [2952:1193856] CHIP: [CTL] Starting commissioning discovery over BLE
Jan  9 10:51:53 iPhone homed(Matter)[2952] <Notice>: \M-p\M^_\M^T\M-5 [1704768713532] [2952:1193856] CHIP: [CTL] Starting commissioning discovery over DNS-SD
Jan  9 10:51:54 iPhone homed(Matter)[2952] <Notice>: \M-p\M^_\M^T\M-5 [1704768714214] [2952:1193855] CHIP: [CTL] Discovered device to be commissioned over BLE
Jan  9 10:51:55 iPhone homed(Matter)[2952] <Notice>: \M-p\M^_\M^_\M-" [1704768715986] [2952:1193856] CHIP: [CTL] Stopping commissioning discovery over DNS-SD
Jan  9 10:51:58 iPhone runningboardd(RunningBoard)[31] <Notice>: Executing launch request for xpcservice<net.mycom.ayapp.commissionRequestHandler([application<com.apple.Home.HomeUIService>:4560])> (Launching extension net.mycom.ayapp.commissionRequestHandler(5D54CF68-CEC8-4FB4-846A-056207031933) for host 4560)
Jan  9 10:51:58 iPhone runningboardd(RunningBoard)[31] <Error>: Failed to get LSApplicationRecord for <RBSLaunchRequest| xpcservice<net.mycom.ayapp.commissionRequestHandler([application<com.apple.Home.HomeUIService>:4560])>; "Launching extension net.mycom.ayapp.commissionRequestHandler(5D54CF68-CEC8-4FB4-846A-056207031933) for host 4560"> with error Error Domain=NSOSStatusErrorDomain Code=-10814 "(null)" UserInfo={_LSLine=1538, _LSFunction=runEvaluator}
Jan  9 10:51:58 iPhone runningboardd(RunningBoard)[31] <Notice>: Creating and launching job for: xpcservice<net.mycom.ayapp.commissionRequestHandler([application<com.apple.Home.HomeUIService>:4560])>
				"i" => <string: 0xd52a3f980> { length = 40, contents = "net.mycom.ayapp.commissionRequestHandler" }
			"i" => <string: 0xd52a3f980> { length = 40, contents = "net.mycom.ayapp.commissionRequestHandler" }
Jan  9 10:51:58 iPhone runningboardd(RunningBoard)[31] <Notice>: [xpcservice<net.mycom.ayapp.commissionRequestHandler([application<com.apple.Home.HomeUIService>:4560])>:4561] Memory Limits: active 80 inactive 80 <private>
Jan  9 10:51:58 iPhone runningboardd(RunningBoard)[31] <Notice>: [xpcservice<net.mycom.ayapp.commissionRequestHandler([application<com.apple.Home.HomeUIService>:4560])>:4561] This process will be managed.
Jan  9 10:51:58 iPhone runningboardd(RunningBoard)[31] <Notice>: Now tracking process: [xpcservice<net.mycom.ayapp.commissionRequestHandler([application<com.apple.Home.HomeUIService>:4560])>:4561]
Jan  9 10:51:58 iPhone runningboardd(RunningBoard)[31] <Notice>: Calculated state for xpcservice<net.mycom.ayapp.commissionRequestHandler([application<com.apple.Home.HomeUIService>:4560])>: running-suspended (role: None)
Jan  9 10:51:58 iPhone runningboardd(RunningBoard)[31] <Notice>: [xpcservice<net.mycom.ayapp.commissionRequestHandler([application<com.apple.Home.HomeUIService>:4560])>:4561] reported to RB as running
Jan  9 10:51:58 iPhone runningboardd(RunningBoard)[31] <Notice>: Acquiring assertion targeting [xpcservice<net.mycom.ayapp.commissionRequestHandler([application<com.apple.Home.HomeUIService>:4560])>:4561] from originator [application<com.apple.Home.HomeUIService>:4560] with description <RBSAssertionDescriptor| "com.apple.extension.session" ID:31-4560-28850 target:4561 attributes:[	<RBSLegacyAttribute| requestedReason:ViewService reason:ViewService flags:( AllowIdleSleep PreventTaskSuspend PreventTaskThrottleDown )>,
Jan  9 10:51:58 iPhone runningboardd(RunningBoard)[31] <Notice>: Assertion 31-4560-28850 (target:[xpcservice<net.mycom.ayapp.commissionRequestHandler([application<com.apple.Home.HomeUIService>:4560])>:4561]) will be created as active
Jan  9 10:51:58 iPhone SpringBoard(RunningBoardServices)[32] <Notice>: Hit the server for a process handle 3529d38000011d1 that resolved to: [xpcservice<net.mycom.ayapp.commissionRequestHandler([application<com.apple.Home.HomeUIService>:4560])>:4561]
Jan  9 10:51:58 iPhone runningboardd(RunningBoard)[31] <Notice>: [xpcservice<net.mycom.ayapp.commissionRequestHandler([application<com.apple.Home.HomeUIService>:4560])>:4561] Set jetsam priority to 40 [0] flag[1]
Jan  9 10:51:58 iPhone runningboardd(RunningBoard)[31] <Notice>: Calculated state for xpcservice<net.mycom.ayapp.commissionRequestHandler([application<com.apple.Home.HomeUIService>:4560])>: running-active (role: UserInteractiveNonFocal)
Jan  9 10:51:58 iPhone runningboardd(RunningBoard)[31] <Notice>: [xpcservice<net.mycom.ayapp.commissionRequestHandler([application<com.apple.Home.HomeUIService>:4560])>:4561] Resuming task.
Jan  9 10:51:58 iPhone runningboardd(RunningBoard)[31] <Notice>: [xpcservice<net.mycom.ayapp.commissionRequestHandler([application<com.apple.Home.HomeUIService>:4560])>:4561] Result 45 setting darwin role to UserInteractiveNonFocal: Ope [... cut ...] p'
Jan  9 10:52:20 iPhone homed(Matter)[2952] <Notice>: \M-p\M^_\M^T\M-5 [1704768740507] [2952:1193855] CHIP: [CTL] Performing next commissioning step 'Cleanup' with completion status = '../../../../../../../../Sources/CHIPFramework/connectedhomeip/src/credentials/CHIPCert.cpp:488: CHIP Error 0x0000004A: CA certificate not found'
Jan  9 10:52:20 iPhone homed(Matter)[2952] <Notice>: \M-p\M^_\M^T\M-5 [1704768740507] [2952:1193855] CHIP: [CTL] Successfully finished commissioning step 'Cleanup'

```

robinmo, 2024-01-09:
related issue:
#30362 #29703 

@bzbarsky-apple

robinmo, 2024-01-09:
On iOS16.7.2 iPhone 8
[sometimeSuccessSometimeFindOperationalFailed.txt](https://github.com/project-chip/connectedhomeip/files/13870666/sometimeSuccessSometimeFindOperationalFailed.txt)

1st: MatterSupport success
```
Jan  9 15:30:02 iPhone8-robingeren homed(Matter)[2692] <Notice>: Performing next commissioning step 'FindOperational'
Jan  9 15:30:04 iPhone8-robingeren homed(Matter)[2692] <Notice>: Successfully finished commissioning step 'FindOperational'
```
2nd commission into our fabric failed
```
Jan  9 15:30:23 iPhone8-robingeren robinHome(Matter)[3237] <Notice>: Error on commissioning step 'FindOperational': '../../../../../../../../Sources/CHIPFramework/connectedhomeip/src/protocols/secure_channel/CASESession.cpp:1723: CHIP Error 0x00000054: Invalid CASE parameter'
Jan  9 15:30:23 iPhone8-robingeren robinHome(Matter)[3237] <Error>: Failed to perform commissioning step 20
```
I reset the iot device and then go to
3rd MatterSupport success
```
Jan  9 15:31:38 iPhone8-robingeren homed(Matter)[2692] <Notice>: Performing next commissioning step 'FindOperational'
Jan  9 15:31:41 iPhone8-robingeren homed(Matter)[2692] <Notice>: Successfully finished commissioning step 'FindOperational'
```
4th  commission into our fabric failed
```
Jan  9 15:32:00 iPhone8-robingeren robinHome(Matter)[3237] <Notice>: Error on commissioning step 'FindOperational': '../../../../../../../../Sources/CHIPFramework/connectedhomeip/src/protocols/secure_channel/CASESession.cpp:1723: CHIP Error 0x00000054: Invalid CASE parameter'
Jan  9 15:32:00 iPhone8-robingeren robinHome(Matter)[3237] <Error>: Failed to perform commissioning step 20
```
RESET then
5th MS(MatterSupport) success
```
Jan  9 15:35:56 iPhone8-robingeren homed(Matter)[2692] <Notice>: Performing next commissioning step 'FindOperational'
Jan  9 15:35:59 iPhone8-robingeren homed(Matter)[2692] <Notice>: Successfully finished commissioning step 'FindOperational'
```
6th same failed for our fabric
```
Jan  9 15:36:21 iPhone8-robingeren robinHome(Matter)[3253] <Notice>: Error on commissioning step 'FindOperational': '../../../../../../../../Sources/CHIPFramework/connectedhomeip/src/protocols/secure_channel/CASESession.cpp:1723: CHIP Error 0x00000054: Invalid CASE parameter'
Jan  9 15:36:21 iPhone8-robingeren robinHome(Matter)[3253] <Error>: Failed to perform commissioning step 20
```
RESET and make the iot device and router into a shield box like this
[image]
7th MS success
```
Jan  9 15:40:01 iPhone8-robingeren homed(Matter)[2692] <Notice>: Performing next commissioning step 'FindOperational'
Jan  9 15:40:03 iPhone8-robingeren homed(Matter)[2692] <Notice>: Successfully finished commissioning step 'FindOperational'
```
8th success for our fabric
```
Jan  9 15:40:24 iPhone8-robingeren robinHome(Matter)[3266] <Notice>: Performing next commissioning step 'FindOperational'
Jan  9 15:40:27 iPhone8-robingeren robinHome(Matter)[3266] <Notice>: Successfully finished commissioning step 'FindOperational'
```

So it seems the 2.4G interference will affect the CASE stage?

bzbarsky-apple, 2024-01-09:
@robinmo Please file a feedback assistant ticket per the following instructions, including the logs you can get with the debugging profiles the instructions ask you to install.  Put the ticket ID here.

Feedback assistant ticket:
- File a feedback assistant ticket.
https://developer.apple.com/bug-reporting/#feedback-assistant
- Click on the link below using your account:
https://feedbackassistant.apple.com/new-form-response
- Choose iOS/iPadOS to file an issue
- Choose the area as Home App and fill in the details of the type of issue and the description
* Technical Information
Please provide the following
a) Sysdiagnose logs for iPhone / iPad with the timestamp / timeframe of the issue
b) Sysdiagnose logs for all the Home Hubs (HomePod mini / HomePod / Apple TV) with the timestamp / timeframe of the issue
c) Accessory logs with the timestamp / timeframe of the issue
d) mDNS network logs with timestamp / timeframe of the issue
e) Video or screenshot describing the issue (optional but beneficial)
Please make sure you download and install the profiles for iOS and tvOS as provided here. The instructions are also in these links.
* iOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=homekit
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=network
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=mdns
* tvOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=homekit
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=network
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=mdns
For installation instructions, refer to https://developer.apple.com/bug-reporting/profiles-and-logs.
Note: Once you have the profile on the iPhone, it will ask you if you want to install it on other devices. Follow the prompts to install on the HomePod, Apple TV, etc., where you have the described issue and reproduce it.

robinmo, 2024-01-10:
@bzbarsky-apple 
I found there is a related pr: #26372 
could this issue
```
Error on commissioning step 'FindOperational': '../../../../../../../../Sources/CHIPFramework/connectedhomeip/src/credentials/CHIPCert.cpp:488: CHIP Error 0x0000004A: CA certificate not found'
```
does matter with the pr (validating expiration time for a root certificate)

robinmo, 2024-01-30:
capture key log: 

```
Jan 30 11:16:26 iPhone14Pro homed(Matter)[173] <Notice>: \M-p\M^_\M^_\M-" [1706584586170] [173:571372] CHIP: [EM] CHIP MessageCounter:1460362 not in RetransTable on exchange 39357i
Jan 30 11:16:26 iPhone14Pro homed(Matter)[173] <Notice>: \M-p\M^_\M^T\M-5 [1706584586170] [173:571372] CHIP: [SC] Received Sigma2 msg
Jan 30 11:16:26 iPhone14Pro homed(Matter)[173] <Notice>: \M-p\M^_\M^_\M-" [1706584586171] [173:571372] CHIP: [SC] Peer assigned session session ID 16575
Jan 30 11:16:26 iPhone14Pro homed(Matter)[173] <Notice>: \M-p\M^_\M^_\M-" [1706584586187] [173:571372] CHIP: [SC] Certificate's mNotAfterTime (756034214) is before current time (759899786)
```
Is it related to #20637: Fix validity times on certificates issued by the Darwin framework
It seems fixed in new version.(So on iOS 16.3 maybe use the old version code?)
@bzbarsky-apple

bzbarsky-apple, 2024-02-03:
That is possible....

closed 2024-02-20 by robinmo

huchuankan, 2024-03-14:
Matter Commission  demo video ： https://www.bilibili.com/video/BV1Nx421y7sa/