project-chip/connectedhomeip #31316: [BUG] TLVReader error "Wrong TLV type" when use CustomArgumentParser.

nykh2010, 2024-01-09:
### Reproduction steps

pair all-cluster-app demo with chip-tool, send command: 
`./chip-tool-1.2.0.1 colorcontrol command-by-id 8 '{"0":10,"1":"s:20","2":0,"3":0}' 12345 1`, 
the all-cluster-app shows an error log below:
```
[1704804890.455678][346309:346309] CHIP:DMG: InvokeRequestMessage =
[1704804890.455680][346309:346309] CHIP:DMG: {
[1704804890.455682][346309:346309] CHIP:DMG: 	suppressResponse = false, 
[1704804890.455685][346309:346309] CHIP:DMG: 	timedRequest = false, 
[1704804890.455687][346309:346309] CHIP:DMG: 	InvokeRequests =
[1704804890.455690][346309:346309] CHIP:DMG: 	[
[1704804890.455692][346309:346309] CHIP:DMG: 		CommandDataIB =
[1704804890.455695][346309:346309] CHIP:DMG: 		{
[1704804890.455697][346309:346309] CHIP:DMG: 			CommandPathIB =
[1704804890.455712][346309:346309] CHIP:DMG: 			{
[1704804890.455714][346309:346309] CHIP:DMG: 				EndpointId = 0x1,
[1704804890.455716][346309:346309] CHIP:DMG: 				ClusterId = 0x300,
[1704804890.455718][346309:346309] CHIP:DMG: 				CommandId = 0x8,
[1704804890.455740][346309:346309] CHIP:DMG: 			},
[1704804890.455744][346309:346309] CHIP:DMG: 			
[1704804890.455747][346309:346309] CHIP:DMG: 			CommandFields = 
[1704804890.455749][346309:346309] CHIP:DMG: 			{
[1704804890.455756][346309:346309] CHIP:DMG: 				0x0 = 10, 
[1704804890.455758][346309:346309] CHIP:DMG: 				0x1 = 20, 
[1704804890.455760][346309:346309] CHIP:DMG: 				0x2 = 0, 
[1704804890.455762][346309:346309] CHIP:DMG: 				0x3 = 0, 
[1704804890.455764][346309:346309] CHIP:DMG: 			},
[1704804890.455765][346309:346309] CHIP:DMG: 		},
[1704804890.455768][346309:346309] CHIP:DMG: 		
[1704804890.455770][346309:346309] CHIP:DMG: 	],
[1704804890.455773][346309:346309] CHIP:DMG: 	
[1704804890.455774][346309:346309] CHIP:DMG: 	InteractionModelRevision = 11
[1704804890.455776][346309:346309] CHIP:DMG: },
[1704804890.455789][346309:346309] CHIP:DMG: AccessControl: checking f=1 a=c s=0x000000000001B669 t= c=0x0000_0300 e=1 p=o
[1704804890.455795][346309:346309] CHIP:DMG: AccessControl: allowed
[1704804890.455798][346309:346309] CHIP:DMG: Received command for Endpoint=1 Cluster=0x0000_0300 Command=0x0000_0008
[1704804890.455803][346309:346309] CHIP:DMG: Endpoint=1 Cluster=0x0000_0300 Command=0x0000_0008 status 0x85 (INVALID_COMMAND) (no additional context)
[1704804890.455807][346309:346309] CHIP:DMG: Command handler moving to [ Preparing]
[1704804890.455810][346309:346309] CHIP:DMG: Command handler moving to [AddingComm]
[1704804890.455813][346309:346309] CHIP:DMG: Command handler moving to [AddedComma]
[1704804890.455816][346309:346309] CHIP:ZCL: Failed to dispatch command, TLVError=src/lib/core/TLVReader.cpp:186: CHIP Error 0x00000026: Wrong TLV type
```
I found that the reason is that the chip-tool treats 10 and 20 as uint8 type. Why doesn't MoveColor automatically treat unsigned integers as signed?

### Bug prevalence

always

### GitHub hash of the SDK that was being used

181b0cb14ff007ec912f2ba6627e05dfb066c008

### Platform

other

### Platform Version(s)

Linux

bzbarsky-apple, 2024-01-09:
> Why doesn't MoveColor automatically treat unsigned integers as signed?

Because that would be a spec violation.  Those are different TLV types, and per spec if the wrong type is used the command is rejected.

As you found, you can use `"0":"s:10"` to indicate that the 10 is signed.

closed 2024-01-09 by bzbarsky-apple

nykh2010, 2024-01-10:
> > Why doesn't MoveColor automatically treat unsigned integers as signed?
> 
> Because that would be a spec violation. Those are different TLV types, and per spec if the wrong type is used the command is rejected.
> 
> As you found, you can use `"0":"s:10"` to indicate that the 10 is signed.

I use like chip-tool any command mode as rpc request, so I'm not sure about the specific data type before send request. Upon your reply, I should reformat the data type according to the cluster data model.

bzbarsky-apple, 2024-01-10:
> I should reformat the data type according to the cluster data model

Yes.