project-chip/connectedhomeip #31317: [Tooling] automated running of python tests in CI

cecille, 2024-01-09:
### Reproduction steps / Feature

We have a script that goes through all the yamls and runs them automatically. We should probably have something similar for the python tests because right now they need to be manually added.

### Platform

python

### Type

Manually tested with SDK

Apollon77, 2024-04-17:
If I could add a wish then it would be cool to allow to specify the started executables comparable to the yaml tooling ... in fact matter.js uses the same tooling to just start a nodejs  scropt as executable instead of the device.

cecille, 2025-04-29:
This is basically done, will track improvements as individual issues.

closed 2025-04-29 by cecille