project-chip/connectedhomeip #31332: [Build] I try to build using android studio, but always get an error.I need the latest generated CHIPController.jar, SetupPayloadParser.jar, AndroidPlatform.jar files.

JasonClark2, 2024-01-10:
### Build issue(s)

1.I need the latest generated CHIPController.jar, SetupPayloadParser.jar, AndroidPlatform.jar files.
2.Now I got three jar packages from Google's demo, but they don't seem to be the latest. Can you create a folder to put the latest jar packages? The experience of compiling by myself is really terrible, there are too many errors.
3.
...

### Platform

android

### Anything else?

Please put the jar package in a separate folder to reduce the bad compilation experience.
The user experience for android studio is very bad, and many people give up because of this, https://stackoverflow.com/questions/76975250/project-chip-connectedhomeip-chip-android-sample-app-setup

yunhanw-google, 2024-01-11:
If this is urgent and you have ubuntu/debian machine, you can do the below to generate these jar.

git clone https://github.com/project-chip/connectedhomeip.git
cd connectedhomeip
git pull
git submodule init
git submodule update
docker run -it -v ~/workspace/connectedhomeip:/connectedhomeip ghcr.io/project-chip/chip-build-android:32 
git config --global --add [safe.directory](http://safe.directory/) "*"
source scripts/bootstrap.sh
./scripts/build/build_examples.py --target android-arm64-chip-tool build

JasonClark2, 2024-03-27:
I use Ubuntu to compile, but I still get an error and cannot generate the final jar package. What else do I need to do?