project-chip/connectedhomeip #31346: [1.2] While reading the event-list of the Operational State cluster, I encountered an unsupported attribute in the result

VaishaliAvhale, 2024-01-10:
### Reproduction steps

1. build and execute linux all-cluster-app and commission it
2.  Execute given command :   ./chip-tool operationalstate read event-list 1 1
Expected result of above command is 0 if the only mandatory event supported or 0 and 1 if both mandatory and optional event are supported

### Bug prevalence

Everytime

### GitHub hash of the SDK that was being used

418b5987ce72f3a20c55be91e232a77f37e9e00b

### Platform

core

### Platform Version(s)

v1.2

### Type

Spec Compliance Issue

bzbarsky-apple, 2024-01-10:
EventList is provisional and not currently supported in the SDK unless you build with `enable_eventlist_attribute=true`

closed 2024-01-10 by bzbarsky-apple

VaishaliAvhale, 2024-01-11:
@bzbarsky-apple I can see the OPSTAT cluster having one mandatory event so it should be listed once I read the event list of that cluster why do I need to enable that? 
(https://github.com/project-chip/connectedhomeip/issues/31346#issuecomment-1885923794)
>> As per 1.2 Matter Application spec event-list is not a provisional attribute and not even event 0x0000 event of OPSTAT is provisional.

bzbarsky-apple, 2024-01-12:
@VaishaliAvhale Again: the EventList attribute is not actually supported by default in the Matter SDK right now.