project-chip/connectedhomeip #31348: [1.3] When an event is triggered on an end device, no reports get sent out

mkardous-silabs, 2024-01-10:
### Reproduction steps

1. Build Light-Switch App and flash
2. Chip-Tool in interactive mode
3. Subscribe to all events 
`any subscribe-event-by-id 0xFFFFFFFF 0xFFFFFFFF 0 10 <nodeid> 0xFFFF`
4. Trigger event - Nothing gets sent out

Attribute reporting works as expected but not events.

### Bug prevalence

Systematically 

### GitHub hash of the SDK that was being used

6b21935fde

### Platform

core

### Type

Core SDK Crash

### Anything else?

**Device Side logs**
```
00> [00:00:40.786][info  ][SC] Received Sigma1 msg
00> [00:00:40.787][detail][SC] Found MRP parameters in the message
00> [00:00:40.787][detail][SC] Peer assigned session key ID 37309
00> [00:00:40.794][info  ][EM] <<< [E:22221r S:0 M:165291602 (Ack:216700616)] (U) Msg TX to 0:0000000000000000 [0000] [UDP:[fd51:b85e:5b91:1:875c:204a:10fb:a185]:54706] --- Type 0000:33 (SecureChannel:CASE_Sigma2Resume)
00> [00:00:40.795][detail][SC] Sent Sigma2Resume msg
00> [00:00:41.284][detail][IN] UDP Message Received packet nb : 7 SrcAddr : fd51:b85e:5b91:1:875c:204a:10fb:a185[54706] DestAddr : fd51:b85e:5b91:1:56e7:13b5:430f:ba33[5540] Payload Length 34
00> [00:00:41.284][info  ][EM] >>> [E:22221r S:0 M:216700617 (Ack:165291602)] (U) Msg RX from 0:03047483AF67520A [0000] --- Type 0000:40 (SecureChannel:StatusReport)
00> [00:00:41.284][detail][EM] Found matching exchange: 22221r, Delegate: 0x200253e4
00> [00:00:41.285][detail][EM] Rxd Ack; Removing MessageCounter:165291602 from Retrans Table on exchange 22221r
00> [00:00:41.285][info  ][SC] Success status report received. Session was established
00> [00:00:41.306][detail][IN] UDP Message Received packet nb : 8 SrcAddr : fd51:b85e:5b91:1:875c:204a:10fb:a185[54706] DestAddr : fd51:b85e:5b91:1:56e7:13b5:430f:ba33[5540] Payload Length 50
00> [00:00:41.341][info  ][SC] SecureSession[0x200245c0, LSID:28544]: State change 'kEstablishing' --> 'kActive'
00> [00:00:41.341][detail][IN] SecureSession[0x200245c0]: Activated - Type:2 LSID:28544
00> [00:00:41.341][detail][IN] New secure session activated for device <000000000001B669, 1>, LSID:28544 PSID:37309!
00> [00:00:41.341][info  ][IN] CASE Session established to peer: <000000000001B669, 1>
00> [00:00:41.342][detail][IN] SecureSession[0x20024690]: Allocated Type:2 LSID:28545
00> [00:00:41.342][detail][SC] Allocated SecureSession (0x20024690) - waiting for Sigma1 msg
00> [00:00:41.343][info  ][EM] <<< [E:22221r S:0 M:165291603 (Ack:216700617)] (U) Msg TX to 0:0000000000000000 [0000] [UDP:[fd51:b85e:5b91:1:875c:204a:10fb:a185]:54706] --- Type 0000:10 (SecureChannel:StandaloneAck)
00> [00:00:41.343][detail][EM] Flushed pending ack for MessageCounter:216700617 on exchange 22221r
00> [00:00:41.344][info  ][EM] >>> [E:22222r S:28544 M:158648690] (S) Msg RX from 1:000000000001B669 [8783] --- Type 0001:03 (IM:SubscribeRequest)
00> [00:00:41.345][detail][EM] Handling via exchange: 22222r, Delegate: 0x2001468c
00> [00:00:41.345][detail][IM] Received Subscribe request
00> [00:00:41.345][info  ][DMG] Final negotiated min/max parameters: Min = 0s, Max = 3600s
00> [00:00:41.346][detail][DMG] IM RH moving to [CanStartReporting]
00> [00:00:41.357][detail][DMG] Building Reports for ReadHandler with LastReportGeneration = 0x0000000000000000 DirtyGeneration = 0x0000000000000000
00> [00:00:41.358][detail][DMG] Fetched 4 events
00> [00:00:41.358][detail][DMG] <RE> Sending report (payload has 176 bytes)...
00> [00:00:41.359][info  ][EM] <<< [E:22222r S:28544 M:179194078 (Ack:158648690)] (S) Msg TX to 1:000000000001B669 [8783] [UDP:[fd51:b85e:5b91:1:875c:204a:10fb:a185]:54706] --- Type 0001:05 (IM:ReportData)
00> [00:00:41.360][detail][DMG] IM RH moving to [AwaitingReportResponse]
00> [00:00:41.360][detail][DMG] <RE> ReportsInFlight = 1 with readHandler 0, RE has no more messages
00> [00:00:41.360][detail][DMG] All ReadHandler-s are clean, clear GlobalDirtySet
00> [00:00:41.804][detail][IN] UDP Message Received packet nb : 9 SrcAddr : fd51:b85e:5b91:1:875c:204a:10fb:a185[54706] DestAddr : fd51:b85e:5b91:1:56e7:13b5:430f:ba33[5540] Payload Length 42
00> [00:00:41.805][info  ][EM] >>> [E:22222r S:28544 M:158648691 (Ack:179194078)] (S) Msg RX from 1:000000000001B669 [8783] --- Type 0001:01 (IM:StatusResponse)
00> [00:00:41.805][detail][EM] Found matching exchange: 22222r, Delegate: 0x200161f4
00> [00:00:41.805][detail][EM] Rxd Ack; Removing MessageCounter:179194078 from Retrans Table on exchange 22222r
00> [00:00:41.806][info  ][IM] Received status response, status is 0x00
00> [00:00:41.807][info  ][EM] <<< [E:22222r S:28544 M:179194079 (Ack:158648691)] (S) Msg TX to 1:000000000001B669 [8783] [UDP:[fd51:b85e:5b91:1:875c:204a:10fb:a185%ot]:54706] --- Type 0001:04 (IM:SubscribeResponse)
00> [00:00:41.807][info  ][DMG] Registered a ReadHandler that will schedule a report between system Timestamp: 0x000000000000A34F and system Timestamp 0x00000000003791CF.
00> [00:00:41.808][detail][DMG] <RE> OnReportConfirm: NumReports = 0
00> [00:00:41.808][detail][DMG] IM RH moving to [CanStartReporting]
[... 262 lines cut ...]
[1704919030.753955][4510:4512] CHIP:TOO:    }
[1704919030.754046][4510:4512] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_0038 Event 0x0000_0003
[1704919030.754070][4510:4512] CHIP:TOO:   Event number: 589827
[1704919030.754090][4510:4512] CHIP:TOO:   Priority: Info
[1704919030.754129][4510:4512] CHIP:TOO:   Timestamp: 946684801499
[1704919030.754156][4510:4512] CHIP:TOO:   TimeFailure: {
[1704919030.754179][4510:4512] CHIP:TOO:    }
[1704919030.754306][4510:4512] CHIP:DMG: MoveToState ReadClient[0xffff8000a540]: Moving to [AwaitingSu]
[1704919030.754638][4510:4512] CHIP:EM: <<< [E:22222i S:37309 M:158648691 (Ack:179194078)] (S) Msg TX to 1:0000000000004C08 [8783] [UDP:[fd51:b85e:5b91:1:56e7:13b5:430f:ba33%wpan0]:5540] --- Type 0001:01 (IM:StatusResponse)
[1704919031.179922][4510:4512] CHIP:EM: >>> [E:22222i S:37309 M:179194079 (Ack:158648691)] (S) Msg RX from 1:0000000000004C08 [8783] --- Type 0001:04 (IM:SubscribeResponse)
[1704919031.180000][4510:4512] CHIP:EM: Found matching exchange: 22222i, Delegate: 0xffff8000a550
[1704919031.180078][4510:4512] CHIP:EM: Rxd Ack; Removing MessageCounter:158648691 from Retrans Table on exchange 22222i
[1704919031.180150][4510:4512] CHIP:DMG: SubscribeResponse is received
[1704919031.180225][4510:4512] CHIP:DMG: SubscribeResponseMessage =
[1704919031.180551][4510:4512] CHIP:DMG: {
[1704919031.180924][4510:4512] CHIP:DMG: 	SubscriptionId = 0xdb88210c,
[1704919031.180982][4510:4512] CHIP:DMG: 	MaxInterval = 0xe10,
[1704919031.181071][4510:4512] CHIP:DMG: 	InteractionModelRevision = 11
[1704919031.181114][4510:4512] CHIP:DMG: }
[1704919031.181164][4510:4512] CHIP:DMG: Subscription established with SubscriptionID = 0xdb88210c MinInterval = 0s MaxInterval = 3600s Peer = 01:0000000000004C08
[1704919031.181248][4510:4512] CHIP:DMG: MoveToState ReadClient[0xffff8000a540]: Moving to [Subscripti]
[1704919031.181328][4510:4512] CHIP:DMG: Refresh LivenessCheckTime for 3604224 milliseconds with SubscriptionId = 0xdb88210c Peer = 01:0000000000004C08
[1704919031.181769][4510:4512] CHIP:EM: <<< [E:22222i S:37309 M:158648692 (Ack:179194079)] (S) Msg TX to 1:0000000000004C08 [8783] [UDP:[fd51:b85e:5b91:1:56e7:13b5:430f:ba33%wpan0]:5540] --- Type 0000:10 (SecureChannel:StandaloneAck)
[1704919031.182014][4510:4512] CHIP:EM: Flushed pending ack for MessageCounter:179194079 on exchange 22222i
```

yunhanw-google, 2024-01-10:
may I know how to trigger event?

mkardous-silabs, 2024-01-10:
On the Linux app, i'm not sure how to trigger the event. Test event has a way (TC_SWTCH_...)
On a silabs boards, BTN1 triggers the switch events

bzbarsky-apple, 2024-01-10:
So I just tried the following steps to reproduce, after compiling the relevant binaries:

Terminal 1:
```
% ./out/debug/standalone/chip-all-clusters-app
```

Terminal 2:
```
% ./out/debug/standalone/chip-tool pairing code 1 MT:-24J042C00KA0648G00
% ./out/debug/standalone/chip-tool interactive start  
>>>  any subscribe-event-by-id 0xFFFFFFFF 0xFFFFFFFF 0 10 1 0xFFFF
```

Terminal 3:
```
% echo '{ "Name": "SoftwareFault" }' > /tmp/chip_all_clusters_fifo_200
```

In terminal 1 I see:
```
[1704929918822] [200:7919384] [EVL] LogEvent event number: 0x0000000000000005 priority: 1, endpoint id:  0x0 cluster id: 0x0000_0034 event id: 0x0 Epoch timestamp: 0x0000018CF5BDFF66
```

After a short delay in terminal 2 I see:
```
[1704929924908] [218:7918881] [DMG] ReportDataMessage =
[1704929924908] [218:7918881] [DMG] {
[1704929924908] [218:7918881] [DMG] 	SubscriptionId = 0xc0936fe8,
[1704929924908] [218:7918881] [DMG] 	EventReportIBs =
...
[1704929924909] [218:7918881] [TOO] Endpoint: 0 Cluster: 0x0000_0034 Event 0x0000_0000
[1704929924909] [218:7918881] [TOO]   Event number: 5
[1704929924909] [218:7918881] [TOO]   Priority: Info
[1704929924909] [218:7918881] [TOO]   Timestamp: 1704929918822
[1704929924909] [218:7918881] [TOO]   SoftwareFault: {
[1704929924909] [218:7918881] [TOO]     Id: 200
[1704929924909] [218:7918881] [TOO]     Name: 200
[1704929924909] [218:7918881] [TOO]     FaultRecording: 576564204A616E2031302031383A33383A33382032303234
[1704929924909] [218:7918881] [TOO]    }
```
```

yunhanw-google, 2024-01-11:
locally when running the below against all-cluster-app
./out/linux-x64-chip-tool/chip-tool interactive start
pairing onnetwork-long  1 20202021 3840
unittesting subscribe-event test-event 0 20 1 1
unittesting test-emit-test-event-request 1 2 false 1 1

I am able to receive the test event whenever maximum period is reached

bzbarsky-apple, 2024-01-11:
I just tried the following also:

1. Compile all-clusters-app for esp32 (m5stack).
2. Commission it with chip-tool and node id 1.
3. Run the following:

```
% ./out/debug/standalone/chip-tool interactive start  
>>>  any subscribe-event-by-id 0xFFFFFFFF 0xFFFFFFFF 0 10 1 0xFFFF --is-urgent true
>>> accesscontrol write acl '[{"privilege": 5, "authMode": 2, "subjects": null, "targets": null}]' 1 0
```
I immediately get subscription reports:
```
[1704990075471] [54161:8230327] [TOO] Endpoint: 0 Cluster: 0x0000_001F Event 0x0000_0000
[1704990075471] [54161:8230327] [TOO]   Event number: 5
[1704990075471] [54161:8230327] [TOO]   Priority: Info
[1704990075471] [54161:8230327] [TOO]   Timestamp: 209453
[1704990075471] [54161:8230327] [TOO]   AccessControlEntryChanged: {
```
etc.

mkardous-silabs, 2024-01-16:
The reason why the event reports were not being sent out was because the subscription request did not have the `--is-urgent` flag which requests the report to be as soon as possible.

closed 2024-01-16 by mkardous-silabs