project-chip/connectedhomeip #31360: [BUG] error while source activate.sh

minq-park-lge, 2024-01-11:
### Reproduction steps

After activate.sh, I see this message

connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/pw_cli/plugins.py", line 96, in from_name
    module = importlib.import_module(module_name)
  File "/usr/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/home/worker/git/repo-manifest/clip.all/CLIPonRTOS/devices/ameba-z2/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/pw_watch/watch.py", line 93, in <module>
    import pw_console.python_logging
  File "/home/worker/git/repo-manifest/clip.all/CLIPonRTOS/devices/ameba-z2/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/pw_console/__init__.py", line 16, in <module>
    from pw_console.embed import PwConsoleEmbed
  File "/home/worker/git/repo-manifest/clip.all/CLIPonRTOS/devices/ameba-z2/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/pw_console/embed.py", line 23, in <module>
    from pw_console.console_app import ConsoleApp
  File "/home/worker/git/repo-manifest/clip.all/CLIPonRTOS/devices/ameba-z2/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/pw_console/console_app.py", line 59, in <module>
    from ptpython.layout import CompletionVisualisation  # type: ignore
  File "/home/worker/git/repo-manifest/clip.all/CLIPonRTOS/devices/ameba-z2/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/ptpython/__init__.py", line 3, in <module>
    from .repl import embed
  File "/home/worker/git/repo-manifest/clip.all/CLIPonRTOS/devices/ameba-z2/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/ptpython/repl.py", line 23, in <module>
    from prompt_toolkit.formatted_text import OneStyleAndTextTuple
ImportError: cannot import name 'OneStyleAndTextTuple' from 'prompt_toolkit.formatted_text' (/home/worker/git/repo-manifest/clip.all/CLIPonRTOS/devices/ameba-z2/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/prompt_toolkit/formatted_text/__init__.py)

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/worker/git/repo-manifest/clip.all/CLIPonRTOS/devices/ameba-z2/connectedhomeip/.environment/pigweed-venv/bin/pw", line 8, in <module>
    sys.exit(main())
  File "/home/worker/git/repo-manifest/clip.all/CLIPonRTOS/devices/ameba-z2/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/pw_cli/__main__.py", line 43, in main
    pw_command_plugins.register()
  File "/home/worker/git/repo-manifest/clip.all/CLIPonRTOS/devices/ameba-z2/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/pw_cli/pw_command_plugins.py", line 65, in register
    _register_builtin_plugins(plugin_registry)
  File "/home/worker/git/repo-manifest/clip.all/CLIPonRTOS/devices/ameba-z2/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/pw_cli/pw_command_plugins.py", line 43, in _register_builtin_plugins
    registry.register_by_name('watch', 'pw_watch.watch', 'main')
  File "/home/worker/git/repo-manifest/clip.all/CLIPonRTOS/devices/ameba-z2/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/pw_cli/plugins.py", line 303, in register_by_name
    Plugin.from_name(name, module_name, member_name, source)
  File "/home/worker/git/repo-manifest/clip.all/CLIPonRTOS/devices/ameba-z2/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/pw_cli/plugins.py", line 104, in from_name
    raise Error(f'Failed to import module "{module_name}"') from err
pw_cli.plugins.Error: Failed to import module "pw_watch.watch" (ImportError: cannot import name 'OneStyleAndTextTuple' from 'prompt_toolkit.formatted_text' (/home/worker/git/repo-manifest/clip.all/CLIPonRTOS/devices/ameba-z2/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/prompt_toolkit/formatted_text/__init__.py))
Error during activate--see messages above.

### Bug prevalence

Whenver I do this

### GitHub hash of the SDK that was being used

52e4a6435f

### Platform

ameba

### Platform Version(s)

1.2

minq-park-lge, 2024-01-11:
bootstrap.sh process is also the same :(

```

  WELCOME TO...

         █
         █
     ▄   █   ▄                                █     █
     ▀▀█████▀▀      ▄▀▀▀▄ ▄▀▀▀▄    ▄▀▀▀▀▄█  ▀▀█▀▀▀▀▀█▀▀   ▄▀▀▀▀▄    ▄▀▀
   ▀█▄       ▄█▀   █     █     █  █      █    █     █    █▄▄▄▄▄▄█  █   
     ▀█▄   ▄█▀     █     █     █  █      █    █     █    █         █   
  ▄██▀▀█   █▀▀██▄  █     █     █   ▀▄▄▄▄▀█    ▀▄▄   ▀▄▄   ▀▄▄▄▄▀   █   
 ▀▀    █   █    ▀▀

  BOOTSTRAP! Bootstrap may take a few minutes; please be patient.

Downloading and installing packages into local source directory:

  Setting up CIPD package manager...done (17.7s)
  Setting up Project actions........skipped (0.1s)
  Setting up Python environment.....done (1m27.5s)
  Setting up pw packages............skipped (0.1s)
  Setting up Host tools.............done (0.1s)

Activating environment (setting environment variables):

  Setting environment variables for CIPD package manager...done
  Setting environment variables for Project actions........skipped
  Setting environment variables for Python environment.....done
  Setting environment variables for pw packages............skipped
  Setting environment variables for Host tools.............done

Checking the environment:

Traceback (most recent call last):
  File "/home/worker/git/repo-manifest/clip.all/CLIPonRTOS/devices/ameba-z2/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/pw_cli/plugins.py", line 96, in from_name
    module = importlib.import_module(module_name)
  File "/usr/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/home/worker/git/repo-manifest/clip.all/CLIPonRTOS/devices/ameba-z2/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/pw_watch/watch.py", line 93, in <module>
    import pw_console.python_logging
  File "/home/worker/git/repo-manifest/clip.all/CLIPonRTOS/devices/ameba-z2/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/pw_console/__init__.py", line 16, in <module>
    from pw_console.embed import PwConsoleEmbed
  File "/home/worker/git/repo-manifest/clip.all/CLIPonRTOS/devices/ameba-z2/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/pw_console/embed.py", line 23, in <module>
    from pw_console.console_app import ConsoleApp
  File "/home/worker/git/repo-manifest/clip.all/CLIPonRTOS/devices/ameba-z2/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/pw_console/console_app.py", line 59, in <module>
    from ptpython.layout import CompletionVisualisation  # type: ignore
  File "/home/worker/git/repo-manifest/clip.all/CLIPonRTOS/devices/ameba-z2/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/ptpython/__init__.py", line 3, in <module>
    from .repl import embed
  File "/home/worker/git/repo-manifest/clip.all/CLIPonRTOS/devices/ameba-z2/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/ptpython/repl.py", line 23, in <module>
    from prompt_toolkit.formatted_text import OneStyleAndTextTuple
ImportError: cannot import name 'OneStyleAndTextTuple' from 'prompt_toolkit.formatted_text' (/home/worker/git/repo-manifest/clip.all/CLIPonRTOS/devices/ameba-z2/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/prompt_toolkit/formatted_text/__init__.py)

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/worker/git/repo-manifest/clip.all/CLIPonRTOS/devices/ameba-z2/connectedhomeip/.environment/pigweed-venv/bin/pw", line 8, in <module>
    sys.exit(main())
  File "/home/worker/git/repo-manifest/clip.all/CLIPonRTOS/devices/ameba-z2/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/pw_cli/__main__.py", line 43, in main
    pw_command_plugins.register()
  File "/home/worker/git/repo-manifest/clip.all/CLIPonRTOS/devices/ameba-z2/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/pw_cli/pw_command_plugins.py", line 65, in register
    _register_builtin_plugins(plugin_registry)
  File "/home/worker/git/repo-manifest/clip.all/CLIPonRTOS/devices/ameba-z2/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/pw_cli/pw_command_plugins.py", line 43, in _register_builtin_plugins
    registry.register_by_name('watch', 'pw_watch.watch', 'main')
  File "/home/worker/git/repo-manifest/clip.all/CLIPonRTOS/devices/ameba-z2/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/pw_cli/plugins.py", line 303, in register_by_name
    Plugin.from_name(name, module_name, member_name, source)
  File "/home/worker/git/repo-manifest/clip.all/CLIPonRTOS/devices/ameba-z2/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/pw_cli/plugins.py", line 104, in from_name
    raise Error(f'Failed to import module "{module_name}"') from err
pw_cli.plugins.Error: Failed to import module "pw_watch.watch" (ImportError: cannot import name 'OneStyleAndTextTuple' from 'prompt_toolkit.formatted_text' (/home/worker/git/repo-manifest/clip.all/CLIPonRTOS/devices/ameba-z2/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/prompt_toolkit/formatted_text/__init__.py))
Error during bootstrap--see messages above.
Installing pip requirements for all...

[notice] A new release of pip is available: 23.2.1 -> 23.3.2
[notice] To update, run: pip install --upgrade pip
```

andy31415, 2024-01-11:
OneStyleAndTextTuple error is due to prompt_toolkit update. you need #30987 patched in I imagine.

andy31415, 2024-01-11:
Closing as duplicate of #30981

closed 2024-01-11 by andy31415