project-chip/connectedhomeip #31379: [ICD] Add TestEventTrigger for LIT that support enable/disable and specify specify the LIT time to maybe 30 seconds or so which can be used

yunhanw-google, 2024-01-12:


yunhanw-google, 2024-01-13:
I think we also need to use TestEventTrigger to adjust operatingMode attribute between sit and lit for dynamic ICD feature validation during certification test. @lpbeliveau-silabs @jmartinez-silabs @mkardous-silabs

jmartinez-silabs, 2024-01-13:
That should be done by registering/unregistering the monitoring client

yunhanw-google, 2024-01-13:
@jmartinez-silabs could you elaborate a little bit more between monitoring client registration and dynamic ICD switch? Thanks

tcarmelveilleux, 2024-01-15:
We need to be able to:

- Disable LIT behavior entirely for test purposes, so that an ICD configured by a commissioner to have LIT registration can be tested for non-LIT tests without crazy time-outs. This could be done at commissioning time before commissioning complete. Otherwise, it's possible you have to wait for long check-in delay before you can unregister.
- Force the LIT behavior to be at a max sleep interval that is much lower than what the product would do in the field, so that test cases can be done with delays of >= ~30s (e.g. just out of SIT bounds), rather than waiting for real hours.

mkardous-silabs, 2024-01-17:
@tcarmelveilleux @yunhanw-google agreed on both posts. The test event trigger will allow to force specific states for different tests cases.

The only thing i don't think is necessary is the bits for dynamic lit / sit ie. Smoke&Co tests. We won't need the test event trigger; not for the ICD bits at least.

yunhanw-google, 2024-01-17:
Agree. Thanks

yunhanw-google, 2024-03-13:
We also need one way to test RefreshKey, maybe create a initial counter + uint32_max via TestEventTrigger.

closed 2024-04-10 by mergify[bot]