project-chip/connectedhomeip #31383: [BUG] About Chip-tool Cross Compilation Running in Android Devices with Errors: Configuration Manager initialization failed: src/platform/android/AndroidConfig.cpp:329: CHIP Error 0x00000003: Incorrect state

shuiYe-704265085, 2024-01-12:
### Reproduction steps

I have compiled and run the openthread agent service into an Android device through cross compilation, and through rcp and esp idf/examples/openthread/ot-cli example successfully connects cli to the thread network and establishes udp communication, but when I compiled the chip-tool through cross compilation i found that it is different from running in Linux, and it requires relevant configuration information(gAndroidConfigObject).  The command is ./chip-tool pairing ble-thread ${NODE-ID} hex: ${DATASET} ${PIN-CODE} ${DISCRIMINATOR}
Upon checking the code, it was found that the configuration information is that the APP initializes the AndroidChipPlatform instance and JNI calls the underlying layer for initialization. 
The error is as follows: Configuration Manager initialization failed: src/platform/android/androidConfiguration. cpp: 329: CHIP Error 0x00000003: Incorrect state. I am not sure if there is any association between the cross compiled binary program of chip tool and the APP. Can you provide me with relevant documentation or help? Thank you very much

### Bug prevalence

every time

### GitHub hash of the SDK that was being used

d1b59df60a74ac5d73e606619e68752d1c65cb27

### Platform

android

yunhanw-google, 2024-01-14:
it seems you wanna c++ chip-tool as binary thing inside android platform?

shuiYe-704265085, 2024-01-15:
> it seems you wanna c++ chip-tool instead of android chip-tool as binary thing inside android platform? why don't you run android chip-tool?

I previously encountered timeout issues and related problems while attempting to run the openthread otbr service on Linux and use the chip-tool Android APP to guide the thread terminal device for network access. This process was unsuccessful, and given that I couldn't find any related documentation for the APP, I'm unsure whether it requires special configurations. However, the binary format of chip-tool can successfully assist the thread device in gaining network access on Linux by generating thread network credentials using the command (chip-tool pairing ble-thread ${NODE_ID} hex:${DATASET} ${PIN_CODE} ${DISCRIMINATOR}). Consequently, I want to try cross-compiling chip-tool into ARM format and execute it in the Android background. However, I found that chip-tool needs to initialize configurations in the background, but the configuration process gets interrupted as it needs to be reset via upper-layer JNI.

shuiYe-704265085, 2024-01-15:
By the way, I'd like to ask if there are any related APP usage documents. I have added a Thread Border Router hub through the app by following these steps: Privision chip device with Wi-Fi -> Input Device Address -> Discover -> Commission. However, I'm not sure how to add thread child devices in this router. If I try to add a thread device directly through Privision chip device with thread -> Scan QR Code -> Use default configuration, it returns error code 50. Looking up the documentation indicates this is a timeout error, specifically CHIP_ERROR_TIMEOUT.

yunhanw-google, 2024-03-08:
@shuiYe-704265085 , @joonhaengHeo has locally tried, and cannot reproduce it , could you attach the detailed log using android chip-tool? thanks