project-chip/connectedhomeip #31389: [Feature] "Audio Player Architecture" with a new"Casting Audio Player device type" and "Audio Player endpoint" ("Casting Audio Player" and adding "Basic Audio Player") + look into adding multi-room audio support?

Hedda, 2024-01-12:
### Feature description

Home it is OK so submit this large feature request that I do not have the skills or resources to implement this myself. That is, this is just a feature suggestion meant for open discussion, not a feature proposal as such.

Anyway, the feature request is that I ask others to consider having a seperate Matter architecture for a "Audio Player Architecture" and adding new "Casting Audio Player device type" and "Audio Player endpoint" ("Casting Audio Player" and "Basic Audio Player").

Would also needs example similar to `tv-casting-app` but perhaps instead for purpose of audio casting, so `music-casting-app` or?

https://github.com/project-chip/connectedhomeip/blob/master/examples/tv-casting-app/APIs.md

Anyway, such "Audio Player Architecture" could be based on the existing Video Player Architecture, and the new "Casting Audio Player device type" could be based on the existing "Casting Video Player device type". However think there needs to be differences.

While some video-specific features could be removed, should preferably also extend with some audio-specific features, such example real-time audio synchronization to allow for synchronized multi-room audio playback of music on several Matter Audio Casting enabled speakers installed in same home, (also known as a distributed audio system). This needs support for "Audio Group", usually named "Speaker Group" and perhaps also "Audio Zone" ("Speaker Zone" or area). Preferably also need to have separate volume controls for each speaker and/or zone to compensate for differences in apparent volume due to room size and shape as well as speaker products used in different rooms.  

https://en.wikipedia.org/wiki/Multi-room_audio

Multi-room audio:
* Single source, single zone = Scenario: One music app streaming to single audio player/speaker).
* Single source, multiple zone = Scenario: One music app streaming to audio group containing multiple audio players/speakers.
* Multiple source, multiple zones =  Scenario: Several music apps streaming to seperate audio players/speakers in same home.

Another argument for a seperate audio-only architecture and audio player specialized for just music playback could enable it to run with even less resources on constrained devices. 

Background: The existing Matter specification does feature a "Video Player Architecture" with a "Casting Video Player device type" and "Video Player endpoint" ("Casting Video Player" and "Basic Video Player"). What looks to currently missing but directly related is a "pure" audio architecture with Matter Casting Audio Player device type, and maybe an Audio Input cluster as well? 

An alternative could be to redesign and rename the "Video Player Architecture" into a more generic "Media Player Architecture"?

Product use case:  A client/server design that works well for music/audio apps and smart speaker products, + products with audio line-in, (i.e. devices designed for only pure audio outout and/or input that are normally used just music playback, including multi-room sound systems. Probably sometimes but not always including microphone input for voice assistant. The point is that it means products that lack any kind of video output like with video screens such as televisions and/or smart control display have).

I think that other than the obvious smart speakers with voice assistant, another real-world market and use-case for a pure audio players are high-quality speakers and Hi-Fi grade sound systems for music playback, wheather or not they would be used for  being used as a single-point and multi-room sound system for music playback,  their primary audience would probably be users of music streaming from example streaming music from different commercial music services apps like Amazon Music, Spotify, SiriusXM, Pandora, Tidal, Qobuz, Deezer, YouTube Music, Apple Music, if and when they add support “Matter Casting” streaming protocol for audio to their apps.

If implemented, please be sure to include support for the concept of audio-output dongles (with TOSLINK and Phono AUX-out or line-out ports for external speakers and sound systems from third-parties), such as the classic  "Chromecast Audio" product which enabled adding Google Cast audio player capability to any third-party speaker / sound system, as well as "Amazon Echo Input" and "Amazon Echo Flex" which similarly also adds AUX output to third-party speaker / sound system (but Amazon Echo products also have embedded voice assistant via built-in microphones).

New product idea to consider would to also include support for the concept of audio-input dongles. That is, audio-steaming server dongles with input ports that act as embedded for streaming "Matter Casting Client” for audio-only which can be streamed to set "Audio Player endpoint", which can be a single endpoint of a grouped endpoints (audio group) for multiroom music playback. This would allow a user to connect any legacy audio source, like an LP record player (phonograph turntable),  cassette deck, or CD-player to a such audio-input dongle and stream that audio to any “Matter Casting” enabled audio player.

### Platform

all

### Anything else?

Perhaps an existing Matter group member would be willing to contribute their existing solution as a base for multi-room support?

Amazon Alexa features multi-room music support:

- https://www.amazon.com/alexa-multi-room-audio/b?ie=UTF8&node=21480962011

Google has "Google Cast" which supports multi-room audio with grouping of speakers and multiroom synchronized playback so maybe they could be convinced to contribute components?

- https://en.wikipedia.org/wiki/Google_Cast

Apple feature multiroom support for AirPlay 2 audio streaming:

- https://support.apple.com/en-us/HT202809

Espressif ESP-ADF (Espressif Audio Development Framework ) do support ESP Multi-Room Music but not synchronized on its own?

- https://github.com/espressif/esp-adf
    - https://github.com/espressif/esp-adf/tree/master/examples/advanced_examples/multi-room

IKEA of Sweden AB currently has a partnership with Sonos to make Wi-Fi speakers with multi-rooms audio support:

- https://www.ikea.com/global/en/stories/design/symfonisk-product-story-230131/
  - https://www.ikea.com/global/en/newsroom/collaborations/symfonisk-is-here--ikea-and-sonos-to-rethink-the-way-we-use-sound-and-light-190604/
    - https://www.ikea.com/us/en/cat/wi-fi-speakers-46194/

There are also other open-source and closed-source multi-room audio solutions for multi-room audio synchronisation. For example:

Snapcast 

- https://github.com/badaix/snapcast
  - https://github.com/bridadan/libsnapcast

strobe audio

- https://strobe.audio

bzbarsky-apple, 2024-01-12:
@decenzo please take a look.

chrisdecenzo, 2024-01-15:
Thanks for the suggestion. Please join Matter so you can volunteer to lead this effort!

Hedda, 2024-01-16:
Sorry, I do not have the capacity for that myself. Would think that Amazon and/or Google might be best suited to look into this? 

Again, both Amazon and Google have competing smart speakers with their own technology that implements all these features.

Is there someone from Amazon who leads Matter's matching "Video Player Architecture" + "Casting Video Player" development?

Referencing to Amazon driving "Matter Casting" for their video playback devices and them already having many smart speakers:

https://www.aboutamazon.com/news/devices/amazon-ces-2024-announcements

https://www.streamtvinsider.com/video/amazon-drops-matter-casting-capabilities-panasonic-and-fire-tv-os-partnership-ces

https://9to5google.com/2024/01/09/amazon-fire-tv-matter-casting/

https://www.theverge.com/2024/1/9/24030324/amazon-matter-casting-echo-show-fire-tv-prime-video

https://www.aftvnews.com/matter-casting-is-coming-to-fire-tvs-and-the-echo-show-15-an-industry-first-by-amazon/

https://www.aftvnews.com/new-matter-casting-video-player-for-fire-tv-gets-certified/

Hedda, 2024-05-21:
@marcelveldt as the Matter guy at Nabu Casa and Home Assistant perhaps this is something that you and ESPHome developers at Nabu Casa would be interested in helping architecture and develop now that Home Assistant has announced both your "Music Assistant" integration and the Open Home Foundation + the Home Assistant's voice assistant work, which are separate things that I believe all align in spirit with this?

https://www.home-assistant.io/blog/2024/05/09/music-assistant-2/

https://www.openhomefoundation.org/blog/announcing-the-open-home-foundation/

https://www.home-assistant.io/blog/2024/04/24/state-of-the-open-home-2024/

PS: I also read that Nabu Casa is developing your own ESPHome based smart-speakers (and/or smart-display) hardware, and for those to also work as music streaming/receiver endpoints without your "Music Assistant" integration I am guessing you are eventually going to want to add cross-ecosystem support for some kind of standard audio streaming protocol like Matter Casting?

Hedda, 2024-06-16:
Any feedback or input on these feature request ideas anout audio-only casting and multi-room support for syncronized music playback in multiple rooms?

chrisdecenzo, 2024-06-17:
There is an effort within Matter right now to define use cases for audio players / smart speakers. Please join us!

jonsmirl, 2024-07-04:
> There is an effort within Matter right now to define use cases for audio players / smart speakers. Please join us!

@chrisdecenzo There are groups of developers (myself included) who work on Matter projects but can't participate in CHIP because our employers won't join the CSA for various reasons. CHIP would benefit from having an 'Invited Expert' membership (like the W3C has) allowing individuals caught in this situation to participate. 'Invited Expert' would allow access to Slack and the draft spec with no voting ability.

jonsmirl, 2024-08-06:
Don't waste your time with the proprietary chips, we wasted a year on them. Instead use a generic ARM CPU running Linux.  Amazon has an Alexa qualified hot word processing library for ARM if you can convince them to give it to you. Another alternative is to use TensorFlow and an ARM CPU with an NPU. Google around github and you'll find example code. Allwinner and Rockchip make low cost CPUs specifically designed for this use case.

Hedda, 2024-08-06:
Off-topic but FYI, even if XMOS is proprietary hardware their chips are very popular and have open-source compatible libraries.

https://github.com/xmos

By the way, I stumbled on this new "***voice-kit***" GitHub repository where [ESPHome](https://esphome.io/) firmware developers are developing new and improved components for I2S audio, including XMOS output and input support, as well as a their own ESP32 native media player with support for FLAC, WAV/PCM, MP3, etc. for the upcoming Home Assistant voice-kit hardware platform from Nabu Casa (that as mentioned will be based on a combination on an XMOS xCORE chip and ESP32-S3):

* https://github.com/esphome/voice-kit/
  * https://github.com/esphome/voice-kit/tree/dev/esphome/components
    * https://github.com/esphome/voice-kit/tree/dev/esphome/components/nabu
    * https://github.com/esphome/voice-kit/tree/dev/esphome/components/i2s_audio
      * https://github.com/esphome/voice-kit/tree/dev/esphome/components/i2s_audio/speaker
      * https://github.com/esphome/voice-kit/tree/dev/esphome/components/i2s_audio/microphone
    * https://github.com/esphome/voice-kit/tree/dev/esphome/components/media_player

They already added features and functions or improvements/enhancements to ESPHome, such as:

* *New: Nabu Media Player - new "nabu" media player from Nabu Casa running natively on ESP32*
  * *[Music Assistant](https://www.home-assistant.io/blog/2024/05/09/music-assistant-2/) streams work (both mp3 and flac), but since it requires resampling, the audio quality isn't great*
* *New: Added support for FLAC files*
* *New: Added a proper WAV decoder (that parses WAV headers with LIST, INFO, etc. chunks.)*
* *New: Initial support for playing back local files*
* *New: Playback Control for the VoiceKit*
* *New: Added an `is_paused` condition for media players.*
* *New: Add Click to Converse to button*
* *New: LED animation*
* *New: Scripts for controlling LEDs*
* *New: Update Button Behaviour for the Voice kit*
* *New: Dial Volume Control*
* *New: Timer basic implementation*
* *New: Dial Volume Control*
* *New: Added HTTP(s) OTA updates*
* *New: Dial Volume Control*
* *New: Added Buttons for force ota update.*
* *New: Software Mute Switch*
* *Improvement: A basic resampler adjusts sample rates*
* *Improvement: Configurable output sample rate (for experimental 48kHz XMOS firmware)*
* *Improvement: The DAC mute state is read on boot*
* *Improvement: volume/mute control via the DAC (the wheel works for increasing/decreasing volume)*
* *Improvement: Logs what element failed if the pipeline breaks*
* *Improvement: Fails gracefully if the incoming stream can't be processed*
* *Improvement: Differentiate between user facing LED Ring and Internal LED ring*
* *Point external component to dev branch*

They also have many TODO inline coments in the code there if anyone are interested in helping them:

https://github.com/search?q=repo%3Aesphome%2Fvoice-kit%20todo&type=code

Note! Be aware that there are many comments there to that most of the new stuff are not yet stable.

Hedda, 2024-11-07:
> There is an effort within Matter right now to define use cases for audio players / smart speakers.

@chrisdecenzo any updates on Matter Casting use cases to add music playback via dedicated audio players and smart speakers?

Read that Matter 1.4 seems to be adding messaging for Matter Casting but again is only focusing on video and smart displays.

* https://www.matteralpha.com/explainer/what-is-matter-casting

Looks at least like HiFi Streamers / Music Streamers (network audio players) are trending and becoming more popular as products.

Again, there are no open standards/protocols for multi-room audio (distributed audio) systems made for home multiroom audio.

chrisdecenzo, 2025-03-12:
Yes, the smart speaker effort is ongoing!

Hedda, 2025-03-17:
Great to hear! Hope for open standard that is able to compete on feature-parity for audio-only with Squeezelite (Squeezebox client): 

- Up to 768 kHz max sample rate for PCM.
- High Resolution Streaming Support.
- Lossless stream support.
- Perfect sync for synchronized multi-room audio between clients.
- Sync Correction.
- Crossfade Support.
- Gapless playback support.
- Stereo Pair support (pairing two separate mono-speakers into one stereo pair).
- Direct streaming.
- Grouping clients.
- Audio input.
- Bi-directional control with syncronization (so volume can be controlled on server or client via the transport protocol)
- Let the audio device control its own clock.

* https://en.wikipedia.org/wiki/Squeezelite

  * https://lyrion.org/players-and-controllers/squeezelite/

    * https://github.com/ralph-irving/squeezelite

      * https://sourceforge.net/projects/lmsclients/files/squeezelite/

Hedda, 2025-04-25:
> > Please join Matter so you can volunteer to lead this effort!
> 
> [@marcelveldt](https://github.com/marcelveldt) as "the Matter guy at Nabu Casa" (and Home Assistant) perhaps this is something that you and ESPHome developers at Nabu Casa would be interested in helping architecture and develop for the Matter project? I would think this functionality might be very relative to your roadmap now that Home Assistant's recent announcents about both your "[Music Assistant](https://github.com/music-assistant/)" and the Open Home Foundation + the Home Assistant's voice assistant work, which are separate things that I believe all align in spirit with this concept at a high-level?
> 
> * https://www.home-assistant.io/blog/2024/05/09/music-assistant-2/
> * https://www.openhomefoundation.org/blog/announcing-the-open-home-foundation/
>   
>   * https://www.home-assistant.io/blog/2024/04/24/state-of-the-open-home-2024/
> 
> PS: I also read that Nabu Casa is developing your own [ESPHome](https://github.com/esphome) based smart-speakers (and/or smart-display) hardware, and for such devices to also work as music streaming and audio player (A/V-receiver endpoint) without your "Music Assistant" integration acting as middleware I am guessing you are eventually going to want to add support for some cross-ecosystem support for some kind of standardized audio streaming protocol like Matter Casting with audio player endpoint features?
> 
> * ["Matter Casting" (a.k.a. MatterCast) audio/music and video player (streaming reciever) for new/upcoming video and music cast standard? esphome/feature-requests#2723](https://github.com/esphome/feature-requests/issues/2723)

In related news and a good argument to work with Home Assistant and Music Assistent developers on this new streaming protocol is that both Home Assistant and Music Assistant as well and the lead developers working on those projects has now officially transfered to the non-profit [Open Home Foundation](https://www.openhomefoundation.org/about/), read:

- https://www.home-assistant.io/blog/2025/04/16/state-of-the-open-home-recap/

Or checkout their ”State of the Open Home 2025” video (live stream recording from studio):

- https://www.youtube.com/live/o4Vctz1_KYE?si=KjDhZs0Rlzdu66fa

FYI, the "Works with Home Assistent" partner program was also transfered to the Open Home Foundation:

- https://www.home-assistant.io/blog/2024/08/08/works-with-home-assistant-becomes-part-ohf/

PS: By the way, another argument to work with Home Assistant and Music Assistent developers on this is that the Home Assistant core was recently announced as the top open-source project on GitHub with contributions from the most amount of people (and that statistic only included the main Home Assistant core repository):

https://github.blog/news-insights/octoverse/octoverse-2024/

Hedda, 2025-09-18:
@Daasin what the heck are you talking about? Actually, do not answer. Let me just tell you that you gone way off on a tangent and are so very so off-topic that there is no point in you replying to explain. First of all, the Matter Casting specification has nothing to do video conferencing or making some kind of ”calls”, and secondly, this request is only discussing audio casting for music playback (as in ”multi-room audio” and ”music streamer”), not about making two-way calls, so doorbells or landline phones will never be relevant to this topic. Please move along.

Hedda, 2026-08-28:
For reference, Matter 1.7 specification will introduce an Audio Control cluster (0x0512) as part of the Streaming/Casting Speaker device types:

It is not the audio streaming for HiFi music platbacl asked for here, but at least it will provide a standardized building block designed to manage audio output features like volume levels and frequency corrections on compatible devices (providing attributes and commands to adjust audio parameters such as output volume and sound frequency adjustments).

FYI, it looks like @LookAtThisTree and @ajkontra begun working on adding some scaffolding/tests for that new cluster to connectedhomeip:

* https://github.com/project-chip/connectedhomeip/pull/72637

* https://github.com/project-chip/connectedhomeip/pull/72869

* https://github.com/project-chip/connectedhomeip/pull/73594