project-chip/connectedhomeip #31436: [BUG] error: 'MATTER_CLUSTER_SCENE_NAME_SUPPORT' macro redefined

nmtoan91, 2024-01-16:
### Reproduction steps

Hi,
I'm trying to add more devices to the all-clusters-app. It ran pretty well at first, but when I tried to add some sensors I got the error:

```
clang++ -MMD -MF obj/third_party/connectedhomeip/src/app/clusters/air-quality-server/all-clusters-common.air-quality-server.cpp.o.d -Wconversion -O0 -g2 -fno-common -ffunction-sections -fdata-sections -fno-exceptions -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -Wall -Werror -Wextra -Wshadow -Wunreachable-code -Wvla -Wformat -Wformat-nonliteral -Wformat-security -Wundef -Wimplicit-fallthrough -Wheader-hygiene -Wshorten-64-to-32 -Wformat-type-confusion -Wthread-safety -Wno-deprecated-declarations -Wno-missing-field-initializers -Wno-unknown-warning-option -Wno-unused-parameter -fdiagnostics-color -fno-strict-aliasing -fsanitize=address -fno-omit-frame-pointer -fmacro-prefix-map=../../examples/all-clusters-app/linux/third_party/connectedhomeip/= -I/usr/include/gio-unix-2.0 -I/usr/include/libmount -I/usr/include/blkid -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -Wno-format-nonliteral -Wno-format-nonliteral -std=gnu++17 -fno-rtti -Wnon-virtual-dtor -DTIME_SYNC_ENABLE_TSC_FEATURE=1 -D_LIBCPP_ENABLE_THREAD_SAFETY_ANNOTATIONS=1 -DCHIP_HAVE_CONFIG_H=1 -DCHIP_ADDRESS_RESOLVE_IMPL_INCLUDE_HEADER=\<lib/address_resolve/AddressResolve_DefaultImpl.h\> -DCHIP_MINMDNS_USE_EPHEMERAL_UNICAST_PORT=1 -DCHIP_MINMDNS_HIGH_VERBOSITY=0 -DCHIP_MINMDNS_DEFAULT_POLICY=1 -I../../examples/all-clusters-app/linux/third_party/connectedhomeip/zzz_generated/all-clusters-app -Igen/third_party/connectedhomeip/examples/all-clusters-app/all-clusters-common -Igen/third_party/connectedhomeip/examples/all-clusters-app/all-clusters-common/zapgen -I../../examples/all-clusters-app/linux/third_party/connectedhomeip/src/include -I../../examples/all-clusters-app/linux/third_party/connectedhomeip/src -Igen/include -I../../examples/all-clusters-app/linux/third_party/connectedhomeip/zzz_generated/app-common -I../../examples/all-clusters-app/linux/third_party/connectedhomeip/examples/all-clusters-app/linux/include -I../../examples/all-clusters-app/linux/third_party/connectedhomeip/config/standalone -I../../examples/all-clusters-app/linux/third_party/connectedhomeip/third_party/nlassert/repo/include -I../../examples/all-clusters-app/linux/third_party/connectedhomeip/third_party/nlio/repo/include -I../../examples/all-clusters-app/linux/third_party/connectedhomeip/third_party/inipp/repo/inipp -I../../examples/all-clusters-app/linux/third_party/connectedhomeip/src/tracing/perfetto/include -I../../examples/all-clusters-app/linux/third_party/connectedhomeip/third_party/perfetto/repo/sdk -c ../../examples/all-clusters-app/linux/third_party/connectedhomeip/src/app/clusters/air-quality-server/air-quality-server.cpp -o obj/third_party/connectedhomeip/src/app/clusters/air-quality-server/all-clusters-common.air-quality-server.cpp.o
2024-01-16 14:37:39 INFO    In file included from ../../examples/all-clusters-app/linux/third_party/connectedhomeip/src/app/clusters/air-quality-server/air-quality-server.cpp:23:
2024-01-16 14:37:39 INFO    In file included from ../../examples/all-clusters-app/linux/third_party/connectedhomeip/src/app/util/attribute-storage.h:23:
2024-01-16 14:37:39 INFO    In file included from ../../examples/all-clusters-app/linux/third_party/connectedhomeip/src/app/util/af.h:23:
2024-01-16 14:37:39 INFO    In file included from ../../examples/all-clusters-app/linux/third_party/connectedhomeip/src/app/util/config.h:29:
2024-01-16 14:37:39 INFO    gen/third_party/connectedhomeip/examples/all-clusters-app/all-clusters-common/zapgen/zap-generated/gen_config.h:157:9: error: 'MATTER_CLUSTER_SCENE_NAME_SUPPORT' macro redefined [-Werror,-Wmacro-redefined]
2024-01-16 14:37:39 INFO      157 | #define MATTER_CLUSTER_SCENE_NAME_SUPPORT (0x000F & MATTER_CLUSTER_SCENE_NAME_SUPPORT_MASK)
2024-01-16 14:37:39 INFO          |         ^
2024-01-16 14:37:39 INFO    gen/third_party/connectedhomeip/examples/all-clusters-app/all-clusters-common/zapgen/zap-generated/gen_config.h:147:9: note: previous definition is here
2024-01-16 14:37:39 INFO      147 | #define MATTER_CLUSTER_SCENE_NAME_SUPPORT (0x0000 & MATTER_CLUSTER_SCENE_NAME_SUPPORT_MASK)
2024-01-16 14:37:39 INFO          |         ^
2024-01-16 14:37:39 INFO    gen/third_party/connectedhomeip/examples/all-clusters-app/all-clusters-common/zapgen/zap-generated/gen_config.h:161:9: error: 'MATTER_CLUSTER_TABLE_SIZE_SUPPORT' macro redefined [-Werror,-Wmacro-redefined]
2024-01-16 14:37:39 INFO      161 | #define MATTER_CLUSTER_TABLE_SIZE_SUPPORT (0x000F & MATTER_CLUSTER_TABLE_SIZE_SUPPORT_MASK)
2024-01-16 14:37:39 INFO          |         ^
2024-01-16 14:37:39 INFO    gen/third_party/connectedhomeip/examples/all-clusters-app/all-clusters-common/zapgen/zap-generated/gen_config.h:151:9: note: previous definition is here
2024-01-16 14:37:39 INFO      151 | #define MATTER_CLUSTER_TABLE_SIZE_SUPPORT (0x0000 & MATTER_CLUSTER_TABLE_SIZE_SUPPORT_MASK)
2024-01-16 14:37:39 INFO          |         ^
2024-01-16 14:37:39 INFO    gen/third_party/connectedhomeip/examples/all-clusters-app/all-clusters-common/zapgen/zap-generated/gen_config.h:172:9: error: 'MATTER_CLUSTER_SCENE_NAME_SUPPORT' macro redefined [-Werror,-Wmacro-redefined]
2024-01-16 14:37:39 INFO      172 | #define MATTER_CLUSTER_SCENE_NAME_SUPPORT (0x0000 & MATTER_CLUSTER_SCENE_NAME_SUPPORT_MASK)
2024-01-16 14:37:39 INFO          |         ^
2024-01-16 14:37:39 INFO    gen/third_party/connectedhomeip/examples/all-clusters-app/all-clusters-common/zapgen/zap-generated/gen_config.h:157:9: note: previous definition is here
2024-01-16 14:37:39 INFO      157 | #define MATTER_CLUSTER_SCENE_NAME_SUPPORT (0x000F & MATTER_CLUSTER_SCENE_NAME_SUPPORT_MASK)
2024-01-16 14:37:39 INFO          |         ^
2024-01-16 14:37:39 INFO    gen/third_party/connectedhomeip/examples/all-clusters-app/all-clusters-common/zapgen/zap-generated/gen_config.h:176:9: error: 'MATTER_CLUSTER_TABLE_SIZE_SUPPORT' macro redefined [-Werror,-Wmacro-redefined]
2024-01-16 14:37:39 INFO      176 | #define MATTER_CLUSTER_TABLE_SIZE_SUPPORT (0x0000 & MATTER_CLUSTER_TABLE_SIZE_SUPPORT_MASK)
2024-01-16 14:37:39 INFO          |         ^
2024-01-16 14:37:39 INFO    gen/third_party/connectedhomeip/examples/all-clusters-app/all-clusters-common/zapgen/zap-generated/gen_config.h:161:9: note: previous definition is here
2024-01-16 14:37:39 INFO      161 | #define MATTER_CLUSTER_TABLE_SIZE_SUPPORT (0x000F & MATTER_CLUSTER_TABLE_SIZE_SUPPORT_MASK)
2024-01-16 14:37:39 INFO          |         ^
2024-01-16 14:37:39 INFO    gen/third_party/connectedhomeip/examples/all-clusters-app/all-clusters-common/zapgen/zap-generated/gen_config.h:182:9: error: 'MATTER_CLUSTER_SCENE_NAME_SUPPORT' macro redefined [-Werror,-Wmacro-redefined]
2024-01-16 14:37:39 INFO      182 | #define MATTER_CLUSTER_SCENE_NAME_SUPPORT (0x000F & MATTER_CLUSTER_SCENE_NAME_SUPPORT_MASK)
2024-01-16 14:37:39 INFO          |         ^
2024-01-16 14:37:39 INFO    gen/third_party/connectedhomeip/examples/all-clusters-app/all-clusters-common/zapgen/zap-generated/gen_config.h:172:9: note: previous definition is here
2024-01-16 14:37:39 INFO      172 | #define MATTER_CLUSTER_SCENE_NAME_SUPPORT (0x0000 & MATTER_CLUSTER_SCENE_NAME_SUPPORT_MASK)
2024-01-16 14:37:39 INFO          |         ^
2024-01-16 14:37:39 INFO    gen/third_party/connectedhomeip/examples/all-clusters-app/all-clusters-common/zapgen/zap-generated/gen_config.h:186:9: error: 'MATTER_CLUSTER_TABLE_SIZE_SUPPORT' macro redefined [-Werror,-Wmacro-redefined]
2024-01-16 14:37:39 INFO      186 | #define MATTER_CLUSTER_TABLE_SIZE_SUPPORT (0x000F & MATTER_CLUSTER_TABLE_SIZE_SUPPORT_MASK)
2024-01-16 14:37:39 INFO          |         ^
2024-01-16 14:37:39 INFO    gen/third_party/connectedhomeip/examples/all-clusters-app/all-clusters-common/zapgen/zap-generated/gen_config.h:176:9: note: previous definition is here
2024-01-16 14:37:39 INFO      176 | #define MATTER_CLUSTER_TABLE_SIZE_SUPPORT (0x0000 & MATTER_CLUSTER_TABLE_SIZE_SUPPORT_MASK)
2024-01-16 14:37:39 INFO          |         ^
2024-01-16 14:37:39 INFO    6 errors generated.
2024-01-16 14:37:40 INFO    [12/106] c++ obj/third_party/connectedhomeip/src/app/util/all-clusters-common.binding-table.cpp.o
```

### Bug prevalence

all the time

### GitHub hash of the SDK that was being used

latest

### Platform

darwin

bzbarsky-apple, 2024-01-16:
@nmtoan91 This can easily happen depending on exactly what got added where and how much things got copy/pasted.  What were the specific changes made?  Please link to those.

bzbarsky-apple, 2024-01-19:
Please list step by step steps to reproduce.  Which ZAP file?  What sensors, etc.

Alternately, please attach the .zap file in question to this issue,

nmtoan91, 2024-02-06:
The error is so random. I load and save it several times and it can build.
However, why the variable ...TABLE_SIZE are annoying. Why don't we just use a map to map them into endpoints?

woody-apple, 2025-04-02:
More information is needed here about how to reproduce this, please re-open with more detail if you're still hitting this issue.

closed 2025-04-02 by woody-apple