project-chip/connectedhomeip #31463: [1.2]

solstice-gao, 2024-01-17:
### Reproduction steps

Use esp-idf to burn normally esp32-c3

### Bug prevalence

every esp32c3 board

### GitHub hash of the SDK that was being used

123

### Platform

esp32

### Platform Version(s)

1.2

### Type

Platform Issue

### Anything else?

**The following is the running log.He has been printing the following logs in the console even if there is no operation.I changed the development board and it's still the same.**

LOG：
[0;32mI (1184) chip[DL]: Configuring CHIPoBLE advertising (interval 25 ms, connectable)[0m
[0;32mI (1184) NimBLE: GAP procedure initiated: advertise; [0m
[0;32mI (1184) NimBLE: disc_mode=2[0m
[0;32mI (1184) NimBLE:  adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=40 adv_itvl_max=40[0m
[0;32mI (1194) NimBLE: 
[0m
[0;32mI (1204) chip[DL]: CHIPoBLE advertising started[0m
[0;32mI (1204) chip[DL]: Starting ESP WiFi layer[0m
I (1214) wifi:mode : sta (48:31:b7:06:47:c0)
I (1214) wifi:enable tsf
W (1214) wifi:Haven't to connect to a suitable AP now!
[0;32mI (1224) chip[DL]: Done driving station state, nothing else to do...[0m
W (1224) wifi:Haven't to connect to a suitable AP now!
[0;32mI (1234) chip[DL]: Done driving station state, nothing else to do...[0m
[0;32mI (1244) chip[SVR]: Initializing subscription resumption storage...[0m
[0;32mI (1244) chip[SVR]: Server initializing...[0m
[0;32mI (1254) chip[TS]: Last Known Good Time: 2023-10-14T01:16:48[0m
[0;32mI (1254) chip[DMG]: AccessControl: initializing[0m
[0;32mI (1264) chip[DMG]: Examples::AccessControlDelegate::Init[0m
[0;32mI (1264) chip[DMG]: AccessControl: setting[0m
[0;32mI (1274) chip[DMG]: DefaultAclStorage: initializing[0m
[0;32mI (1284) chip[DMG]: DefaultAclStorage: 0 entries loaded[0m
[0;32mI (1294) chip[ZCL]: Using ZAP configuration...[0m
[0;32mI (1294) chip[DMG]: AccessControlCluster: initializing[0m
[0;32mI (1294) chip[ZCL]: Initiating Admin Commissioning cluster.[0m
[0;32mI (1304) chip[ZCL]: Door Lock server initialized[0m
[0;32mI (1304) lock-devicecallbacks: PostAttributeChangeCallback - Cluster ID: '0x30', EndPoint ID: '0x0', Attribute ID: '0x0'[0m
[0;32mI (1314) lock-devicecallbacks: Unhandled cluster ID: 48[0m
[0;32mI (1324) lock-devicecallbacks: Current free heap: 100420
[0m
[0;32mI (1334) chip[DIS]: Updating services using commissioning mode 1[0m
[0;32mI (1334) chip[DIS]: CHIP minimal mDNS started advertising.[0m
[0;32mI (1344) chip[DIS]: Advertise commission parameter vendorID=65521 productID=32774 discriminator=3840/15 cm=1[0m
[0;32mI (1354) chip[DIS]: CHIP minimal mDNS configured as 'Commissionable node device'; instance name: 0AEAE7826CF80BE3.[0m
[0;32mI (1364) chip[DIS]: mDNS service published: _matterc._udp[0m
[0;32mI (1374) chip[IN]: CASE Server enabling CASE session setups[0m
[0;32mI (1374) chip[SVR]: Joining Multicast groups[0m
[0;32mI (1384) chip[SVR]: Server Listening...[0m
[0;32mI (1384) lock-app: ------------------------Starting App Task---------------------------[0m
[0;32mI (1394) app-devicecallbacks: Current free heap: 95992
[0m
[... 189 lines cut ...]
[0m
[0;32mI (1464) NimBLE: GAP procedure initiated: advertise; [0m
[0;32mI (1474) NimBLE: disc_mode=2[0m
[0;32mI (1474) NimBLE:  adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=40 adv_itvl_max=40[0m
[0;32mI (1484) NimBLE: 
[0m
[0;32mI (1484) chip[DL]: Device Configuration:[0m
[0;32mI (1494) chip[DL]:   Serial Number: TEST_SN[0m
[0;32mI (1494) chip[DL]:   Vendor Id: 65521 (0xFFF1)[0m
[0;32mI (1504) chip[DL]:   Product Id: 32774 (0x8006)[0m
[0;32mI (1504) chip[DL]:   Product Name: TEST_PRODUCT[0m
[0;32mI (1514) chip[DL]:   Hardware Version: 0[0m
[0;32mI (1514) chip[DL]:   Setup Pin Code (0 for UNKNOWN/ERROR): 20202021[0m
[0;32mI (1524) chip[DL]:   Setup Discriminator (0xFFFF for UNKNOWN/ERROR): 3840 (0xF00)[0m
[0;32mI (1534) chip[DL]:   Manufacturing Date: (not set)[0m
[0;32mI (1534) chip[DL]:   Device Type: 0 (0x0)[0m
[0;32mI (1544) chip[SVR]: SetupQRCode: [MT:8IXS142C00KA0648G00][0m
[0;32mI (1544) chip[SVR]: Copy/paste the below URL in a browser to see the QR Code:[0m
[0;32mI (1554) chip[SVR]: https://project-chip.github.io/connectedhomeip/qrcode.html?data=MT%3A8IXS142C00KA0648G00[0m
[0;32mI (1564) chip[SVR]: Manual pairing code: [34970112332][0m
[0;32mI (1574) lock-app: App Task started[0m
[0;32mI (1574) lock-devicecallbacks: PostAttributeChangeCallback - Cluster ID: '0x6', EndPoint ID: '0x1', Attribute ID: '0x0'[0m
[0;32mI (1584) lock-devicecallbacks: Current free heap: 95912
[0m
[0;32mI (5034) lock-app: Factory Reset Triggered. Release button within 3000ms to cancel.[0m

bzbarsky-apple, 2024-01-17:
@solstice-gao So what is the actual issue?  There is no summary for the issue and no clear description of the problem.

dhrishi, 2024-01-21:
@solstice-gao This issue should be fixed with https://github.com/project-chip/connectedhomeip/pull/31500

dhrishi, 2024-10-25:
Closing this as the fix is provided

closed 2024-10-25 by dhrishi