project-chip/connectedhomeip #31472: [CERT-TEST-FAILURE] TC-APPOBSERVER-13.2 - Test failed

Rajashreekalmane, 2024-01-17:
### Feature Area

Other

### Test Case

TC-APPOBSERVER-13.2

### Reproduction steps

Step 1 : executed the below mentioned command to put DUT into a commissionable state
`./chip-tv-casting-app`
Step 2 : Once DUT reach the commissionable state sent below mentioned command on TH.
`./chip-tool pairing onnetwork 1 20202021`

According to the spec, when sending the ContentAppMessage command, it is required to include the mandatory 'Data' field and the optional 'EncodingHint' field as parameters. However, the current chip-tool does not include the 'Data' field

**PFA Spec Screenshot :** 
[image]

**PFA chip-tool screenshot below** 
[image]

### Bug prevalence

whenever I do

### GitHub hash of the SDK that was being used

09f014cebf114980e6174277d8d45f301f77679f

### Platform

raspi

chrisdecenzo, 2024-01-17:
You should be able to provide the data field using the "--Data" command line argument:

`./out/debug/chip-tool contentappobserver content-app-message foo 1234 1 --Data bar`

chrisdecenzo, 2024-01-17:
That said I am seeing an error come back when trying this. I will fix it as part of https://github.com/project-chip/connectedhomeip/pull/31432

Rajashreekalmane, 2024-02-05:
it's resolved in latest TOT

closed 2024-02-05 by Rajashreekalmane