project-chip/connectedhomeip #31475: [TC-DGGEN-2.4] - The Python script should be updated in accordance with the latest changes in the Test Plan.

Rajashreekalmane, 2024-01-17:
### Feature Area

Other

### Test Case

TC-DGGEN-2.4

### Reproduction steps

The Python script should be updated in accordance with the latest changes in the Test Plan.

 Please refer to the Test Plan PR that has been updated for the relevant test case : https://github.com/CHIP-Specifications/chip-test-plans/pull/3824

Python SCript : https://github.com/project-chip/connectedhomeip/blob/master/src/python_testing/TC_DGGEN_2_4.py

### Bug prevalence

whenever I do

### GitHub hash of the SDK that was being used

09f014cebf114980e6174277d8d45f301f77679f

### Platform

raspi

### Anything else?

Test-Plan Reference : 
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/general_diagnostics.adoc#tc-dggen-2-4-timesnapshot-command-tests-dut_server

Test-Plan PR : https://github.com/CHIP-Specifications/chip-test-plans/pull/3824

Rajashreekalmane, 2024-01-22:
@tcarmelveilleux script missed to validate the test step 1a and 1b 
[image]
[image]

tcarmelveilleux, 2024-01-23:
If you look at the code on the whole:

```
    @async_test_body
    async def test_TC_GEN_2_4(self):
        self.print_step("1a", "Detect Time Synchronization UTCTime attribute presence")
        root_descriptor = await self.default_controller.ReadAttribute(self.dut_node_id, [(0, Clusters.Descriptor)])
        root_server_list = root_descriptor[0][Clusters.Descriptor][Clusters.Descriptor.Attributes.ServerList]
        has_timesync = (Clusters.TimeSynchronization.id in root_server_list)

        testvar_TimeSyncSupported = False

        if has_timesync:
            ts_attributes = await self.read_single_attribute(self.default_controller, node_id=self.dut_node_id, endpoint=0, attribute=Clusters.TimeSynchronization.Attributes.AttributeList)
            has_utc_time = (Clusters.TimeSynchronization.Attributes.UTCTime.attribute_id in ts_attributes)

            if has_utc_time:
                testvar_TimeSyncSupported = True
            else:
                asserts.fail("Time Synchronization cluster present but UTCTime missing. This should not happen!")

            self.print_step("1b", "Write current time to DUT")
            # Get current time in the correct format to set via command.
            th_utc = matter_epoch_us_from_utc_datetime(desired_datetime=None)

            await self.set_time_in_timesync(th_utc)

            self.print_step("1c", "Read current time from DUT")
            testvar_UTCTime1 = await self.read_timesync_attribute_expect_success(Clusters.TimeSynchronization.Attributes.UTCTime)
            asserts.assert_true(testvar_UTCTime1 != NullValue,
                                "UTCTime1 readback must not be null after SetUTCTime (per Time Synchronization cluster spec)")

            testvar_TimeSyncSupported = True
```

- Step 1a reads the flags
- Step 1b sets current epoch time if possible
- Step 1c reads backs the UTCTime

The fact that variables such as `TimeSyncSupported` have a particular way of getting set to respect the logic changes nothing.

Please indicate what specifically detracts from the test plan.

PR #31526 (2024-01-20): Fix DGGEN and TC-DGGEN-2.4 to latest spec
- https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/8737 introduced simplifications to the TimeSnapshot command. The test plans were also updated.
- This PR updates the test case to follow, by removing granularity check, and reordering some sub-steps of step3.

Fixes: #31475

Testing done:

- Tested with all-clusters-app with both `--simulate-no-internal-time` and without (this app has TimeSync, so tests step 3, not 4).
- Tested with lighting-app (this app doesn't have TimeSync, so tests step 4, not 3)
commit 8489d381 (2024-01-20): Fix DGGEN and TC-DGGEN-2.4 to latest spec (#31526)

* Fix DGGEN and TC-DGGEN-2.4 to latest spec

- https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/8737 introduced simplifications
  to the TimeSnapshot command. The test plans were also updated.
- This PR updates the test case to follow, by removing granularity check,
  and reordering some sub-steps of step3.

Fixes: #31475

Testing done:

- Tested with all-clusters-app with both `--simulate-no-internal-time` and without
  (this app has TimeSync, so tests step 3, not 4).
- Tested with lighting-app (this app doesn't have TimeSync [... cut]