project-chip/connectedhomeip #31478: [BUG] crash at chip.platform.NsdManagerServiceResolver

AriesHaw, 2024-01-17:
### Reproduction steps

Reproduction steps
android app encouters this crash sometimes:
01-16 21:06:28.611 12314 12356 I MatterClientManager: getPublicKey
01-16 21:06:28.612 12314 12356 I MatterClientManager: check public key222222[48, 89, 48, 19, 6, 7, 42, -122, 72, -50, 61, 2, 1, 6, 8, 42, -122, 72, -50, 61, 3, 1, 7, 3, 66, 0, 4, 124, 32, 64, 80, 120, 69, -89, -48, 41, -32, -88, 100, -111, -27, 46, 62, 82, 62, -100, 14, 90, -19, 69, 112, -100, -38, -124, 44, 71, -50, -117, -40, -76, 67, 101, -33, 18, -59, 56, 54, 66, 36, -12, -31, -88, -40, -35, 116, -73, -10, 61, -28, 84, 78, -128, -90, 63, -4, 115, 94, -8, 35, 70, -90]
01-16 21:06:28.612 12314 12356 I MatterClientManager: check public key333333[4, 124, 32, 64, 80, 120, 69, -89, -48, 41, -32, -88, 100, -111, -27, 46, 62, 82, 62, -100, 14, 90, -19, 69, 112, -100, -38, -124, 44, 71, -50, -117, -40, -76, 67, 101, -33, 18, -59, 56, 54, 66, 36, -12, -31, -88, -40, -35, 116, -73, -10, 61, -28, 84, 78, -128, -90, 63, -4, 115, 94, -8, 35, 70, -90]
01-16 21:06:28.613 12314 12356 E CTL     : Device Controller Factory already initialized...
01-16 21:06:28.613 12314 12356 D CTL     : Intermediate CA is not needed
01-16 21:06:28.614 12314 12356 D TOO     : KVS: Getting key f/1/r
01-16 21:06:28.615 12314 12356 D TOO     : KVS: Getting key f/1/r
01-16 21:06:28.616 12314 12356 D TOO     : KVS: Getting key f/1/n
01-16 21:06:28.616 12314 12356 D TOO     : KVS: Getting key f/1/r
01-16 21:06:28.616 12314 12356 D FP      : Validating NOC chain
01-16 21:06:28.658 12314 12796 D NsdServiceFinderAndResolver: Service discovery timed out after 30000 ms
01-16 21:06:28.659 12314 12314 D NsdManagerServiceResolver: resolve: Timing out
01-16 21:06:28.663 12314 12314 D AndroidRuntime: Shutting down VM
01-16 21:06:28.664 12314 12314 E AndroidRuntime: FATAL EXCEPTION: main
01-16 21:06:28.664 12314 12314 E AndroidRuntime: Process: com.ewelinkcastapp, PID: 12314
01-16 21:06:28.664 12314 12314 E AndroidRuntime: java.lang.RuntimeException: MulticastLock under-locked chipMulticastLock
01-16 21:06:28.664 12314 12314 E AndroidRuntime: 	at android.net.wifi.WifiManager$MulticastLock.release(WifiManager.java:3390)
01-16 21:06:28.664 12314 12314 E AndroidRuntime: 	at chip.platform.NsdManagerServiceResolver$1.run(NsdManagerServiceResolver.java:94)
01-16 21:06:28.664 12314 12314 E AndroidRuntime: 	at android.os.Handler.handleCallback(Handler.java:794)
01-16 21:06:28.664 12314 12314 E AndroidRuntime: 	at android.os.Handler.dispatchMessage(Handler.java:99)
01-16 21:06:28.664 12314 12314 E AndroidRuntime: 	at android.os.Looper.loop(Looper.java:173)
01-16 21:06:28.664 12314 12314 E AndroidRuntime: 	at android.app.ActivityThread.main(ActivityThread.java:6651)
01-16 21:06:28.664 12314 12314 E AndroidRuntime: 	at java.lang.reflect.Method.invoke(Native Method)
01-16 21:06:28.664 12314 12314 E AndroidRuntime: 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:547)
01-16 21:06:28.664 12314 12314 E AndroidRuntime: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:822)
01-16 21:06:28.664 12314 12342 I NsdServiceFinderAndResolver: Discovery stopped: _matter._tcp
01-16 21:06:28.670 12314 12314 E CrashManager: java.lang.RuntimeException: MulticastLock under-locked chipMulticastLock
01-16 21:06:28.670 12314 12314 E CrashManager: 	at android.net.wifi.WifiManager$MulticastLock.release(WifiManager.java:3390)
01-16 21:06:28.670 12314 12314 E CrashManager: 	at chip.platform.NsdManagerServiceResolver$1.run(NsdManagerServiceResolver.java:94)
01-16 21:06:28.670 12314 12314 E CrashManager: 	at android.os.Handler.handleCallback(Handler.java:794)
01-16 21:06:28.670 12314 12314 E CrashManager: 	at android.os.Handler.dispatchMessage(Handler.java:99)
01-16 21:06:28.670 12314 12314 E CrashManager: 	at android.os.Looper.loop(Looper.java:173)
01-16 21:06:28.670 12314 12314 E CrashManager: 	at android.app.ActivityThread.main(ActivityThread.java:6651)
01-16 21:06:28.670 12314 12314 E CrashManager: 	at java.lang.reflect.Method.invoke(Native Method)
01-16 21:06:28.670 12314 12314 E CrashManager: 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:547)
01-16 21:06:28.670 12314 12314 E CrashManager: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:822)

### Bug prevalence

3-7times a day

### GitHub hash of the SDK that was being used

sdk version 1.1.0.1

### Platform

android

### Platform Version(s)

Android 8.1/Android 10

### Anything else?

There's a chance that this crash will happen when you leave the APP hanging in the foreground but it's not stable and sometimes the app is hanging in the foreground for an entire day without crashing

joonhaengHeo, 2024-03-11:
@AriesHaw 
For accurate analysis, please share the full log if it is reproduced in the latest version.

yunhanw-google, 2024-03-27:
@AriesHaw Feel free to reopen it with the full logging, thanks

closed 2024-03-27 by yunhanw-google