project-chip/connectedhomeip #31495: [Feature] locate GSON libraries as a prebuilt JAR in the repo?

kencottrell, 2024-01-17:
### Feature description

...
Hi, am attempting to manually build Kotlin and Java controllers in Android studio, and am getting these unresolved references in my JsonToTLv.kt file. I assume that I need to add a JAR to my environment and was hoping to not have to pick through the github and do a Maven build since I am the incorporate Gradle within Android studio.  Is this JAR located somewhjere in the repo, or if not can it added as a feature?

I've looked around to see if I could download the jar but dont see one, 

****import com.google.gson.JsonArray
import com.google.gson.JsonElement
import com.google.gson.JsonObject
import com.google.gson.JsonParser****

### Platform

windows

### Anything else?

e: file:///C:/Users/kjcot/AndroidStudioProjects/KensMatterControllerSample/app/src/main/java/matter/jsontlv/JsonToTlv.kt:21:12 Unresolved reference: google

unresolved refs shown:  
- import com.google.gson.JsonArray
- import com.google.gson.JsonElement
- import com.google.gson.JsonObject
- import com.google.gson.JsonParser

yunhanw-google, 2024-01-18:
https://github.com/project-chip/connectedhomeip/blob/master/third_party/java_deps/set_up_java_deps.sh

kencottrell, 2024-01-18:
thank you @yunhanw-google

closed 2024-01-18 by kencottrell

kencottrell, 2024-01-18:
one more comment, @yunhanw-google  I couldn't get the refs recognized by adding them to the ./libs directory - I also had to setup the project structure |  dependencies path to point to the jars. now they are recognized. I'm not sure why Android didn't look at the libs directory after synching the Gradle build.