project-chip/connectedhomeip #31499: Nordic off-with-effect command cannot immediately transition the global-scene-control state.

chuckrui, 2024-01-18:
NCS：V2.3.0
Project：light_bulb
Test step：

1. ./chip-tool onoff on 1 1
2. ./chip-tool onoff read on-off 1 1
3. ./chip-tool onoff read global-scene-control 1 1
4. ./chip-tool onoff off-with-effect 0 0 1 1
5. ./chip-tool onoff read on-off 1 
6. ./chip-tool onoff read global-scene-control 1 1
command result：
1.success
2.true
3.true
4.false
5.true（But during about 4-5seconds will change from true to false）

And on the contrary Test step：

1. ./chip-tool onoff off 1 1
2. ./chip-tool onoff read on-off 1 1
3. ./chip-tool onoff read global-scene-control 1 1
4. ./chip-tool onoff on-with-recall-global-scene 1 
5. ./chip-tool onoff read on-off 1 1
6. ./chip-tool onoff read global-scene-control 1 1
command result:

1. success
2. false
3. false
4. success
5. true
6. true
All the command run normal.

Log：
[GlobalSceneControl.txt](https://github.com/project-chip/connectedhomeip/files/13972046/GlobalSceneControl.txt)

So I want to know that when the light changes from on state to off state through the off-with-effect command, the global-scene-control value needs to respond to the change after a few seconds.However, when the state is switched from off to on, the state change can be immediately responded to.

bzbarsky-apple, 2024-01-18:
@chuckrui Is this only an issue with Nordic devices?

@Damian-Nordic @kkasperczyk-no

bzbarsky-apple, 2024-01-18:
In particular, I cannot reproduce with the "linux" all-clusters-app: I get back "false" at step 6 in the steps above.

chuckrui, 2024-01-19:
@bzbarsky-apple right，I encountered this problem temporarily on nordic platforms

kkasperczyk-no, 2024-01-19:
@chuckrui well I couldn't reproduce your problem on the current master branch (c1c9237bccd5ff062049e913df4d6c283788b20e) + NCS 2.5.0, it works for me correctly. You wrote you tested it with NCS 2.3.0 that was released ~10 months ago and included Matter 1.0.0.2, so it's possible that something changed in the meantime. Let me try to go back to the older versions and try to reproduce it again.

kkasperczyk-no, 2024-01-19:
Ok, indeed I can see this issue on the Matter 1.0.0.2 + NCS 2.3.0 versions.

chuckrui, 2024-01-19:
@kkasperczyk-no I know that currently ncs has reached v2.5.0, but our certification is still in v2.3.0 Matter1.0, so I want to know how to solve this problem, because we found this problem when we sent the certification. And this is a feature that 0x0101 Matter Dimmable Light device type must support.

chuckrui, 2024-01-19:
We have been stuck here for a long time and have been unable to solve it. Once this problem is solved, we can pass the certification. We hope to get your reply as soon as possible.Thank you very much~

kkasperczyk-no, 2024-01-19:
@chuckrui I think I've got a fix. The problem is not with the `on-off` server, but with the `level-control`. The `on-off` server sets `global-scene-control` to false correctly, but then `level-control` puts it back to true, as `off-with-effect` triggers `moveToLevelHandler`: https://github.com/project-chip/connectedhomeip/blob/v1.0.0.2/src/app/clusters/level-control/level-control.cpp#L770 

Please try to apply this commit: https://github.com/project-chip/connectedhomeip/commit/836390ed636ca36126dbcbe763d0f127626cba8d and let me know if everything works as expected.

At least I don't see this issue anymore after I made a cherry-pick.

chuckrui, 2024-01-23:
Yes, I have verified it.Thank you very much~

closed 2024-01-23 by chuckrui