project-chip/connectedhomeip #31502: [CERT-TEST-FAILURE] [TC-MEDIAPLAYBACK-6.8] - Test-failed

Rajashreekalmane, 2024-01-18:
### Feature Area

Other

### Test Case

TC-MEDIAPLAYBACK-6.8

### Reproduction steps

Step 1 : executed the below mentioned command to put DUT into a commissionable state
./chip-tv-app
Step 2 : Once DUT reach the commissionable state sent below mentioned command on TH.
./chip-tool pairing onnetwork 1 20202021 

Step 3 : TH reads the ActiveTextTrack attribute attribute from the DUT by sending mentioned command
./chip-tool mediaplayback read active-text-track 1 3
**Expected value :**
> [1705561901.609219][4148:4150] CHIP:TOO: Endpoint: 3 Cluster: 0x0000_0506 Attribute 0x0000_0009 DataVersion: 2880188153
> [1705561901.609389][4148:4150] CHIP:TOO:   ActiveTextTrack: {
> [1705561901.609426][4148:4150] CHIP:TOO:     Id: 
> [1705561901.609459][4148:4150] CHIP:TOO:     TrackAttributes: null
> [1705561901.609490][4148:4150] CHIP:TOO:    }

**Actual value :**
> [1705561628.435827][4129:4131] CHIP:DMG: 	InteractionModelRevision = 11
> [1705561628.435882][4129:4131] CHIP:DMG: }
> [1705561628.436296][4129:4131] CHIP:TOO: Endpoint: 3 Cluster: 0x0000_0506 Attribute 0x0000_0009 DataVersion: 2880188153
> [1705561628.436859][4129:4131] CHIP:TOO:   ActiveTextTrack: {
> [1705561628.436923][4129:4131] CHIP:TOO:     Id: activeTextTrackId_0
> [1705561628.436979][4129:4131] CHIP:TOO:     TrackAttributes: {
> [1705561628.437032][4129:4131] CHIP:TOO:       LanguageCode: languageCode1
> [1705561628.437131][4129:4131] CHIP:TOO:       DisplayName: displayName1
> [1705561628.437187][4129:4131] CHIP:TOO:      }
> [1705561628.437237][4129:4131] CHIP:TOO:    }
> [1705561628.437705][4129:4131] CHIP:EM: <<< [E:54602i S:60341 M:230830833 (Ack:35510143)] (S) Msg TX to 
> 

### Bug prevalence

whenever I do

### GitHub hash of the SDK that was being used

 b044b5e486a625fb68dd684317221c65b3fa61c7

### Platform

raspi

### Anything else?

As per the test plan, step 1 expects that when TH reads the ActiveTextTrack attribute, the DUT should respond with a null value. However, in the current SDK, we are receiving a list.

Additionally, after successfully sending the DeactivateTextTrack command, when TH attempts to activate by sending the ActiveTextTrack command, it doesn't actually activate, although a success response is received

PFA log below: 
[TC-MEDIAPLAYBACK-6.8.txt](https://github.com/project-chip/connectedhomeip/files/13973337/TC-MEDIAPLAYBACK-6.8.txt)

**Test-Plan Reference :** 
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/media.adoc#tc-mediaplayback-6-8-text-track-verification

chrisdecenzo, 2024-01-20:
should be fixed now

Rajashreekalmane, 2024-01-22:
resolved in latest TOT.

closed 2024-01-22 by Rajashreekalmane