project-chip/connectedhomeip #31503: [CERT-TEST-FAILURE] TC-MEDIAPLAYBACK-6.9 - Test-Failed

Rajashreekalmane, 2024-01-18:
### Feature Area

Other

### Test Case

TC-MEDIAPLAYBACK-6.9

### Reproduction steps

Step 1 : executed the below mentioned command to put DUT into a commissionable state
./chip-tv-app
Step 2 : Once DUT reach the commissionable state sent below mentioned command on TH.
./chip-tool pairing onnetwork 1 20202021 

Step 3 : TH reads the  attribute from the DUT by sending mentioned command
`./chip-tool mediaplayback activate-text-track activeAudioTrackId_1 1 3`
> [1705562844.672707][4187:4189] CHIP:DMG: Received Command Response Status for Endpoint=3 Cluster=0x0000_0506 Command=0x0000_000D Status=0x0
> [1705562844.672774][4187:4189] CHIP:DMG: ICR moving to [AwaitingDe]
> [1705562844.673095][4187:4189] CHIP:EM: <<< [E:59477i S:27452 M:232671343 (Ack:53715641)] (S) Msg TX to 1:0000000000000001 [5564] [UDP:[fe80::e65f:1ff:fe0f:19ff%wlan0]:5640] --- Type 0000:10 (SecureChannel:StandaloneAck)

`./chip-tool mediaplayback  read active-audio-track 1 3`
**Expected value :**
> [1705562877.751018][4190:4192] CHIP:TOO: Endpoint: 3 Cluster: 0x0000_0506 Attribute 0x0000_0007 DataVersion: 2880188153
> [1705562877.751204][4190:4192] CHIP:TOO:   ActiveAudioTrack: {
> [1705562877.751235][4190:4192] CHIP:TOO:     Id: activeAudioTrackId_1
> [1705562877.751264][4190:4192] CHIP:TOO:     TrackAttributes: {
> [1705562877.751291][4190:4192] CHIP:TOO:       LanguageCode: languageCode1
> [1705562877.751341][4190:4192] CHIP:TOO:       DisplayName: displayName1
> [1705562877.751369][4190:4192] CHIP:TOO:      }
> [1705562877.751395][4190:4192] CHIP:TOO:    }
> [1705562877.751672][4190:4192] CHIP:EM: <<< [E:6492i S:10909 M:114938002 (Ack:78123649)] (S) Msg TX to 1:0000000000000001 [5564] [UDP:[fe80::e65f:1ff:fe0f:19ff%wlan0]:5640] --- Type 0000:10 (SecureChannel:StandaloneAck)

**Actual value :**
> [1705562877.751018][4190:4192] CHIP:TOO: Endpoint: 3 Cluster: 0x0000_0506 Attribute 0x0000_0007 DataVersion: 2880188153
> [1705562877.751204][4190:4192] CHIP:TOO:   ActiveAudioTrack: {
> [1705562877.751235][4190:4192] CHIP:TOO:     Id: activeAudioTrackId_0
> [1705562877.751264][4190:4192] CHIP:TOO:     TrackAttributes: {
> [1705562877.751291][4190:4192] CHIP:TOO:       LanguageCode: languageCode1
> [1705562877.751341][4190:4192] CHIP:TOO:       DisplayName: displayName1
> [1705562877.751369][4190:4192] CHIP:TOO:      }
> [1705562877.751395][4190:4192] CHIP:TOO:    }
> [1705562877.751672][4190:4192] CHIP:EM: <<< [E:6492i S:10909 M:114938002 (Ack:78123649)] (S) Msg TX to 1:0000000000000001 [5564] [UDP:[fe80::e65f:1ff:fe0f:19ff%wlan0]:5640] --- Type 0000:10 (SecureChannel:StandaloneAck)

### Bug prevalence

whenevre I do

### GitHub hash of the SDK that was being used

 b044b5e486a625fb68dd684317221c65b3fa61c7

### Platform

raspi

### Anything else?

As per Test-Plan test step 3, after successfully sending the ActivateAudioTrack command, if TH tries to read the ActiveAudioTrack attribute, it should list the value sent in step 3. However, it's currently listing the default value, i.e., Id: activeAudioTrackId_0

**Test-Plan Reference :**
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/media.adoc#tc-mediaplayback-6-9-audio-track-verification

**PFA log below :** 
[TC-MEDIAPLAYBACK-6.9.txt](https://github.com/project-chip/connectedhomeip/files/13973510/TC-MEDIAPLAYBACK-6.9.txt)

chrisdecenzo, 2024-01-20:
should be fixed now

Rajashreekalmane, 2024-01-22:
resolved in latest TOT.

closed 2024-01-22 by Rajashreekalmane