project-chip/connectedhomeip #31504: [Feature] [chip-tool] i would like to know node id list by chip-tool

mmt-10, 2024-01-18:
### Feature description

Hi,
I would like to confirm node ids on fabric.
does chip-tool have this option?
If No, please add this feature.
If Yes, please add explanation to [https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/chip_tool_guide.md](https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/chip_tool_guide.md).

### Platform

all

bzbarsky-apple, 2024-01-18:
> I would like to confirm node ids on fabric.

What do you mean by "node ids on fabric"?  Do you mean "currently discoverable devices", or "all things that have ever been commissioned on the fabric, even if they are not on the networks right now?

The latter is not really doable.  The former could be done in theory, but there is no support for operational browse in the core Matter mdns libraries right now, so that would need to be added first.  See also https://github.com/project-chip/connectedhomeip/issues/25091

mmt-10, 2024-01-18:
Thank you for your reply.

I made the fabric by chip-tool.
I would like to dump the node ids which chip-tool commissioned in the fabric.
As my understanding, I need to remember the node ids.
If I forget the node ids, chip-tool showed the commissioned node ids.