project-chip/connectedhomeip #31511: [1.3] Scenes Management Cluster Implantation has to updated as per Spec Changes.

Ashwinigrl, 2024-01-18:
### Reproduction steps

1. Below mentioned Attributes Id's has to be updated as per Spec.

- [ ] LastConfiguredBy(0x0000)
- [ ] SceneTableSize(0x0001)
- [ ] FabricSceneInfo(0x0002)
2. `EX`, `TS` and  `FS` features has been removed from Spec, So Feature map value has to be updated as per Spec.
3. Some  Attributes(NameSupport) are removed and Id's has been changed in Spec,  So Attribute Id's has to be updated as per Spec in Attribute List.
4. `EnhancedAddScene` and `EnhancedViewScene` commands has been removed from Spec, So It should be removed from AcceptedCommandList.
5. `EnhancedAddSceneResponse` and` EnhancedViewSceneResponse` commands has been removed from Spec, So It should be removed from GeneratedCommandList.
6. Below mentioned commands has to be removed from SDK as per Spec.

- [ ] EnhancedAddScene
- [ ] EnhancedViewScene
- [ ] EnhancedAddSceneResponse
- [ ] EnhancedViewSceneResponse

### Bug prevalence

Whenever I do

### GitHub hash of the SDK that was being used

09f014cebf114980e6174277d8d45f301f77679f

### Platform

raspi

### Platform Version(s)

1.3

### Type

Test Improvement

### Anything else?

**Spec References:**
- https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/app_clusters/Scenes.adoc#4-features

**Test Plan References:**
- https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/scenes.adoc#411-tc-s-11-global-attributes-with-dut_server

lpbeliveau-silabs, 2024-01-18:
A PR is opened in the SDK for those changes here https://github.com/project-chip/connectedhomeip/pull/31178

lpbeliveau-silabs, 2024-01-18:
On the spec side, those changes were merged here: https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/8668

Ashwinigrl, 2024-02-02:
Issue Fixed

closed 2024-02-02 by Ashwinigrl