project-chip/connectedhomeip #31512: [Tooling] ZAP bombs out with unhelpful message if derived xml clusters (e.g. xyz-mode-cluster.xml is not in mode-base-cluster.xml)

jamesharrow, 2024-01-18:
### Reproduction steps / Feature

Follow the process to add new XML cluster definitions, but due to lack of documentation there was no clear instruction to add the <cluster code="0x----"> into mode-base-cluster.xml

Running  ./scripts/tools/zap_regen_all.py bombs out after 5 mins with this error:

Having more debug to explain what cluster is affected and what type is effected would give the developer a clue where the issue was. Instead I had to reverse engineer zap tool.

🤖 Generating files:
    ✍  /__w/connectedhomeip/connectedhomeip/src/controller/python/chip/clusters/Objects.py
⚠️  Errors:
    👎  /__w/connectedhomeip/connectedhomeip/src/controller/python/chip/clusters/CHIPClusters.py: ⛔ TypeError: Cannot read properties of undefined (reading 'discriminatorName')
Stack trace:

TypeError: Cannot read properties of undefined (reading 'discriminatorName')
    at Object.as_underlying_python_zcl_type (/snapshot/zap/dist/src-electron/generator/matter/controller/python/templates/helper.js:76:23)
    at async Promise.all (index 0)
    at async Promise.all (index 0)
    at async Promise.all (index 1)
    at async Promise.all (index 70)
    at async Promise.all (index 0)
    at produceContent (/snapshot/zap/dist/src-electron/generator/template-engine.js:150:19)
    at generateSingleTemplate (/snapshot/zap/dist/src-electron/generator/generation-engine.js:522:27)
    at async Promise.all (index 1)
    at generateAllTemplates (/snapshot/zap/dist/src-electron/generator/generation-engine.js:499:5)
🕐 Generation time: 6s 905ms 
{"src/controller/python/chip/clusters/CHIPClusters.py":{}}
Zap generation error: Error: Generation failed: -- blank session --
Error: Generation failed: -- blank session --
    at generateSingleFile (/snapshot/zap/dist/src-electron/main-process/startup.js:490:15)
    at startGeneration (/snapshot/zap/dist/src-electron/main-process/startup.js:543:5)
    
    
Propose adding something like:

```
   if (!dataType)
   {
     throw new Error(`dataType is undefined for clusterId: ${clusterId} ${type} ${packageIds}`);
   }
```
into https://github.com/project-chip/zap/blob/9c4b5244168a512112d98c5807285a7dbc8b5f3d/src-electron/generator/matter/controller/python/templates/helper.js#L82
  
    

### Platform

core (please add to version below)

### Platform Version(s)

1.3

### Type

Manually tested with SDK

bzbarsky-apple, 2024-01-18:
@jamesharrow ZAP issues should be filed in https://github.com/project-chip/zap/issues

jamesharrow, 2024-01-18:
Raised issue on ZAP instead here: https://github.com/project-chip/zap/issues/1247

closed 2024-01-18 by jamesharrow