project-chip/connectedhomeip #31539: [1.3] [TC-RVCCLEANM-2.2] Test failing due to Status code is mismatching

Ashwinigrl, 2024-01-19:
### Reproduction steps

Executed the below mentioned command to put DUT into a commissionable state
**./chip-all-clusters-app**

Once DUT reach the commissionable state sent below mentioned command on TH.
**./chip-tool pairing onnetwork 1 20202021**

**After the successful commissioning Send the below commands**

1. Read SupportedModes attribute of the RVC Run Mode cluster.

./chip-tool rvcrunmode read supported-modes 1 1

```
[1690177742.429087][6315:6317] CHIP:TOO: Endpoint: 1 Cluster: 0x0000_0054 Attribute 0x0000_0000 DataVersion: 551764487
[1690177742.429341][6315:6317] CHIP:TOO:   SupportedModes: 3 entries
[1690177742.429483][6315:6317] CHIP:TOO:     [1]: {
[1690177742.429540][6315:6317] CHIP:TOO:       Label: Idle
[1690177742.429595][6315:6317] CHIP:TOO:       Mode: 0
[1690177742.429658][6315:6317] CHIP:TOO:       ModeTags: 1 entries
[1690177742.429732][6315:6317] CHIP:TOO:         [1]: {
[1690177742.429791][6315:6317] CHIP:TOO:           Value: 16384
[1690
177742.429842][6315:6317] CHIP:TOO:          }
[1690177742.429892][6315:6317] CHIP:TOO:      }
[1690177742.429967][6315:6317] CHIP:TOO:     [2]: {
[1690177742.430020][6315:6317] CHIP:TOO:       Label: Cleaning
[1690177742.430070][6315:6317] CHIP:TOO:       Mode: 1
[1690177742.430188][6315:6317] CHIP:TOO:       ModeTags: 1 entries
[1690177742.430263][6315:6317] CHIP:TOO:         [1]: {
[1690177742.430320][6315:6317] CHIP:TOO:           Value: 16385
[1690177742.430374][6315:6317] CHIP:TOO:          }
[1690177742.430427][6315:6317] CHIP:TOO:      }
[1690177742.430501][6315:6317] CHIP:TOO:     [3]: {
[1690177742.430555][6315:6317] CHIP:TOO:       Label: Mapping
[1690177742.430605][6315:6317] CHIP:TOO:       Mode: 2
[1690177742.430664][6315:6317] CHIP:TOO:       ModeTags: 1 entries
[1690177742.430729][6315:6317] CHIP:TOO:         [1]: {
[1690177742.430784][6315:6317] CHIP:TOO:           Value: 16384
[1690177742.430901][6315:6317] CHIP:TOO:          }
[1690177742.430966][6315:6317] CHIP:TOO:      }
```

3. Read the CurrentMode attribute from the RVC Run Mode cluster.

./chip-tool rvcrunmode read current-mode 1 1

```
[1690203158.762178][18282:18284] CHIP:TOO: Endpoint: 1 Cluster: 0x0000_0054 Attribute 0x0000_0001 DataVersion: 468915105
[1690203158.762194][18282:18284] CHIP:TOO:   CurrentMode: 0
```
4. If CurrentMode is in Idle state, change the mode to non-idle mode(Cleaning)

./chip-tool rvcrunmode change-to-mode 1 1 1

```
[1690178355.021382][6358:6360] CHIP:TOO: Endpoint: 1 Cluster: 0x0000_0054 Command 0x0000_0001
[1690178355.021533][6358:6360] CHIP:TOO:   ChangeToModeResponse: {
[1690178355.021692][6358:6360] CHIP:TOO:     status: 0
[1690178355.021750][6358:6360] CHIP:TOO:    }
```
5. Read SupportedModes attribute of the RVC Clean Mode cluster.

./chip-tool  rvccleanmode read supported-modes 1 1
```
[1690182376.583260][14996:14998] CHIP:TOO: Endpoint: 1 Cluster: 0x0000_0055 Attribute 0x0000_0000 DataVersion: 3145953005
[1690182376.583288][14996:14998] CHIP:TOO:   SupportedModes: 3 entries
[1690182376.583301][14996:14998] CHIP:TOO:     [1]: {
[1690182376.583304][14996:14998] CHIP:TOO:       Label: Vacuum
[1690182376.583307][14996:14998] CHIP:TOO:       Mode: 0
[1690182376.583311][14996:14998] CHIP:TOO:       ModeTags: 1 entries
[1690182376.583316][14996:14998] CHIP:TOO:         [1]: {
[1690182376.583319][14996:14998] CHIP:TOO:           Value: 16385
[1690182376.583322][14996:14998] CHIP:TOO:          }
[1690182376.583325][14996:14998] CHIP:TOO:      }
[1690182376.583331][14996:14998] CHIP:TOO:     [2]: {
[1690182376.583333][14996:14998] CHIP:TOO:       Label: Wash
[1690182376.583336][14996:14998] CHIP:TOO:       Mode: 1
[1690182376.583340][14996:14998] CHIP:TOO:       ModeTags: 1 entries
[1690182376.583344][14996:14998] CHIP:TOO:         [1]: {
[1690182376.583347][14996:14998] CHIP:TOO:           Value: 16386
[1690182376.583349][14996:14998] CHIP:TOO:          }
[1690182376.583352][14996:14998] CHIP:TOO:      }
[1690182376.583357][14996:14998] CHIP:TOO:     [3]: {
[1690182376.583360][14996:14998] CHIP:TOO:       Label: Deep clean
[1690182376.583362][14996:14998] CHIP:TOO:       Mode: 2
[1690182376.583366][14996:14998] CHIP:TOO:       ModeTags: 2 entries
[1690182376.583369][14996:14998] CHIP:TOO:         [1]: {
[1690182376.583372][14996:14998] CHIP:TOO:           Value: 7
[1690182376.583375][14996:14998] CHIP:TOO:          }
[1690182376.583378][14996:14998] CHIP:TOO:         [2]: {
[1690182376.583380][14996:14998] CHIP:TOO:           Value: 16384
[1690182376.583383][14996:14998] CHIP:TOO:          }
[1690182376.583385][14996:14998] CHIP:TOO:      }
```
6. Then try to change the mode to any mode listed in the SupportedModes attribute of the RVC Clean Mode cluster.

**Actual Outcome:**

./chip-tool  rvccleanmode  change-to-mode 2 1 1

```
[1705656302.541257][334243:334245] CHIP:DMG: Received Command Response Data, Endpoint=1 Cluster=0x0000_0055 Command=0x0000_0001
[1705656302.541266][334243:334245] CHIP:TOO: Endpoint: 1 Cluster: 0x0000_0055 Command 0x0000_0001
[1705656302.541290][334243:334245] CHIP:TOO:   ChangeToModeResponse: {
[1705656302.541298][334243:334245] CHIP:TOO:     status: 64
[1705656302.541301][334243:334245] CHIP:TOO:     statusText: Cannot change the cleaning mode during a clean
[1705656302.541304][334243:334245] CHIP:TOO:    }
```
**Expected Outcome:**

./chip-tool  rvccleanmode  change-to-mode 2 1 1

```
[1705656302.541257][334243:334245] CHIP:DMG: Received Command Response Data, Endpoint=1 Cluster=0x0000_0055 Command=0x0000_0001
[1705656302.541266][334243:334245] CHIP:TOO: Endpoint: 1 Cluster: 0x0000_0055 Command 0x0000_0001
[1705656302.541290][334243:334245] CHIP:TOO:   ChangeToModeResponse: {
[1705656302.541298][334243:334245] CHIP:TOO:     status: 3
[1705656302.541301][334243:334245] CHIP:TOO:     statusText: Cannot change the cleaning mode during a clean
[1705656302.541304][334243:334245] CHIP:TOO:    }

### Bug prevalence

Whenever I do

### GitHub hash of the SDK that was being used

b044b5e486a625fb68dd684317221c65b3fa61c7

### Platform

raspi

### Platform Version(s)

1.3

### Type

Test Improvement

### Anything else?

As per Spec, When we try to change the mode for RVCCLEAN Mode while the RVC Run Mode cluster’s CurrentMode attribute is set to a mode without the Idle mode tag. The ChangeToModeResponse command SHALL have the StatusCode field set to the `InvalidInMode value `. But currently getting response as StatusCode field is Set to `64(0x40)`

**Specification References:**
- https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/app_clusters/Mode_RVCClean.adoc#8-mode-change-restrictions

![image](https://github.com/project-chip/connectedhomeip/assets/98016634/93f1edfd-3a67-4a77-b63b-23ad6d4d89d2)

**Test Plan References:**
- https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/mode_rvc_clean.adoc#331-tc-rvccleanm-22-change-to-mode-restrictions-with-dut_server
 
**Reference Log:**
[TC-RVCCLEANM-2.2.txt](https://github.com/project-chip/connectedhomeip/files/13987131/TC-RVCCLEANM-2.2.txt)

hicklin, 2024-01-19:
Please note that there has been a [change to the RVC Clean Mode cluster spec](https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/6cb817ca4ea5efcafd7f04cb741432cc556a70c1/src/app_clusters/Mode_RVCClean.adoc?plain=1#L166) that introduce this new error. The test scripts are currently being updated. This update is being tracked by [this issue](https://github.com/project-chip/connectedhomeip/issues/31394).

Ashwinigrl, 2024-01-29:
Issue Fixed in commit f63b505179fd326c2cb5df91eed330ecd71a39b8. Hence closing this Issue.