project-chip/connectedhomeip #31545: [CERT-TEST-FAILURE] - TC-ACE-2.1 - Test failed

KishokG, 2024-01-19:
### Feature Area

Other

### Test Case

TC-ACE-2.1

### Reproduction steps

Read privilege enforcement fails using all-clusters-app.

### Bug prevalence

Whenever I do this 

### GitHub hash of the SDK that was being used

039cbab7aeb6387d56b6196249b45ea71350752a

### Platform

raspi

### Anything else?

Error Log : [TC-ACE-2.1 Failure responses](https://github.com/project-chip/connectedhomeip/files/13989161/accesschecker2.txt)

KishokG, 2024-01-22:
@cecille Write privilege test case is PASSED when I ran the script in this commit 039cbab7aeb6387d56b6196249b45ea71350752a. But I tried with latest commit 0ff709f25f6a7af68314c1b0a147a76c30c234f6 and now both read and write privileges gets fail. So I have added `TC-ACE-2.2` test case in this ticket and updated the log as well.

cecille, 2024-02-01:
Discussed on slack, but didn't update issue. This appears to be a timeout. https://github.com/project-chip/connectedhomeip/pull/31643

Survensa, 2024-02-06:
Issue fixed

closed 2024-02-06 by Survensa