project-chip/connectedhomeip #31551: [1.3]   `FeatureMap list` and `AttributeList` has to be updated as per Test Plan for Mode base Clusters

Ashwinigrl, 2024-01-19:
### Reproduction steps

1. In Test Plan `DEPONOFF` feature has been removed from the `FeatureMap list`, So In SDK `FeatureMap list` has to be updated for below mentioned mode base clusters as per Test Plan.

- [ ] Refrigerator And Temperature Controlled Cabinet Mode
- [ ] RVC Run Mode
- [ ] RVC Clean Mode
- [ ] Dishwasher Mode Cluster
- [ ]  Laundry Washer Mode

**Test Plan References:**
**Took Example as  Laundry Washer Mode cluster**
- https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/mode_laundry_washer.adoc
[image]

**Log Reference:**
./chip-tool laundrywashermode read feature-map  1 1

```
[1705673126.534207][5460:5462] CHIP:DMG: 	],
[1705673126.534224][5460:5462] CHIP:DMG: 	
[1705673126.534234][5460:5462] CHIP:DMG: 	SuppressResponse = true, 
[1705673126.534244][5460:5462] CHIP:DMG: 	InteractionModelRevision = 11
[1705673126.534253][5460:5462] CHIP:DMG: }
[1705673126.535045][5460:5462] CHIP:TOO: Endpoint: 1 Cluster: 0x0000_0051 Attribute 0x0000_FFFC DataVersion: 886603343
[1705673126.535391][5460:5462] CHIP:TOO:   FeatureMap: 1

```

2. In Test Plan `OnMode` attribute has been removed from `AttributeList`, So In SDK `AttributeList`  has to be updated for below mentioned mode base clusters as per Test Plan.

- [ ] Refrigerator And Temperature Controlled Cabinet Mode
- [ ] RVC Run Mode
- [ ] RVC Clean Mode
- [ ] Dishwasher Mode Cluster
- [ ]  Laundry Washer Mode

**Test Plan References:**
**Took Example as  Laundry Washer Mode cluster**
- https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/mode_laundry_washer.adoc
[image]

**Log Reference:**
./chip-tool laundrywashermode read attribute-list 1 1

```
[1705673395.717456][5519:5521] CHIP:DMG: }
[1705673395.717534][5519:5521] CHIP:TOO: Endpoint: 1 Cluster: 0x0000_0051 Attribute 0x0000_FFFB DataVersion: 886603343
[1705673395.717812][5519:5521] CHIP:TOO:   AttributeList: 9 entries
[1705673395.717824][5519:5521] CHIP:TOO:     [1]: 0
[1705673395.717828][5519:5521] CHIP:TOO:     [2]: 1
[1705673395.717833][5519:5521] CHIP:TOO:     [3]: 2
[1705673395.717838][5519:5521] CHIP:TOO:     [4]: 3
[1705673395.717842][5519:5521] CHIP:TOO:     [5]: 65528
[1705673395.717847][5519:5521] CHIP:TOO:     [6]: 65529
[1705673395.717851][5519:5521] CHIP:TOO:     [7]: 65531
[1705673395.717855][5519:5521] CHIP:TOO:     [8]: 65532
[1705673395.717859][5519:5521] CHIP:TOO:     [9]: 65533
```

### Bug prevalence

Whenever I do

### GitHub hash of the SDK that was being used

b044b5e486a625fb68dd684317221c65b3fa61c7

### Platform

raspi

### Platform Version(s)

1.3

### Type

Test Improvement

Ashwinigrl, 2024-01-30:
FeatureMap value has been updated for `RVC Run Mode` and `RVC Clean Mode` clusters in commit f63b505179fd326c2cb5df91eed330ecd71a39b8

cecille, 2024-04-12:
We should close this as "do not fix". 1.3 is out the door already, all of these tests are being removed in 1.4 - https://github.com/CHIP-Specifications/chip-test-plans/pull/4104

Ashwinigrl, 2024-05-23:
1.1 Test case has been removed from test-plan, hence closing this issue

closed 2024-05-23 by Ashwinigrl