project-chip/connectedhomeip #31554: [Build] Substitute SDK Docker Image

jscarey1, 2024-01-19:
### Build issue(s)

1. Attempting to build a Docker Image for Test Harness (not on RPI).
2. Using SDK SHA `85a48aabbc4e43c433aa4b2d35b2817d611946de`
3. Fails to Build due to ImportError: cannot import name 'OneStyleAndTextTuple' from 'prompt_toolkit.formatted_text' (/root/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/prompt_toolkit/formatted_text/__init__.py)
...

### Platform

python

### Anything else?

[docker_fail.txt](https://github.com/project-chip/connectedhomeip/files/13990615/docker_fail.txt)

bzbarsky-apple, 2024-01-19:
@andy31415

andy31415, 2024-01-19:
You will need a new SDK SHA to have the fix or cherry-pick the update to prompt-toolkit.

closed 2024-01-19 by andy31415