project-chip/connectedhomeip #31574: [Tooling] Adjust run_test_suite.py to no t use SIGKILL on the test device

Apollon77, 2024-01-20:
### Reproduction steps / Feature

I'm currently changing the matter.js testing to use the new tooling from here and I run into the issue that it seems that the started Device executable is ended by SIGKILL or such "hard way" when it gets restarted. 
I use a nodejs script and use this e.g. as --all-clusters-app parameter, but SIGKILL do not give the process time to really tear down anything. e.g. sending MDNS expires and such and this leads to issues with the tests.

It would be very helpful of the kill signal could be sonfigurable - or just SIGTERM or such is used.

Thank you

### Platform

core (please add to version below)

### Platform Version(s)

NA

### Type

YAML tested

bzbarsky-apple, 2024-01-22:
The Python used is `process.kill()`.  We could `process.terminate()` but there is no guarantee that this will actually kill the process, and some of the bits here absolutely need to make sure the process dies (e.g. so it releases the port).

If you have specific suggestions on how to improve this while preserving the existing invariants, a PR is probably best....

Apollon77, 2024-01-22:
I'm unfortunately not a python developer. I created the issue on request of @cecille. 

I just read that `process.kill("SIGINT")` could be used to send a SIGINT, but yes that would require maybe to check delayed afterwards if the process was really stopped and maybe send a kill 1-2s afterwards again or such ...

Apollon77, 2024-01-24:
@bzbarsky-apple @cecille Ok, I tried my best and came up with that after digging a bit into python ... Pleas ebe nice to my ... my first python changes ;-))

https://github.com/project-chip/connectedhomeip/pull/31650

I tested it locally and it works from my perspective

PR #31650 (2024-01-24): Use SIGTERM to end test process for yaml tests
fixes #31574

I tested it with chip all clusters-app and also matter.js and it works, also the case where it falls back to killing if process did not terminates after 10s

I also verified that this solves the testing error I have with the TestDiscovery test which was caused by missing MDNS expires
closed 2024-01-24 by mergify[bot]