project-chip/connectedhomeip #31577: [1.3] Kotlin Matter controller  Main.kt type mismatch between getPairingCommands() and PairOnNetworkLongCommand()

kencottrell, 2024-01-21:
### Reproduction steps

android build routine that includes these imports and tries to instance the getPairingCommands()function
import com.matter.controller.commands.common.Command
import com.matter.controller.commands.common.CommandManager
import com.matter.controller.commands.common.CredentialsIssuer
import com.matter.controller.commands.pairing.PairOnNetworkLongCommand
import com.matter.controller.commands.pairing.PairOnNetworkLongImInvokeCommand
import com.matter.controller.commands.pairing.PairOnNetworkLongImReadCommand
import com.matter.controller.commands.pairing.PairOnNetworkLongImSubscribeCommand
import com.matter.controller.commands.pairing.PairOnNetworkLongImWriteCommand
import matter.controller.ControllerParams
import matter.controller.MatterController
import matter.controller.MatterControllerImpl
....
// sends Matter Controller, but PairOnNetworkLongCommands is expecting DeviceController
private fun getPairingCommands(
    controller: MatterController,
    credentialsIssuer: CredentialsIssuer
): List<Command> {
    return listOf(
        PairOnNetworkLongCommand(controller, credentialsIssuer),
    )
}

### Bug prevalence

everytime I run the main file

### GitHub hash of the SDK that was being used

not sure where to find this?

### Platform

windows

### Type

Common Cluster Logic

### Anything else?

: file:///C:/Users/kjcot/AndroidStudioProjects/Jan19MainLoop/app/src/main/java/com/matter/controller/commands/pairing/PairOnNetworkLongCommand.kt:31:5 Type mismatch: inferred type is MatterController but ChipDeviceController was expected

yunhanw-google, 2024-01-22:
@kencottrell Kotlin Matter controller is a little far away from the readiness, which still need some time development on it, please still use java device controller https://github.com/project-chip/connectedhomeip/tree/master/src/controller/java/src/chip/devicecontroller, and refer the android-chip-tool for usage, thanks

kencottrell, 2024-01-24:
@yunhanw-google  I'll close this if ok with you.

closed 2024-01-24 by kencottrell