project-chip/connectedhomeip #31578: [CERT-TEST-FAILURE] - TC-DEM-1.1 - Test failed

KishokG, 2024-01-22:
### Feature Area

Other

### Test Case

TC-DEM-1.1

### Reproduction steps

1. Bring up the DUT with All-clusters-app.
2. Pair the DUT using chip-tool through "Onnetwork" transport.
3. Run the command `./chip-tool deviceenergymanagement read feature-map 1 1`

**Expected Outcome:**

> [1705565302.904580][7054:7056] CHIP:TOO: Endpoint: 1 Cluster: 0x0000_0098 Attribute 0x0000_FFFC DataVersion: 1117764527
> [1705565302.904631][7054:7056] CHIP:TOO:   FeatureMap: 31

**Actual Outcome:**

> [1705565302.904580][7054:7056] CHIP:TOO: Endpoint: 1 Cluster: 0x0000_0098 Attribute 0x0000_FFFC DataVersion: 1117764527
> [1705565302.904631][7054:7056] CHIP:TOO:   FeatureMap: 127

### Bug prevalence

Whenever I try to read FeatureMap.

### GitHub hash of the SDK that was being used

0ff709f25f6a7af68314c1b0a147a76c30c234f6

### Platform

raspi

### Anything else?

In SPEC, the `ForecastAdjustment(FA)` and `ConstraintBasedAdjustment(CON)` feature is `PROVISIONAL` but in ACA these features are enabled. So FA and CON features has to be disabled in ACA as it is provisional.

**Specification Reference:**
[image]

jamesharrow, 2024-01-24:
I think this Issue may be raised incorrectly since in order to support PA,STA, PAU features you need to support SFR or PFR as well.
If you support PFR you may optionally also support SFR, and vice versa (if you support SFR you may optionally also support PFR).

So at the very least the expected value should allow 0x1F (both SFR & PFR), 0x1b (PFR is on), or 0x17 (SFR is on).

The other question is just because a feature is marked as provisional in the AppClusters spec, should we be stopping these features being enabled and failing them? @cecille raised this question yesterday.
The current test plan says that they shall not be set. But maybe the test plan needs changing here?

jamesharrow, 2024-01-24:
NOTE: 
The test spec PR https://github.com/CHIP-Specifications/chip-test-plans/pull/3866 should have been used for the testing.

jamesharrow, 2024-01-24:
Based on discussion with Cecille, and clarification that '31' is a decimal value = 0x1F.
I will update the EnergyManagementApp to remove bits 5 & 6

PR #31663 (2024-01-25): Fixes #31578 Removed provisional features from DEM to align to 1.3 TE2 test plans
Fixes #31578 

Removed provisional features from:
- all-clusters-app/device-energy-management-stub.cpp
- energy-management-app/linux/main.cpp
- energy-management-app/esp32/main.cpp
commit ce800c26 (2024-01-25): Fixes #31578 Removed provisional features from implementation to align with test plan for TE2. (#31663)
closed 2024-01-25 by mergify[bot]