project-chip/connectedhomeip #31594: [CERT-TEST-FAILURE] TC-MEDIAPLAYBACK-6.9 - Test-failed

Rajashreekalmane, 2024-01-22:
### Feature Area

Other

### Test Case

TC-MEDIAPLAYBACK-6.9

### Reproduction steps

Step 1 : executed the below mentioned command to put DUT into a commissionable state
`./chip-tv-app`
Step 2 : Once DUT reach the commissionable state sent below mentioned command on TH.
`./chip-tool pairing onnetwork 1 20202021`

Step 3 : TH sends a ActivateAudioTrack command to DUT specifying the ID value from step 2
`./chip-tool mediaplayback activate-text-track activeAudioTrackId_1 1 3`
**Verify response is a successful (value 0) status response**

> [1705921025.735102][14382:14385] CHIP:DMG: InvokeResponseMessage =
> [1705921025.735139][14382:14385] CHIP:DMG: {
> [1705921025.735175][14382:14385] CHIP:DMG: 	suppressResponse = false, 
> [1705921025.735213][14382:14385] CHIP:DMG: 	InvokeResponseIBs =
> [1705921025.735262][14382:14385] CHIP:DMG: 	[
> [1705921025.735301][14382:14385] CHIP:DMG: 		InvokeResponseIB =
> [1705921025.735353][14382:14385] CHIP:DMG: 		{
> [1705921025.735395][14382:14385] CHIP:DMG: 			CommandStatusIB =
> [1705921025.735447][14382:14385] CHIP:DMG: 			{
> [1705921025.735494][14382:14385] CHIP:DMG: 				CommandPathIB =
> [1705921025.735542][14382:14385] CHIP:DMG: 				{
> [1705921025.735646][14382:14385] CHIP:DMG: 					EndpointId = 0x3,
> [1705921025.735700][14382:14385] CHIP:DMG: 					ClusterId = 0x506,
> [1705921025.735752][14382:14385] CHIP:DMG: 					CommandId = 0xd,
> [1705921025.735846][14382:14385] CHIP:DMG: 				},
> [1705921025.735906][14382:14385] CHIP:DMG: 				
> [1705921025.735953][14382:14385] CHIP:DMG: 				StatusIB =
> [1705921025.736002][14382:14385] CHIP:DMG: 				{
> [1705921025.736095][14382:14385] CHIP:DMG: 					status = 0xcb (INVALID_IN_STATE),
> [1705921025.736147][14382:14385] CHIP:DMG: 				},
> [1705921025.736198][14382:14385] CHIP:DMG: 				
> [1705921025.736278][14382:14385] CHIP:DMG: 			},
> [1705921025.736336][14382:14385] CHIP:DMG: 			
> [1705921025.736377][14382:14385] CHIP:DMG: 		},
> [1705921025.736424][14382:14385] CHIP:DMG: 		
> [1705921025.736461][14382:14385] CHIP:DMG: 	],
> [1705921025.736548][14382:14385] CHIP:DMG: 	
> [1705921025.736586][14382:14385] CHIP:DMG: 	InteractionModelRevision = 11
> [1705921025.736622][14382:14385] CHIP:DMG: },
> [1705921025.736763][14382:14385] CHIP:DMG: Received Command Response Status for Endpoint=3 Cluster=0x0000_0506 Command=0x0000_000D Status=0xcb
> [1705921025.736852][14382:14385] CHIP:TOO: Error: IM Error 0x000005CB: General error: 0xcb (INVALID_IN_STATE)
> 

### Bug prevalence

whenevre I do

### GitHub hash of the SDK that was being used

0ff709f25f6a7af68314c1b0a147a76c30c234f6

### Platform

raspi

### Anything else?

AS per Test-Plan, At step 3, when TH sends a ActivateAudioTrack command to DUT specifying the ID value from step 2 DUT should responds with success response but in current SDK got status = 0xcb (INVALID_IN_STATE).
**PFA Test-Plan Reference :** 
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/media.adoc#tc-mediaplayback-6-9-audio-track-verification

**PFA Log below :** 
[mediaplaypack.txt](https://github.com/project-chip/connectedhomeip/files/14009061/mediaplaypack.txt)

chrisdecenzo, 2024-01-22:
Hi @Rajashreekalmane 

I think there is a typo in the command you used (should be activate-audio-track and not activate-text-track). I confirmed that this works for me:
`./out/debug/chip-tool mediaplayback activate-audio-track activeAudioTrackId_1 0 1234 1`

Rajashreekalmane, 2024-01-23:
Thank you, It's worked as expected

closed 2024-01-23 by Rajashreekalmane