project-chip/connectedhomeip #31602: [Tooling] update pyspelling in SDK to support camel-case

cecille, 2024-01-22:
### Reproduction steps / Feature

newer versions fixed the camel-case option to aspell. The spec repo uses this version. We should update the SDK to do the same. Then we can remove some of the excluded files from the spell checker.

### Platform

python

### Type

CI tested

cecille, 2024-01-22:
After updating, we should also prune the wordlist to exclude the things that were added to work around not having camel case. When we did this in the spec, we found lots of spelling errors.

kliao-csa, 2024-02-15:
Pyspelling updated, Reviewdog commented out
Slowly removing camelcase entries from the wordlist - regex indicates there are a few hundred

closed 2024-02-22 by kliao-csa