project-chip/connectedhomeip #31618: [1.2] cannot decode payload_hex on chip-tool trace_file

mmt-10, 2024-01-23:
### Reproduction steps

```
$ ./chip-tool doorlock read door-state 1234 1 --trace_log 1 --trace_file ./psoc6-log/doorstate.log
$ ./chip-tool payload parse-setup-payload 15360017240201250301012404031818290324ff0b18
```
then, chip-tool shows error.

> [1705975377.180310][16350:16350] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_tool_kvs
> [1705975377.180353][16350:16350] CHIP:SPL: Parsing decimalRepresentation: 15360017240201250301012404031818290324ff0b18
> [1705975377.180364][16350:16350] CHIP:TOO: Run command failure: src/setup_payload/ManualSetupPayloadParser.cpp:49: Error 0x00000013

I would like to decode `payload_hex` on below trace file.
[doorstate.log](https://github.com/project-chip/connectedhomeip/files/14017380/doorstate.log)

### Bug prevalence

always

### GitHub hash of the SDK that was being used

commit 181b0cb14ff007ec912f2ba6627e05dfb066c008 (HEAD, tag: v1.2.0.1)

### Platform

core

### Type

Platform Issue

bzbarsky-apple, 2024-01-23:
@mmt-10 I don't understand what you are trying to do.  Message payloads are not the same as either setup payloads or the "additional data" payload in a QR code.  Why do you think "chip-tool payload" (which specifically works with Matter setup payloads) is supposed to be a generic "parse any payload of any packet" thing?

mmt-10, 2024-01-24:
Thank you for your reply.

I understand that I misunderstood `chip-tool payload` function.
I want to know how to decode `payload_hex` in trace file [doorstate.log](https://github.com/project-chip/connectedhomeip/files/14017380/doorstate.log).

bzbarsky-apple, 2024-01-26:
It's just a hex dump of the bytes of the payload, no?  Then you have to look at the exact message type to figure out what it might mean.

mmt-10, 2024-02-29:
@Chapoly1305 
Thank you for your reply.
`--trace_decode 1` could print decoded message.

closed 2024-02-29 by mmt-10