project-chip/connectedhomeip #31624: [BUG] Pairing with Apple Homepod fails

os-mec, 2024-01-23:
### Reproduction steps

Hello CHIP Maintainers,

Im trying to implement a MATTER Device on an embedded linux system and im encountering some problems with the Apple Home eco-system.

When trying to pair the MATTER Device with my Apple Homepod, it goes through the steps for setting up the location and name for the device after which it is added to the Apple Home App but is constantly marked as being not responsive.

The pairing with the Google-Ecosystem works perfectly fine and the only issues I'm encountering are some logic issues on my side with the device.

Thank you in advance for your help :)

### How to reproduce
1. Initiate pairing with Apple Homepod in the Apple Home app via the QR-Code provided by the MATTER application
2. Follow through the steps in the app

### Log
[matter.2.log](https://github.com/project-chip/connectedhomeip/files/14025178/matter.2.log)

### Bug prevalence

Whenever I do this

### GitHub hash of the SDK that was being used

f50e12b0afdfbc940f69d126fd109c5d289aa5ba

### Platform

core

bzbarsky-apple, 2024-01-23:
@os-mec Please report a Feedback Assistant ticket following the instructions below and put the ticket ID here, and I will get people to look at it.

Feedback assistant ticket:
- File a feedback assistant ticket.
https://developer.apple.com/bug-reporting/#feedback-assistant
- Click on the link below using your account:
https://feedbackassistant.apple.com/new-form-response
- Choose iOS/iPadOS to file an issue
- Choose the area as Home App and fill in the details of the type of issue and the description
* Technical Information
Please provide the following
a) Sysdiagnose logs for iPhone / iPad with the timestamp / timeframe of the issue
b) Sysdiagnose logs for all the Home Hubs (HomePod mini / HomePod / Apple TV) with the timestamp / timeframe of the issue
c) Accessory logs with the timestamp / timeframe of the issue
d) mDNS network logs with timestamp / timeframe of the issue
e) Video or screenshot describing the issue (optional but beneficial)
Please make sure you download and install the profiles for iOS and tvOS as provided here. The instructions are also in these links.
* iOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=homekit
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=network
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=mdns
* tvOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=homekit
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=network
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=mdns
For installation instructions, refer to https://developer.apple.com/bug-reporting/profiles-and-logs.
Note: Once you have the profile on the iPhone, it will ask you if you want to install it on other devices. Follow the prompts to install on the HomePod, Apple TV, etc., where you have the described issue and reproduce it.

os-mec, 2024-01-24:
Thanks, the Ticket number is FB13558316. A thing that I want to note here is that I can't seem to reproduce this issue the same way all of the time, occasionally the CASE Session fails on the Sigma1 message. Please let me know if I open a seperate issue for that or if it belongs here too. Also let me know if anything is missing.

Thank you :)

bzbarsky-apple, 2024-01-27:
@os-mec Sorry for the lag; I asked folks to look into that ticket.  Just to confirm: the issue here is `matter.2.log` in the ticket, right?

os-mec, 2024-01-29:
@bzbarsky-apple Yes, the matter.3.log is the log that was generated when generating the Homepod syslog.

bjoern-brandt, 2024-01-31:
Update - I'm working on this device with @os-mec: after installing some profiles (especially the EnableBluetoothCentralMatterClientDeveloperMode one) on the HomePod, and changing some things around in the zap file, the CASE sessions now consistently succeed, but the Apple Home app still shows the device as not responding. The matter log shows no traffic after the initial setup. It appears the HomePod doesn't even try to subscribe to lock events.
Here's a log of our device's view of things: [apple.log](https://github.com/project-chip/connectedhomeip/files/14110676/apple.log), 
[google.log](https://github.com/project-chip/connectedhomeip/files/14112102/google.log).
The Apple and Google devices seem to be doing very similar things, up until the point where the HomePod decides not to communicate with us anymore.

According to another colleague, the same firmware that produced these logs is working fine for them (except that the app won't let them enter the correct PIN, but that's a separate issue). What could cause this discrepancy? I'm pretty sure we're using the same hardware, and I've already checked that we're using the same iOS and tvOS versions (both 17.3).

bjoern-brandt, 2024-01-31:
After creating a new home in the Apple Home app, everything works now. I'm still not sure what caused this behavior, but I'll take it. Thank you for your time.

bzbarsky-apple, 2024-02-03:
Please update the feedback tickets with the relevant info....  If creating a new home fixes things, this sounds like a Home-side problem, which means I'm not the right person to be looking into it (but the people looking at the feedback bits should be).

os-mec, 2024-02-05:
Alright, im closing this issue since the SDK side seems to be resolved. Thank you for your help @bzbarsky-apple !

closed 2024-02-05 by os-mec