project-chip/connectedhomeip #31655: [Tooling] yaml tests: chip-tool websocket ping timeout too low for some CADMIN tests

Apollon77, 2024-01-24:
### Reproduction steps / Feature

It seems that there is no websocket ping-pong happening when the test e.g. waits around 900s till a commissioning window times out.

The exception that happens in the test is then

```
2024-01-24 14:44:18.406 ERROR   14:44:17.615 - TEST ERR  : Traceback (most recent call last):
2024-01-24 14:44:18.406 ERROR   14:44:17.615 - TEST ERR  :   File "/Users/ingof/Dev/matter/connectedhomeip/scripts/tests/yaml/chiptool.py", line 118, in <module>
2024-01-24 14:44:18.406 ERROR   14:44:17.619 - TEST ERR  :     chiptool_py()
2024-01-24 14:44:18.406 ERROR   14:44:17.619 - TEST ERR  :   File "/Users/ingof/Dev/matter/connectedhomeip/.environment/pigweed-venv/lib/python3.9/site-packages/click/core.py", line 1130, in __call__
2024-01-24 14:44:18.407 ERROR   14:44:17.621 - TEST ERR  :     return self.main(*args, **kwargs)
2024-01-24 14:44:18.407 ERROR   14:44:17.621 - TEST ERR  :   File "/Users/ingof/Dev/matter/connectedhomeip/.environment/pigweed-venv/lib/python3.9/site-packages/click/core.py", line 1055, in main
2024-01-24 14:44:18.407 ERROR   14:44:17.621 - TEST ERR  :     rv = self.invoke(ctx)
2024-01-24 14:44:18.407 ERROR   14:44:17.658 - TEST ERR  :   File "/Users/ingof/Dev/matter/connectedhomeip/.environment/pigweed-venv/lib/python3.9/site-packages/click/core.py", line 1404, in invoke
2024-01-24 14:44:18.407 ERROR   14:44:17.659 - TEST ERR  :     return ctx.invoke(self.callback, **ctx.params)
2024-01-24 14:44:18.407 ERROR   14:44:17.659 - TEST ERR  :   File "/Users/ingof/Dev/matter/connectedhomeip/.environment/pigweed-venv/lib/python3.9/site-packages/click/core.py", line 760, in invoke
2024-01-24 14:44:18.407 ERROR   14:44:17.659 - TEST ERR  :     return __callback(*args, **kwargs)
2024-01-24 14:44:18.407 ERROR   14:44:17.659 - TEST ERR  :   File "/Users/ingof/Dev/matter/connectedhomeip/.environment/pigweed-venv/lib/python3.9/site-packages/click/decorators.py", line 26, in new_func
2024-01-24 14:44:18.407 ERROR   14:44:17.659 - TEST ERR  :     return f(get_current_context(), *args, **kwargs)
2024-01-24 14:44:18.407 ERROR   14:44:17.659 - TEST ERR  :   File "/Users/ingof/Dev/matter/connectedhomeip/scripts/tests/yaml/chiptool.py", line 106, in chiptool_py
2024-01-24 14:44:18.407 ERROR   14:44:17.660 - TEST ERR  :     success = send_yaml_command(chiptool, commands[1], server_path, server_arguments, show_adapter_logs, specifications_paths, pics,
2024-01-24 14:44:18.407 ERROR   14:44:17.660 - TEST ERR  :   File "/Users/ingof/Dev/matter/connectedhomeip/.environment/pigweed-venv/lib/python3.9/site-packages/click/decorators.py", line 26, in new_func
2024-01-24 14:44:18.407 ERROR   14:44:17.660 - TEST ERR  :     return f(get_current_context(), *args, **kwargs)
2024-01-24 14:44:18.407 ERROR   14:44:17.660 - TEST ERR  :   File "/Users/ingof/Dev/matter/connectedhomeip/scripts/tests/yaml/tests_tool.py", line 44, in send_yaml_command
2024-01-24 14:44:18.407 ERROR   14:44:17.660 - TEST ERR  :     return ctx.forward(test_tool)
2024-01-24 14:44:18.407 ERROR   14:44:17.660 - TEST ERR  :   File "/Users/ingof/Dev/matter/connectedhomeip/.environment/pigweed-venv/lib/python3.9/site-packages/click/core.py", line 781, in forward
2024-01-24 14:44:18.407 ERROR   14:44:17.660 - TEST ERR  :     return __self.invoke(__cmd, *args, **kwargs)
2024-01-24 14:44:18.407 ERROR   14:44:17.660 - TEST ERR  :   File "/Users/ingof/Dev/matter/connectedhomeip/.environment/pigweed-venv/lib/python3.9/site-packages/click/core.py", line 760, in invoke
2024-01-24 14:44:18.408 ERROR   14:44:17.660 - TEST ERR  :     return __callback(*args, **kwargs)
2024-01-24 14:44:18.408 ERROR   14:44:17.661 - TEST ERR  :   File "/Users/ingof/Dev/matter/connectedhomeip/.environment/pigweed-venv/lib/python3.9/site-packages/click/decorators.py", line 26, in new_func
2024-01-24 14:44:18.408 ERROR   14:44:17.661 - TEST ERR  :     return f(get_current_context(), *args, **kwargs)
2024-01-24 14:44:18.408 ERROR   14:44:17.661 - TEST ERR  :   File "/Users/ingof/Dev/matter/connectedhomeip/scripts/tests/yaml/runner.py", line 361, in chiptool
2024-01-24 14:44:18.408 ERROR   14:44:17.661 - TEST ERR  :     return ctx.forward(websocket)
2024-01-24 14:44:18.408 ERROR   14:44:17.661 - TEST ERR  :   File "/Users/ingof/Dev/matter/connectedhomeip/.environment/pigweed-venv/lib/python3.9/site-packages/click/core.py", line 781, in forward
2024-01-24 14:44:18.408 ERROR   14:44:17.661 - TEST ERR  :     return __self.invoke(__cmd, *args, **kwargs)
2024-01-24 14:44:18.408 ERROR   14:44:17.661 - TEST ERR  :   File "/Users/ingof/Dev/matter/connectedhomeip/.environment/pigweed-venv/lib/python3.9/site-packages/click/core.py", line 760, in invoke
2024-01-24 14:44:18.408 ERROR   14:44:17.661 - TEST ERR  :     return __callback(*args, **kwargs)
2024-01-24 14:44:18.408 ERROR   14:44:17.661 - TEST ERR  :   File "/Users/ingof/Dev/matter/connectedhomeip/.environment/pigweed-venv/lib/python3.9/site-packages/click/decorators.py", line 84, in new_func
2024-01-24 14:44:18.408 ERROR   14:44:17.662 - TEST ERR  :     return ctx.invoke(f, obj, *args, **kwargs)
2024-01-24 14:44:18.408 ERROR   14:44:17.662 - TEST ERR  :   File "/Users/ingof/Dev/matter/connectedhomeip/.environment/pigweed-venv/lib/python3.9/site-packages/click/core.py", line 760, in invoke
2024-01-24 14:44:18.408 ERROR   14:44:17.662 - TEST ERR  :     return __callback(*args, **kwargs)
2024-01-24 14:44:18.408 ERROR   14:44:17.662 - TEST ERR  :   File "/Users/ingof/Dev/matter/connectedhomeip/scripts/tests/yaml/runner.py", line 333, in websocket
2024-01-24 14:44:18.408 ERROR   14:44:17.662 - TEST ERR  :     return loop.run_until_complete(runner.run(parser_group.builder_config, runner_config))
2024-01-24 14:44:18.408 ERROR   14:44:17.662 - TEST ERR  :   File "/Users/ingof/Dev/matter/connectedhomeip/.environment/cipd/packages/python/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
2024-01-24 14:44:18.408 ERROR   14:44:17.664 - TEST ERR  :     return future.result()
2024-01-24 14:44:18.408 ERROR   14:44:17.664 - TEST ERR  :   File "/Users/ingof/Dev/matter/connectedhomeip/scripts/py_matter_yamltests/matter_yamltests/runner.py", line 150, in run
2024-01-24 14:44:18.408 ERROR   14:44:17.665 - TEST ERR  :     raise (result)
2024-01-24 14:44:18.409 ERROR   14:44:17.706 - TEST ERR  :   File "/Users/ingof/Dev/matter/connectedhomeip/scripts/py_matter_yamltests/matter_yamltests/runner.py", line 200, in _run
2024-01-24 14:44:18.409 ERROR   14:44:17.706 - TEST ERR  :     encoded_response = await self.execute(encoded_request)
2024-01-24 14:44:18.409 ERROR   14:44:17.706 - TEST ERR  :   File "/Users/ingof/Dev/matter/connectedhomeip/scripts/py_matter_yamltests/matter_yamltests/websocket_runner.py", line 69, in execute
2024-01-24 14:44:18.409 ERROR   14:44:17.706 - TEST ERR  :     await instance.send(request)
2024-01-24 14:44:18.409 ERROR   14:44:17.706 - TEST ERR  :   File "/Users/ingof/Dev/matter/connectedhomeip/.environment/pigweed-venv/lib/python3.9/site-packages/websockets/legacy/protocol.py", line 647, in send
2024-01-24 14:44:18.409 ERROR   14:44:17.706 - TEST ERR  :     await self.write_frame(True, opcode, data)
2024-01-24 14:44:18.409 ERROR   14:44:17.706 - TEST ERR  :   File "/Users/ingof/Dev/matter/connectedhomeip/.environment/pigweed-venv/lib/python3.9/site-packages/websockets/legacy/protocol.py", line 1214, in write_frame
2024-01-24 14:44:18.409 ERROR   14:44:17.706 - TEST ERR  :     await self.drain()
2024-01-24 14:44:18.409 ERROR   14:44:17.706 - TEST ERR  :   File "/Users/ingof/Dev/matter/connectedhomeip/.environment/pigweed-venv/lib/python3.9/site-packages/websockets/legacy/protocol.py", line 1203, in drain
2024-01-24 14:44:18.409 ERROR   14:44:17.707 - TEST ERR  :     await self.ensure_open()
2024-01-24 14:44:18.409 ERROR   14:44:17.707 - TEST ERR  :   File "/Users/ingof/Dev/matter/connectedhomeip/.environment/pigweed-venv/lib/python3.9/site-packages/websockets/legacy/protocol.py", line 944, in ensure_open
2024-01-24 14:44:18.409 ERROR   14:44:17.707 - TEST ERR  :     raise self.connection_closed_exc()
2024-01-24 14:44:18.409 ERROR   14:44:17.707 - TEST ERR  : websockets.exceptions.ConnectionClosedError: no close frame received or sent

```

First I thought it is because of the used ping client timout values set to 240s ... these might also be it, but also increasing them did not help.

The reason seems that chip-tool which is used in interactive mode for the tests uses the defaults of the libwebsocket:
[... cut ...]
I increased the 300s default which was in that libwebsocket ... no change ...

With this current "too short setting somewhere" tests like Test_TC_CADMIN_1_21 or Test_TC_CADMIN_1_22 will not be successful at all (I know these tests are flagged as manual and long running tests but in fact they are good to validate correct behavior nd it would be cool if they can be executed as yaml tests as it was possible as they were still built in in chip tool.

Does anyone have an idea where/how to solve that?

### Platform

core (please add to version below)

### Type

YAML tested

cecille, 2024-01-24:
@vivien-apple

GetUpSii, 2024-04-15:
Hello, "No close frame received or sent" This issue seem to come from websocket，I found a solution on the official website, but I don't know how to implement it on TH. If this is effective, please take a look. Link: https://websockets.readthedocs.io/en/stable/faq/common.html#
[image]
what-does-connectionclosederror-no-close-frame-received-or-sent-mean